aaronwmorris/indi-allsky #1497: Update SyncAPI/Distribution Support Documentation

cfmorrell, 2024-09-02:
I've been able to successfully run a remote SyncAPI server on Ubuntu 24.04 with Python 3.12.  In web_only_setup.sh, I updated lines 509, 515, 523, 524, and 525 to include 24.04 and 3.12 as appropriate.  Other than that, install, run, configure, and sync all worked as documented.  It seems to run without issue.

I did struggle a bit to figure out which OS distros were supported.  

A possible solution is to add a distro support table to the SyncAPI Wiki page or add a line to the notes column of the Distro Support 
table in the README.  I'm not sure how many folks are playing with a remote server, so this may be a non-issue.

aaronwmorris, 2024-09-03:
I will get that script updated.  I was going through adding Ubuntu 24.04 support and I got distracted because there are problems with Python 3.12 and the pyindi-client module, but that is not an issue for a web-only deployment.

aaronwmorris, 2024-09-13:
Closing this for now.  Please let me know if there is anything else I can do.

closed 2024-09-13 by aaronwmorris