aaronwmorris/indi-allsky #1503: JPG or PNG not working after update

sonny486, 2024-09-05:
Updated this morning and now JPG daytime image processing not working.  Shows a quarter image only.  Not the full image.

I had to revert to RAW DNG.

Will check on this some more.

sonny486, 2024-09-05:
Got it working by adding --width --height to the libcamera options

sonny486, 2024-09-05:
[image]

aaronwmorris, 2024-09-08:
That is weird...  I am currently on vacation and I will be back later this week.

sonny486, 2024-09-08:
No worries, enjoy your time off! 

I see if I can see anything in the code.

sonny486, 2024-09-08:
Well it ended up being the LIBCAMERA library.  It must have had an issue, I did a system update and I noticed it installed new libcamera utilities and system files.

It's working fine now with the latest update.
Closing this issue.

closed 2024-09-08 by sonny486

aaronwmorris, 2024-09-08:
Glad you figured it out!