aaronwmorris/indi-allsky #1504: Difficulty connecting to BME280 sensor on RPi 5 despite system seeing it and being able to display results

abdiastroboy, 2024-09-08:
Hi

I tried with much frustration to get this to work.
I was using the web server to -> config -> sensors and choose sensor A as the BME sensor to 280, on bus 77 
I was able to check via a python script that the sensor was reading and I could display results

The other thing of note is the system time seems to be always off despite the ripi showing correct time (its connected to wifi all the time) and even after clicking update time it still says some wrong time and date in august. I m not sure if this is relevant.

I can tinker and modify code and files if I am shown where to and know my way around the terminal etc and thonny but I'm no programmer at all, just learning as I go.

code

from smbus2 import SMBus
import bme280

# Initialize the I2C bus and BME280 address
bus = SMBus(1)  # I2C bus 1
address = 0x77  # BME280 sensor address, change to 0x76 if necessary

# Load calibration parameters
calibration_params = bme280.load_calibration_params(bus, address)

# Read and display sensor data
data = bme280.sample(bus, address, calibration_params)

# Safely print the degree symbol using Unicode
print(f"Temperature: {data.temperature} \u00B0C")
print(f"Pressure: {data.pressure} hPa")
print(f"Humidity: {data.humidity} %")

Shell output

>>> %Run -c $EDITOR_CONTENT
Temperature: 16.822673378232867 Â°C
Pressure: 993.2317995054409 hPa
Humidity: 43.792248340356586 %
>>> 

However for the life of me I can't get it to display anything in the charts it doesn't seem to see it.
I set the correct i2c address
I believe the pin is not needed for i2c

I am connecting the bme280 from piicodev via piicodev adapter on the rasberry pi 5 with a NVME base and rpi5 PD board feeding power via usb c

https://core-electronics.com.au/piicodev-adapter-for-raspberry-pi.html
https://core-electronics.com.au/piicodev-atmospheric-sensor-bme280.html
https://core-electronics.com.au/piicodev-cable-100mm.html
https://core-electronics.com.au/pd-power-expansion-board-raspberry-pi-5.html which doesnt interface with i2c controller at all
https://core-electronics.com.au/nvme-base-for-raspberry-pi-5-nvme-base.html which also doesn't interface with i2c controller

I am not sure what I need to do
I thought the bme280 library were already installed when installing the Indi-allsky however it seems it wasn the case or perhaps a different version so I installed with sudo apt install python3-bme280

then ran the script above and it works perfectly fine

I am not entirely clear if there is other setup involved and how the program handles the sensors I know the wiki says it works but Im not sure what I have to do on my end to get it to work

CHATGPT tells me to modify the json file I can only see config.json.save and I didn't want to mess with it and besides I couldn't even open it due to permissions but that is a different problem.

I also have a second bme 280 sensor and a VEML 7700 sensor that I m hoping to setup with the one bme sensor being internal and the trigger via a relay switch to trigger a fan in the case and also the dew heater (still figuring out that) and the second one being housed in an custom enclosure to send the surrounding temp/humidty etc.

aaronwmorris, 2024-09-08:
Can you post the Support Info information here?

abdiastroboy, 2024-09-08:
HelloYou mean what I posted on Github?HiI tried with much frustration to get this to work.I was using the web server to -> config -> sensors and choose sensor A as the BME sensor to 280, on bus 77I was able to check via a python script that the sensor was reading and I could display resultsThe other thing of note is the system time seems to be always off despite the ripi showing correct time (its connected to wifi all the time) and even after clicking update time it still says some wrong time and date in august. I m not sure if this is relevant.I can tinker and modify code and files if I am shown where to and know my way around the terminal etc and thonny but I'm no programmer at all, just learning as I go.codefrom smbus2 import SMBusimport bme280#Initialize the I2C bus and BME280 addressbus = SMBus(1) # I2C bus 1address = 0x77 # BME280 sensor address, change to 0x76 if necessary#Load calibration parameterscalibration_params = bme280.load_calibration_params(bus, address)#Read and display sensor datadata = bme280.sample(bus, address, calibration_params)#Safely print the degree symbol using Unicodeprint(f"Temperature: {data.temperature} \u00B0C")print(f"Pressure: {data.pressure} hPa")print(f"Humidity: {data.humidity} %")Shell output%Run -c $EDITOR_CONTENTTemperature: 16.822673378232867 Â°CPressure: 993.2317995054409 hPaHumidity: 43.792248340356586 %However for the life of me I can't get it to display anything in the charts it doesn't seem to see it.I set the correct i2c addressI believe the pin is not needed for i2cI am connecting the bme280 from piicodev via piicodev adapter on the rasberry pi 5 with a NVME base and rpi5 PD board feeding power via usb chttps://core-electronics.com.au/piicodev-adapter-for-raspberry-pi.htmlhttps://core-electronics.com.au/piicodev-atmospheric-sensor-bme280.htmlhttps://core-electronics.com.au/piicodev-cable-100mm.htmlhttps://core-electronics.com.au/pd-power-expansion-board-raspberry-pi-5.html which doesnt interface with i2c controller at allhttps://core-electronics.com.au/nvme-base-for-raspberry-pi-5-nvme-base.html which also doesn't interface with i2c controllerI am not sure what I need to doI thought the bme280 library were already installed when installing the Indi-allsky however it seems it wasn the case or perhaps a different version so I installed with sudo apt install python3-bme280then ran the script above and it works perfectly fineI am not entirely clear if there is other setup involved and how the program handles the sensors I know the wiki says it works but Im not sure what I have to do on my end to get it to workCHATGPT tells me to modify the json file I can only see config.json.save and I didn't want to mess with it and besides I couldn't even open it due to permissions but that is a different problem.I also have a second bme 280 sensor and a VEML 7700 sensor that I m hoping to setup with the one bme sensor being internal and the trigger via a relay switch to trigger a fan in the case and also the dew heater (still figuring out that) and the second one being housed in an custom enclosure to send the surrounding temp/humidty etc.

aaronwmorris, 2024-09-08:
No, there is a web view under Tools -> Support Info that has configuration information about your system.  I would like to look at that to help.

Since you are using a Raspi 5, you may also need to perform these actions for the GPIO modules to work correctly.

```
source virtualenv/indi-allsky/bin/activate

pip uninstall RPi.GPIO rpi.lgpio

pip install rpi.lgpio
```

abdiastroboy, 2024-09-08:
Ok thanks I will try to do that ASAP, I am working for next 36hrs so might be a bit of a delay.I will try to get this info to you and Ill try the gpio things (i do remember reading the way rpi5 queries /managed the pins is different cw previous generations)Regards

abdiastroboy, 2024-09-08:
I presume the gpio issue is the one discussed hereRaspberry Pi 5 - gpiod vs RPi.GPIO - Raspberry Pi Forumsforums.raspberrypi.comRegards

abdiastroboy, 2024-09-09:
> Here is the Support info
>
> /home/ascar5/indi-allsky/misc/support_info.sh: line 62: warning: command
> substitution: ignored null byte in input
> #################################
> ### indi-allsky support info  ###
> #################################
> ```
>
> Distribution: debian
> Release: 12
> Arch: aarch64
> Bits: 64
>
> CPUs: 4
> Memory: 8245584 kB
>
> System: Raspberry Pi 5 Model B Rev 1.0
>
> Linux asc 6.6.47+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.47-1+rpt1
> (2024-09-02) aarch64 GNU/Linux
>
> Time
> Mon  9 Sep 00:23:03 AEST 2024
>
> [... 1359 lines cut ...]
>
> ```
> #################################
> ###     end support info      ###
> #################################

Also I tried those commands and could not run the

***@***.***:~ $ source virtualenv/indi-allsky/bin/activate
it said the following

bash: virtualenv/indi-allsky/bin/activate: No such file or directory

Also the ;

pip uninstall RPi.GPIO rpi.lgpio

pip install rpi.lgpio

wouldnt work said something about managed in external environment

I used chatgpt to figure out the following which worked

sudo apt remove python3-rpi.gpio python3-lgpio

sudo apt install python3-lgpio

So thats done and I still cant see the sensor show up must be doing
something silly

But i can still see it on the 77 bus of the i2c with sudo i2cdetect command
and running the python script in thonny i can get a reading from it.

aaronwmorris, 2024-09-09:
You will have to be in the indi-allsky folder to run the pip commands.

aaronwmorris, 2024-09-13:
@abdiastroboy Were you able to work through this yet?

abdiastroboy, 2024-10-16:
HiYes i managed to get it all working, I had some issue with graphs not showing. But they show up once the camera is connected. is this expected behaviour (took me awhile to figure it out)I wrote a py script to control a relay to control a 12v noctua 80mm fan and dew heater based on temp/dew/humidity conditions, that’s working well.I realised that you have those options under sensors in the config but although i can see the sensor the fans and 12v heater are unable to be controlled via indi all sky.I was trying to get pwm working for the fan and managed to set gpio 12 to pwmo but despite being connected appropriately and the settings being changed via sysfs it would never do anything. I thought it was the relay initially but even direct connection doesn’t seem to do anything. I dont have an oscilloscope and i know very little programming to do anymore to make sure the pin is outputting appropriate pwm signal. The blue pin of the noctua (pwm) was connected to gpio 12 and fan was connected via relay and the software was written to turn on relay and then the pwm control logic, but although software ran without issues the fan was either fully on or off even a direct connection didnt help.  because its a 12v fan i had to wire it to a 12v supply and then connect the pwm pin to gpio 12 maybe this way is not how its done(the pwm pin 5v but probably is ok with getting signalled at 3.3) i gave up with pwm after trying for 2 months of writing code.i also have the light sensor veml7700 im not sure how indi all sky integrates this input to do something? am i missing something, is it possible to get it to sense night and day or sky brightness and adjust camera values for gain/exposure etc . I was planning to write a script where it would use the flux to turn on or off the rpi system but haven’t figured out if thats useful.So far it all works will assemble the final wiring of the box and the soldering this weekend.Have a rpi5 with a nvme adapter and m2 ssd and rpi5 power distribution board sandwiched on top. Connected to x2 bme280 sensors one inside box and one outside with 2 different i2c addresses.fan/heater 12 v12v power to box via dc coming in to power distribution board sending power to piand then power distribution board connected to supply 12 v to fan/heater and relayrelay in turn is connected to fan/heater and back to gpio pins on pi for control.It all works, only missing thing is what to do with the veml output.I have to make my py script a service such that its always running even on reboot similar to what indi all sky does.Been an interesting experiment.AR

aaronwmorris, 2024-10-17:
For now, the Lux sensors like the VEML7700 data is only able to really be used as graph data on the Chart view.

As for the Fan, as you said, the blue pwm pin MIGHT work on 3.3v logic.  There is no info in the white paper on this.

There is a warning in the Noctua whitepaper that hooking up 12v to the PWM pin will damage the fan.  If that was done by accident, that may be why it never worked.

GH-Bill, 2024-10-29:
Aaron, I'm also having the same issue. I can run a BME test and it shows 
[image]
However, I get no Chart Activity and Logs show  
24-10-29T16:17:42.693829-04:00 AllSky [WARNING] Capture-3-3181/MainThread capture._periodic_tasks() [1064]: Periodic tasks triggered
2024-10-29T16:17:42.134475-04:00 AllSky [INFO] MainProcess-2450/MainThread allsky._queueManualTasks() [1223]: Checking for manually submitted tasks
2024-10-29T16:17:42.129737-04:00 AllSky [INFO] MainProcess-2450/MainThread allsky._startSensorWorker() [519]: Starting Sensor-33 worker
2024-10-29T16:17:42.129708-04:00 AllSky [ERROR] MainProcess-2450/MainThread allsky._startSensorWorker() [512]: Sensor worker exception: 
2024-10-29T16:17:42.129673-04:00 AllSky [ERROR] MainProcess-2450/MainThread allsky._startSensorWorker() [512]: Sensor worker exception: IndentationError: unexpected indent
2024-10-29T16:17:42.129636-04:00 AllSky [ERROR] MainProcess-2450/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:     var1 = (var3 + self._pressure_calib[1] * var1) / 524288.0
2024-10-29T16:17:42.129609-04:00 AllSky [ERROR] MainProcess-2450/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:   File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/adafruit_bme280/basic.py", line 226
2024-10-29T16:17:42.129582-04:00 AllSky [ERROR] MainProcess-2450/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:     from adafruit_bme280.basic import Adafruit_BME280
2024-10-29T16:17:42.129557-04:00 AllSky [ERROR] MainProcess-2450/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:   File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/adafruit_bme280/advanced.py", line 33, in <module>
2024-10-29T16:17:42.129530-04:00 AllSky [ERROR] MainProcess-2450/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:     from adafruit_bme280 import advanced as adafruit_bme280
2024-10-29T16:17:42.129503-04:00 AllSky [ERROR] MainProcess-2450/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:   File "/home/pi/indi-allsky/indi_allsky/devices/sensors/tempSensorBme280.py", line 105, in __init__
2024-10-29T16:17:42

I can not provide a Support Log as it just spins and shows loading. I have been able to generate Support information in the past. Not sure why it's acting up now. Any suggestions so I can provide the Support file?

Bill
PS Raspberry Pi5

aaronwmorris, 2024-10-29:
There is a test script on the command line that you may test the sensor configurations without starting indi-allsky.

```
source virtualenv/indi-allsky/bin/activate

./misc/sensor_test.py
```

Similarly, the Support Info may be generated from the command line.

```
./misc/support_info.sh
```

GH-Bill, 2024-10-29:
[image]

GH-Bill, 2024-10-29:
I do see an error wit /etc/pip.conf..  thanks for taking a look.  
Bill

Linux AllSky 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
Hostname
AllSky
AllSky 

Time
Tue 29 Oct 16:52:11 EDT 2024

System timezone
US/Michigan

Uptime
 16:52:11 up 20 min,  2 users,  load average: 0.46, 0.33, 0.23

Memory
               total        used        free      shared  buff/cache   available
Mem:         4142160     1045152     2013856       85040     1234912     3097008
Swap:         524272           0      524272

Filesystems
Filesystem      1K-blocks     Used  Available Use% Mounted on
udev              1896000        0    1896000   0% /dev
tmpfs              414224     8432     405792   3% /run
/dev/nvme0n1p2 1922735800 16231248 1808839148   1% /
tmpfs             2071072     1136    2069936   1% /dev/shm
tmpfs                5120       48       5072   1% /run/lock
/dev/nvme0n1p1     522230    77094     445136  15% /boot/firmware
tmpfs              414208      176     414032   1% /run/user/1000
/dev/sda1      2047934464 43230592 2004703872   3% /media/pi/ALLSKY_USB

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
37500

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether d8:3a:dd:8c:88:c5 brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.20/24 brd 192.168.2.255 scope global dynamic noprefixroute eth0
       valid_lft 85160sec preferred_lft 85160sec
    inet6 2600:1702:3200:81d0:2bb6:663c:21ff:5d72/64 scope global dynamic noprefixroute 
       valid_lft 3551sec preferred_lft 3551sec
    inet6 fe80::c7ff:9297:21f0:d164/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether d8:3a:dd:8c:88:c7 brd ff:ff:ff:ff:ff:ff
    inet 192.168.68.66/22 brd 192.168.71.255 scope global dynamic noprefixroute wlan0
       valid_lft 5965sec preferred_lft 5965sec
    inet6 fe80::c95b:c90d:4a5e:9a77/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever

User info
uid=1000(pi) gid=1000(pi) groups=1000(pi),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),102(input),105(render),106(netdev),115(lpadmin),993(gpio),994(i2c),995(spi),999(systemd-journal)

Process info
[... 695 lines cut ...]
ii  python3-wheel                            0.38.4-2                            all          built-package format for Python
ii  python3-wheel-whl                        0.38.4-2                            all          built-package format for Python (wheel wheel)
ii  python3-wrapt                            1.14.1-2+b2                         arm64        decorators, wrappers and monkey patching. - Python 3.x
ii  python3-yaml                             6.0-3+b2                            arm64        YAML parser and emitter for Python3
ii  python3-zipp                             1.0.0-6                             all          pathlib-compatible Zipfile object wrapper - Python 3.x
ii  python3.11                               3.11.2-6+deb12u3                    arm64        Interactive high-level object-oriented language (version 3.11)
ii  python3.11-dev                           3.11.2-6+deb12u3                    arm64        Header files and a static library for Python (v3.11)
ii  python3.11-minimal                       3.11.2-6+deb12u3                    arm64        Minimal subset of the Python language (version 3.11)
ii  python3.11-venv                          3.11.2-6+deb12u3                    arm64        Interactive high-level object-oriented language (pyvenv binary, version 3.11)

Detected indi-allsky virtualenv
virtualenv python: Python 3.11.2
virtualenv PATH: /home/pi/indi-allsky/virtualenv/indi-allsky/bin:/usr/local/bin:/usr/bin:/bin
flask command: /home/pi/indi-allsky/virtualenv/indi-allsky/bin/flask

virtualenv python modules
Configuration file could not be loaded.
While reading from '/etc/pip.conf' [line  3]: option 'break-system-packages' in section 'global' already exists

```
###############
###  ERROR  ###
###############

The script exited abnormally, please try to run again...

GH-Bill, 2024-10-29:
Took a look at the pip.conf and removed the duplicate lines. Here is the latest Support Log

/home/pi/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 4142160 kB

System: Raspberry Pi 5 Model B Rev 1.0

Linux AllSky 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
Hostname
AllSky
AllSky 

Time
Tue 29 Oct 17:08:24 EDT 2024

System timezone
US/Michigan

Uptime
 17:08:24 up 3 min,  2 users,  load average: 0.86, 0.72, 0.34

Memory
               total        used        free      shared  buff/cache   available
Mem:         4142160     1437536     1377808      184336     1577296     2704624
Swap:         524272           0      524272

Filesystems
Filesystem      1K-blocks     Used  Available Use% Mounted on
udev              1896000        0    1896000   0% /dev
tmpfs              414224     8368     405856   3% /run
/dev/nvme0n1p2 1922735800 16237388 1808833008   1% /
tmpfs             2071072    23328    2047744   2% /dev/shm
tmpfs                5120       48       5072   1% /run/lock
/dev/nvme0n1p1     522230    77094     445136  15% /boot/firmware
tmpfs              414208      176     414032   1% /run/user/1000
/dev/sda1      2047934464 43230592 2004703872   3% /media/pi/ALLSKY_USB

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
37500

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether d8:3a:dd:8c:88:c5 brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.20/24 brd 192.168.2.255 scope global dynamic noprefixroute eth0
       valid_lft 86179sec preferred_lft 86179sec
[... 1400 lines cut ...]
    "CERT_BYPASS": true,
    "USERNAME": "",
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}Â° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,300 (Left)\n# anchor:la (Left Justified)\n# color:150.150.150\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": true,
    "DAYTIME_TRACK": true,
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}Â° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,200 (Right)\n# anchor:ra (Right Justified)\n# color:150.150.150\nSatellites"
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-10-29:
What is really weird is this message:

```
2024-10-29T16:17:42.129673-04:00 AllSky [ERROR] MainProcess-2450/MainThread allsky._startSensorWorker() [512]: Sensor worker exception: IndentationError: unexpected indent
2024-10-29T16:17:42.129636-04:00 AllSky [ERROR] MainProcess-2450/MainThread allsky._startSensorWorker() [512]: Sensor worker exception: var1 = (var3 + self._pressure_calib[1] * var1) / 524288.0
2024-10-29T16:17:42.129609-04:00 AllSky [ERROR] MainProcess-2450/MainThread allsky._startSensorWorker() [512]: Sensor worker exception: File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/adafruit_bme280/basic.py", line 226
```

This is saying there are syntax errors in the python modules.

As for the /etc/pip.conf error... That file is not on any of my systems.  I am assuming this is not a fresh system image.  It appears to have customizations configured.  You may want to start with a fresh image.

aaronwmorris, 2024-10-29:
You can also try re-installing the python modules.

```
deactivate

cd indi-allsky

rm -fR virtualenv/indi-allsky/

pip cache purge

# re-run setup.sh
```

GH-Bill, 2024-10-30:
Aaron, thanks. I removed the USB Dongle and USB stick and reloaded the OS, Ran Setup and the BME280 works perfectly now.
Just need to take a look at why the GPS Dongle is recognized but not getting a Fix. 
Thanks much for all you do with the program and the support you provide.

Bill

aaronwmorris, 2024-10-30:
Excellent!

I have a GPS dongle (no external antenna) that has trouble getting a fix if it is inside.  Outside (in the enclosure) it works fine.

closed 2025-02-20 by aaronwmorris