aaronwmorris/indi-allsky #1505: Newbie (sorry) QHY5iii224c or ASI?

khobar95, 2024-09-09:
I have been using Thomas Jacquin's allsky with my RPi4 setup for quite some time. I am also using his software on a Pi2W that has started to become unreliable. I wanted to test my QHY5iii224C and found Indi Allsky. 

I have Indi Allsky installed on a 2GB Le Potato running Bullseye. I got all the way through the setup, selected QHY support as well as ASI and a couple of others as I have a variety of cameras. I plugged in my QHY5iii224C camera and it is not detected by Indi Allsky.

I checked using lsusb, and the camera shows up as 
Bus 001 Device 011: ID 1618:0224 Cypress WestBridge

I also tried an infamous ASI120Mini that shows up as:
Bus 001 Device 012: ID 03c3:120c ZWO ASI120MM Mini

But it doesn't show up in the available cameras either. 

I'm sure this is a newbie issue, but where do I go from here? Thanks!

aaronwmorris, 2024-09-09:
I am on mobile, pardon any spelling mistakes.  This is probably due to the firmware tool fxload not supporting USB3.

I have a wiki page to help with the fix.  https://github.com/aaronwmorris/indi-allsky/wiki/Build-fxload-for-USB3-support

khobar95, 2024-09-09:
Wow - thank you for the quick reply!

I got all the way through without incident, and now lsusb shows: 
Bus 001 Device 012: ID 1618:0225 QHYCCD Titan224U

I had tried the ASI120MM-Mini, and Indi-allsky seems to be stuck on that cam for some reason even though it's not connected. 

Will continue working on this project. Thanks again for the help!

khobar95, 2024-09-09:
Good news! I have an image of my desk from my qhy camera! 

Lat/Long are also suddenly working!

This is brilliant! 

Now, to resize and reprint my 3D parts to support this cam!

aaronwmorris, 2024-09-09:
Fantastic!

aaronwmorris, 2024-09-09:
You may hide the ASI120 camera in the web interface by using the instructions here:  https://github.com/aaronwmorris/indi-allsky/wiki/SQL-queries

khobar95, 2024-09-17:
Thank you so much for your help. I've been building an enclosure for a Le Potato and the QHY224 camera, and while I was at it I went through the compile process from scratch (since I installed a 16GB emmc). Everything went smooth as silk!

closed 2024-09-17 by khobar95