aaronwmorris/indi-allsky #1507: Setup fails after Ubuntu 24.04 LTS upgrade

wjschmitt, 2024-09-14:
This is the error I get:

**** Python virtualenv setup ****
./setup.sh: line 1823: /home/uwsd/indi-allsky/virtualenv/indi-allsky/bin/pip3: cannot execute: required file not found

I've verified that pip3 is there and version is:
pip 24.0 from /usr/lib/python3/dist-packages/pip (python 3.12)

Any suggestions?  Thanks.

aaronwmorris, 2024-09-14:
You may just want to nuke the virtualenv and recreate it.

```
# This command only needed if you need to deactivate the virtualenv
deactivate

rm -fR virtualenv/indi-allsky

pip cache purge

./setup.sh
```

wjschmitt, 2024-09-14:
OK.  That did the trick.  Setup is proceeding.  One question while you're there - when selecting the camera driver, I don't see anything for indi_pylibcamera, so I've been selecting indi_webcam_ccd.  That seems to work, but is there a better option?

aaronwmorris, 2024-09-14:
You will have to manually configure the indiserver service with the indi_pylibcamera driver.

I have the instructions for setup here:  https://github.com/aaronwmorris/indi-allsky/wiki/indi_pylibcamera-setup

wjschmitt, 2024-09-14:
Yes. The indi_pylibcamera driver is already setup.  There is just no selection for it when running the indi-allsky setup.

And I spoke too soon.  Setup is now failing on the apache2.service failing to start.  I made sure its installed and enabled.  Didn't see this on 22.04.

aaronwmorris, 2024-09-14:
Did you happen to every try to install TJ's allsky on this system?

wjschmitt, 2024-09-14:
No.  Never installed that.

aaronwmorris, 2024-09-14:
Can you show the output of this:

```
sudo ss -anp | grep ":80\|:443"
```

wjschmitt, 2024-09-14:
tcp   TIME-WAIT 0      0                                               192.168.1.224:33390       38.91.101.174:443                                                                                                                                                                                                              
tcp   TIME-WAIT 0      0                                               192.168.1.224:53724        192.168.1.34:80

aaronwmorris, 2024-09-14:
Okay, that did not really show anything.

Try this:

```
sudo tail -n 30 /var/log/apache2/error.log
```

wjschmitt, 2024-09-14:
[Sat Sep 14 00:00:02.108524 2024] [ssl:warn] [pid 198566:tid 131512926881664] AH01909: 127.0.1.1:443:0 server certificate does NOT include an ID which matches the server name
[Sat Sep 14 00:00:02.109346 2024] [mpm_event:notice] [pid 198566:tid 131512926881664] AH00489: Apache/2.4.52 (Ubuntu) OpenSSL/3.0.2 configured -- resuming normal operations
[Sat Sep 14 00:00:02.109355 2024] [core:notice] [pid 198566:tid 131512926881664] AH00094: Command line: '/usr/sbin/apache2'
[Sat Sep 14 09:42:22.783557 2024] [mpm_event:notice] [pid 198566:tid 131512926881664] AH00492: caught SIGWINCH, shutting down gracefully
[Sat Sep 14 09:42:22.843883 2024] [ssl:warn] [pid 626930:tid 138381158397824] AH01909: 127.0.1.1:443:0 server certificate does NOT include an ID which matches the server name
[Sat Sep 14 09:42:22.851959 2024] [ssl:warn] [pid 626931:tid 138381158397824] AH01909: 127.0.1.1:443:0 server certificate does NOT include an ID which matches the server name
[Sat Sep 14 09:42:22.853923 2024] [mpm_event:notice] [pid 626931:tid 138381158397824] AH00489: Apache/2.4.52 (Ubuntu) OpenSSL/3.0.2 configured -- resuming normal operations
[Sat Sep 14 09:42:22.853946 2024] [core:notice] [pid 626931:tid 138381158397824] AH00094: Command line: '/usr/sbin/apache2'
[Sat Sep 14 09:53:59.458604 2024] [mpm_event:notice] [pid 626931:tid 138381158397824] AH00491: caught SIGTERM, shutting down

aaronwmorris, 2024-09-14:
Does `sudo systemctl restart apache2` work without error?

wjschmitt, 2024-09-14:
No.  Same error as during setup.sh about failed to start.  sudo  systemctl status apache2.service shows:
× apache2.service - The Apache HTTP Server
     Loaded: loaded (/usr/lib/systemd/system/apache2.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Sat 2024-09-14 15:27:17 EDT; 1min 51s ago
       Docs: https://httpd.apache.org/docs/2.4/
    Process: 4888 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)
        CPU: 17ms

Sep 14 15:27:17 INDI-VM systemd[1]: Starting apache2.service - The Apache HTTP Server...
Sep 14 15:27:17 INDI-VM apachectl[4890]: AH00526: Syntax error on line 1 of /etc/apache2/sites-enabled/indi-allsky.conf:
Sep 14 15:27:17 INDI-VM apachectl[4890]: Cannot define multiple Listeners on the same IP:port
Sep 14 15:27:17 INDI-VM systemd[1]: apache2.service: Control process exited, code=exited, status=1/FAILURE
Sep 14 15:27:17 INDI-VM systemd[1]: apache2.service: Failed with result 'exit-code'.
Sep 14 15:27:17 INDI-VM systemd[1]: Failed to start apache2.service - The Apache HTTP Server.

wjschmitt, 2024-09-14:
I looked at /etc/apache2/sites-enabled/indi-allsky.conf but didn't see any obvious syntax error.

aaronwmorris, 2024-09-14:
The problem is somewhere in the apache config, the listen directive has multiple entries for the same port.

Try commenting all of the `Listen` directives out of `/etc/apache2/ports.conf` and restart apache.

wjschmitt, 2024-09-14:
No errors in sudo systemctl status apache2.service that time:
● apache2.service - The Apache HTTP Server
     Loaded: loaded (/usr/lib/systemd/system/apache2.service; enabled; preset: enabled)
     Active: active (running) since Sat 2024-09-14 15:45:15 EDT; 1min 18s ago
       Docs: https://httpd.apache.org/docs/2.4/
    Process: 5053 ExecStart=/usr/sbin/apachectl start (code=exited, status=0/SUCCESS)
   Main PID: 5056 (apache2)
      Tasks: 105 (limit: 9446)
     Memory: 9.8M (peak: 10.2M)
        CPU: 49ms
     CGroup: /system.slice/apache2.service
             ├─5056 /usr/sbin/apache2 -k start
             ├─5058 /usr/sbin/apache2 -k start
             └─5059 /usr/sbin/apache2 -k start

Sep 14 15:45:15 INDI-VM systemd[1]: Starting apache2.service - The Apache HTTP Server...
Sep 14 15:45:15 INDI-VM apachectl[5055]: AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 127.0.1.1. Set the 'ServerName' directive globally to suppress this message
Sep 14 15:45:15 INDI-VM systemd[1]: Started apache2.service - The Apache HTTP Server.

wjschmitt, 2024-09-14:
Re-running setup to see what happens.  I'll let you know.  I didn't mean to hijack your Saturday afternoon.

wjschmitt, 2024-09-14:
Yay.  That worked.  Setup completed.  Indi-allsky starts successfully.
Thanks for all your help Aaron.

aaronwmorris, 2024-09-14:
No worries.  Glad it worked out!

closed 2024-09-22 by aaronwmorris