aaronwmorris/indi-allsky #1510: upgrade from indi_v2024.05.1 to indi_v2024.09.1

leforet, 2024-09-16:
Hello, I followed the instructions at https://github.com/aaronwmorris/indi-allsky/wiki/Updating-indi-allsky to upgrade 

The download of the update from GitHub appears to have gone well.

The upgrade seemed to go well, except for this final stretch.
Synchronizing state of apache2.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable apache2
**** Setup HTDOCS folder ****
**** Setup image folder ****
**** Ensure user is a member of the dialout, video, i2c, spi groups ****
**** Disabling Thomas Jacquin's allsky (ignore errors) ****
**Failed to stop allsky.service: Unit allsky.service not loaded.
Failed to disable unit: Unit file allsky.service does not exist.**
**** Starting gunicorn-indi-allsky.socket
**** Update config camera interface ****
2024-09-16 17:00:50,481 [INFO] MainProcess config._load() [912]: Loading configuration from file

There appears to be a small bug or problem, when I check the website (https://192.168.XxX.75/indi-allsky/system) after restarting the service.  I see that the release is still at 
Release indi_v2024.05.1

I checked the git directory after the update and it appears to be at the latest release
pi@piCOMET:~/indi-allsky $ systemctl --user stop indi-allsky
Warning: Stopping indi-allsky.service, but it can still be activated by:
  indi-allsky.timer
pi@piCOMET:~/indi-allsky $ systemctl --user stop gunicorn-indi-allsky
Warning: Stopping gunicorn-indi-allsky.service, but it can still be activated by:
  gunicorn-indi-allsky.socket
pi@piCOMET:~/indi-allsky $ git pull origin main
From https://github.com/aaronwmorris/indi-allsky
 * branch              main       -> FETCH_HEAD
Already up to date.

I've restarted the service & rebooted the pi, with no appe

aaronwmorris, 2024-09-16:
It appears the upgrade was completely successful.  I just did not change the version text in the release.  I will put out an update shortly.

leforet, 2024-09-16:
great, thanks for the update.

aaronwmorris, 2024-09-16:
Updated the version string as part of #1511

closed 2024-09-17 by leforet