aaronwmorris/indi-allsky #1512: Altairastro Gpcam678C

Startupa, 2024-09-17:
I want to change the camera I use for Indi-allsky. Previously I was running an Altair Gpcam 130M. I've done an update and run sh.setup as per the instructions. I still can't see the new 678 on indi-allsky. It now says no camera present. This is a usb3 camera, my 130M was usb2. I have tried running the new camera using a usb2 port on my RPI4.

On reading through the discussion board I note that I would need to install usb3 support. This was for a different make of camera, QHY, I think. 

Also I notice a reference in the documentation to having to change the 678C from RGB to RAW setting. But I don't know where to get the original file to amend.

I don't speak Linux! I'd like to get this sorted out before the new camera is fitted into a new Allsky case in a fortnight, if possible.
Thanks
Anne

aaronwmorris, 2024-09-17:
When you plug in the 678, you will need to restart the indiserver.  The port you pick should not matter.

```
systemctl --user restart indiserver
```

I would just suggest rebooting after plugging in the new camera.  Once you reboot, can you post the output of the Support Info view (under Tools)?  You may also use `./misc/support_info.sh` on the command line.

Startupa, 2024-09-18:
Here's the output of the Support Info view as requested. I've followed your
instructions but I don't think it' worked, it still says on the side of
screen "no camera"
Anne

/home/pi/indi-allsky/misc/support_info.sh: line 62: warning: command
substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
###############################
```

Distribution: debian
Release: 11
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 7999812 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023
aarch64 GNU/Linux

Time
Wed 18 Sep 12:07:05 BST 2024

System timezone
Europe/London

Uptime
 12:07:05 up 2 min,  2 users,  load average: 1.96, 1.17, 0.46

Memory
               total        used        free      shared  buff/cache
available
Mem:         7999812      380264     6828432       49088      791116
7452688
Swap:         102396           0      102396

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root      122541024 11182608 106319784  10% /
devtmpfs         3735680        0   3735680   0% /dev
tmpfs            3999904        0   3999904   0% /dev/shm
tmpfs            1599964     2900   1597064   1% /run
tmpfs               5120        4      5116   1% /run/lock
/dev/mmcblk0p1    261108    31380    229728  13% /boot
tmpfs             799980       28    799952   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
58426

system python: Python 3.9.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group
default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
[... 1245 lines cut ...]
    "TSL2591_GAIN_DAY": "GAIN_LOW",
    "TSL2591_INT_NIGHT": "INTEGRATIONTIME_100MS",
    "TSL2591_INT_DAY": "INTEGRATIONTIME_100MS",
    "VEML7700_GAIN_NIGHT": "ALS_GAIN_1",
    "VEML7700_GAIN_DAY": "ALS_GAIN_1_8",
    "VEML7700_INT_NIGHT": "ALS_100MS",
    "VEML7700_INT_DAY": "ALS_100MS",
    "SI1145_VIS_GAIN_NIGHT": "GAIN_ADC_CLOCK_DIV_32",
    "SI1145_VIS_GAIN_DAY": "GAIN_ADC_CLOCK_DIV_1",
    "SI1145_IR_GAIN_NIGHT": "GAIN_ADC_CLOCK_DIV_32",
    "SI1145_IR_GAIN_DAY": "GAIN_ADC_CLOCK_DIV_1"
  },
  "CHARTS": {
    "CUSTOM_SLOT_1": 10,
    "CUSTOM_SLOT_2": 11,
    "CUSTOM_SLOT_3": 12,
    "CUSTOM_SLOT_4": 13
  },
  "GPS_TIMESYNC": false
}

```
#################################
###     end support info      ###
#################################

Startupa, 2024-09-18:
> That all looks good. My guess is you just need to upgrade INDI which
> should include more up to date camera drivers. This should take about 45-60
> minutes on your system.
>
> ./misc/build_indi.sh
>
> ./misc/rebuild_pyindi-client.sh

Thanks Aaron, I've done the above and the camera has taken a couple of
images. I believe that I need to change the image format to raw. Where is
the file that I need to amend?

Anne

aaronwmorris, 2024-09-18:
Nice!  Can you provide another Support Info dump.  The Altair cameras sometimes have other options that need to be changed for resolution.

In the Config view, near the bottom of the Camera tab, in the `INDI Camera Config (Default)` entry, you can past this to switch to RAW mode.

```json
{
    "PROPERTIES": {},
    "SWITCHES": {
        "CCD_CAPTURE_FORMAT": {
            "on": [
                "INDI_RAW"
            ]
        }
    }
}
```

Startupa, 2024-09-19:
Hi Aaron

Here's the log as requested. I haven't connected the camera or tried
changing any camera settings for the 678.

Regards
Anne

/home/pi/indi-allsky/misc/support_info.sh: line 62: warning: command
substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 11
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 7999812 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023
aarch64 GNU/Linux

Time
Thu 19 Sep 08:27:52 BST 2024

System timezone
Europe/London

Uptime
 08:27:53 up 1 min,  2 users,  load average: 2.43, 1.20, 0.46

Memory
               total        used        free      shared  buff/cache
available
Mem:         7999812      419576     6690940       44920      889296
7410648
Swap:         102396           0      102396

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root      122541024 11561308 105941084  10% /
devtmpfs         3735680        0   3735680   0% /dev
tmpfs            3999904        4   3999900   1% /dev/shm
tmpfs            1599964     2896   1597068   1% /run
tmpfs               5120        4      5116   1% /run/lock
/dev/mmcblk0p1    261108    31380    229728  13% /boot
tmpfs             799980       28    799952   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
34076

system python: Python 3.9.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group
default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
[... 1235 lines cut ...]
    "TSL2591_GAIN_DAY": "GAIN_LOW",
    "TSL2591_INT_NIGHT": "INTEGRATIONTIME_100MS",
    "TSL2591_INT_DAY": "INTEGRATIONTIME_100MS",
    "VEML7700_GAIN_NIGHT": "ALS_GAIN_1",
    "VEML7700_GAIN_DAY": "ALS_GAIN_1_8",
    "VEML7700_INT_NIGHT": "ALS_100MS",
    "VEML7700_INT_DAY": "ALS_100MS",
    "SI1145_VIS_GAIN_NIGHT": "GAIN_ADC_CLOCK_DIV_32",
    "SI1145_VIS_GAIN_DAY": "GAIN_ADC_CLOCK_DIV_1",
    "SI1145_IR_GAIN_NIGHT": "GAIN_ADC_CLOCK_DIV_32",
    "SI1145_IR_GAIN_DAY": "GAIN_ADC_CLOCK_DIV_1"
  },
  "CHARTS": {
    "CUSTOM_SLOT_1": 10,
    "CUSTOM_SLOT_2": 11,
    "CUSTOM_SLOT_3": 12,
    "CUSTOM_SLOT_4": 13
  },
  "GPS_TIMESYNC": false
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-09-19:
Thanks!  Let me know when you have the new camera running.  Most of the other Altair cameras I have experience with start in a low resolution mode and have to be configured to run in the max resolution.  Once you have the camera running, the support info will have the information I need.

Startupa, 2024-09-20:
Hi Aeron
Here's the log with the 678 connected.
Anne

/home/pi/indi-allsky/misc/support_info.sh: line 62: warning: command
substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 11
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 7999812 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023
aarch64 GNU/Linux

Time
Fri 20 Sep 08:17:50 BST 2024

System timezone
Europe/London

Uptime
 08:17:50 up 3 min,  2 users,  load average: 0.49, 0.85, 0.40

Memory
               total        used        free      shared  buff/cache
available
Mem:         7999812      553860     6600080       46268      845872
7275492
Swap:         102396           0      102396

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root      122541024 11553400 105948992  10% /
devtmpfs         3735680        0   3735680   0% /dev
tmpfs            3999904      168   3999736   1% /dev/shm
tmpfs            1599964     2904   1597060   1% /run
tmpfs               5120        4      5116   1% /run/lock
/dev/mmcblk0p1    261108    31380    229728  13% /boot
tmpfs             799980       28    799952   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
36511

system python: Python 3.9.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group
default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN
[... 1472 lines cut ...]
    "TSL2591_GAIN_DAY": "GAIN_LOW",
    "TSL2591_INT_NIGHT": "INTEGRATIONTIME_100MS",
    "TSL2591_INT_DAY": "INTEGRATIONTIME_100MS",
    "VEML7700_GAIN_NIGHT": "ALS_GAIN_1",
    "VEML7700_GAIN_DAY": "ALS_GAIN_1_8",
    "VEML7700_INT_NIGHT": "ALS_100MS",
    "VEML7700_INT_DAY": "ALS_100MS",
    "SI1145_VIS_GAIN_NIGHT": "GAIN_ADC_CLOCK_DIV_32",
    "SI1145_VIS_GAIN_DAY": "GAIN_ADC_CLOCK_DIV_1",
    "SI1145_IR_GAIN_NIGHT": "GAIN_ADC_CLOCK_DIV_32",
    "SI1145_IR_GAIN_DAY": "GAIN_ADC_CLOCK_DIV_1"
  },
  "CHARTS": {
    "CUSTOM_SLOT_1": 10,
    "CUSTOM_SLOT_2": 11,
    "CUSTOM_SLOT_3": 12,
    "CUSTOM_SLOT_4": 13
  },
  "GPS_TIMESYNC": false
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-09-20:
Perfect.  For some reason, Altair cameras start in a lower resolution mode.  You can increase the resolution (and other settings) using this below.  If you chose to increase the resolution, the image sizes are going to be much larger, so you may want to reduce the number of days images are retain to something like 10 to conserve space.

Use this for the `(Default)` config.  This will enable high conversion gain at night.
```json
{
    "PROPERTIES": {},
    "SWITCHES": {
        "CCD_RESOLUTION": {
            "on": [
                "3840 x 2160"
            ]
        },
        "TC_CONVERSION_GAIN": {
            "on": [
                "GAIN_HIGH"
            ]
        },
        "CCD_CAPTURE_FORMAT": {
            "on": [
                "INDI_RAW"
            ]
        }
    }
}
```

Use this for the `Day` config.  Disables HCG during the day.
```json
{
    "PROPERTIES": {},
    "SWITCHES": {
        "CCD_RESOLUTION": {
            "on": [
                "3840 x 2160"
            ]
        },
        "TC_CONVERSION_GAIN": {
            "on": [
                "GAIN_LOW"
            ]
        },
        "CCD_CAPTURE_FORMAT": {
            "on": [
                "INDI_RAW"
            ]
        }
    }
}
```

Startupa, 2024-09-20:
Hi
Where do I do these changes? I assume there's a file somewhere that I need
to edit or replace. (Some time ago I spent days looking for the folder the
images were in!) I'm sorry to be a pain.

Anne

aaronwmorris, 2024-09-20:
No problem.  You would add those to the `INDI Camera Config (Default)` and `INDI Camera Config (Day)` on the Camera Tab under the Config page.

Startupa, 2024-09-20:
I'm still not getting it. Here's a photo of the camera tab in the config
area. There's nothing further down the page.
Anne

aaronwmorris, 2024-09-20:
The image will not come through via email.  The section looks something like this:

[image]

Startupa, 2024-09-20:
Hi again, I found the place to copy the text into. The iPad showed it when
I used that to remote in, the laptop didn't. I've now rebooted with the 678
connected. It isn't taking any images though and has timed out after 305
secs, it wanted me to acknowledge the fact.

I did do a paste of the support log but Gmail appears to have lost it. I'll
send it using the pc shortly.
Anne

Startupa, 2024-09-20:
/home/pi/indi-allsky/misc/support_info.sh: line 62: warning: command
substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 11
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 7999812 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023
aarch64 GNU/Linux

Time
Fri 20 Sep 19:55:01 BST 2024

System timezone
Europe/London

Uptime
 19:55:01 up 3 min,  2 users,  load average: 0.80, 0.88, 0.39

Memory
               total        used        free      shared  buff/cache
available
Mem:         7999812      556952     6564660       45656      878200
7277396
Swap:         102396           0      102396

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root      122541024 11447764 106054628  10% /
devtmpfs         3735680        0   3735680   0% /dev
tmpfs            3999904      168   3999736   1% /dev/shm
tmpfs            1599964     2904   1597060   1% /run
tmpfs               5120        4      5116   1% /run/lock
/dev/mmcblk0p1    261108    31380    229728  13% /boot
tmpfs             799980       28    799952   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
57452

system python: Python 3.9.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group
default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN
group default qlen 1000
    link/ether 2c:cf:67:09:46:54 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state
UP group default qlen 1000
[... 1543 lines cut ...]
    "TSL2591_GAIN_DAY": "GAIN_LOW",
    "TSL2591_INT_NIGHT": "INTEGRATIONTIME_100MS",
    "TSL2591_INT_DAY": "INTEGRATIONTIME_100MS",
    "VEML7700_GAIN_NIGHT": "ALS_GAIN_1",
    "VEML7700_GAIN_DAY": "ALS_GAIN_1_8",
    "VEML7700_INT_NIGHT": "ALS_100MS",
    "VEML7700_INT_DAY": "ALS_100MS",
    "SI1145_VIS_GAIN_NIGHT": "GAIN_ADC_CLOCK_DIV_32",
    "SI1145_VIS_GAIN_DAY": "GAIN_ADC_CLOCK_DIV_1",
    "SI1145_IR_GAIN_NIGHT": "GAIN_ADC_CLOCK_DIV_32",
    "SI1145_IR_GAIN_DAY": "GAIN_ADC_CLOCK_DIV_1"
  },
  "CHARTS": {
    "CUSTOM_SLOT_1": 10,
    "CUSTOM_SLOT_2": 11,
    "CUSTOM_SLOT_3": 12,
    "CUSTOM_SLOT_4": 13
  },
  "GPS_TIMESYNC": false
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-09-20:
I do not see anything wrong at this level.  Can you review the log file at `/var/log/indi-allsky/indi-allsky.log` and see if you can find any exceptions there that might be interesting?

Startupa, 2024-09-21:
Hi
The 678 wouldn't connect on usb3 so I tried usb2. Then I turned the pi off
and tried usb3 again, camera seen this time. I've posted the log below.

From what I can see, the camera starts the exposure with gain for daytime
etc but doesn't complete it. It then eventually puts a window up at 305
seconds. I did have some images before I changed the settings to give the
correct sensor size. I had to do that using my ipad as the laptop still
doesn't show that information

Anne

aaronwmorris, 2024-09-21:
This is rather frustrating.  I guess just remove the `CCD_RESOLUTION` sections from the config.  You will have to reboot for it to change the config back to the defaults.

If that does not help, remove the `TC_CONVERSION_GAIN` seconds.

I will hook up my GPCAM3 224 and see if I can recreate this type of problem.

Startupa, 2024-09-21:
I'll try it tomorrow morning. When I did get images before doing the config
edit there was a large strip at the top of each image.

The 678 was only introduced this year by Altairastro. I shall be seeing
them end of next week, so could check they've amended the indi driver to
support it.  My other alternative is to see if I can exchange it for the
178, the older model. I've been waiting for them to print their Allsky
enclosure as the original one I used with the GPcam isn't deep enough for
the gpcam3 cameras. Incidentally I tried it out using IndigoSky a while ago
and it didn't work at all with that. It's got to be a driver problem.

Anne

Startupa, 2024-09-22:
A further update. I've removed the two text bits, no change. I then removed
all the config text apart from the curly brackets, I couldn't save unless I
left a pair of brackets in the day and night config. It looks as if the 678
can't finish its first exposure. It gives up at 305 secs and indi-allsky
displays a box telling me it's stopped the exposure and show an acknowledge
box for me to click.

I then switched to my Hypercam 178m. That worked perfectly and did a number
of exposures with no issues. I wonder if there's an issue with the new
Altairastro indi driver. I'm therefore going to ask Altair if I can return
the 678c and get a 178c instead, they've still got some on their website.

The 224c you've got has the same size sensor as the gpcam130m that I'm
replacing, I wanted a larger sensor to match the lens I have.

Any other suggestions?

Anne

aaronwmorris, 2024-09-22:
I am quite sure this is just a driver issue.

I am planning on writing a more extensive test script for the camera to help out with debugging issues remotely.

I just noticed another setting I would like you to try with the 678.  I was looking for the USB bandwidth setting, but I did not see it initially, it is possible it was renamed.

Can you try the following settings?

Default
```json
{
    "PROPERTIES": {
        "CCD_CONTROLS": {
            "Speed": 40
        }
    },
    "SWITCHES": {
        "CCD_RESOLUTION": {
            "on": [
                "3840 x 2160"
            ]
        },
        "TC_CONVERSION_GAIN": {
            "on": [
                "GAIN_HIGH"
            ]
        },
        "CCD_CAPTURE_FORMAT": {
            "on": [
                "INDI_RAW"
            ]
        }
    }
}
```

Day
```json
{
    "PROPERTIES": {
        "CCD_CONTROLS": {
            "Speed": 40
        }
    },
    "SWITCHES": {
        "CCD_RESOLUTION": {
            "on": [
                "3840 x 2160"
            ]
        },
        "TC_CONVERSION_GAIN": {
            "on": [
                "GAIN_LOW"
            ]
        },
        "CCD_CAPTURE_FORMAT": {
            "on": [
                "INDI_RAW"
            ]
        }
    }
}
```

Startupa, 2024-09-22:
Thanks Aaron
I was looking at some of the Altair camera configs and noticed the 678
didn't mention band width! I'll have a go in the morning.
Anne

Startupa, 2024-09-23:
Hi Aaron
It didn't make any difference.

I've also contacted Ian at Altairastro, he is trying to find out what's
going out too.

Anne

aaronwmorris, 2024-09-30:
The Altair library/SDK update has just been merged into the indi-3rdparty info.  This will be included in the next release of indi.  There is no information if this will fix this problem, but there is reason to be optimistic.

https://github.com/indilib/indi-3rdparty/pull/984

Startupa, 2024-09-30:
> indilib/indi-3rdparty#984

Thanks for updating me. In the meantime I've switched the 678c to a 178c.
As to the Altair SDK do I just do a indi update as before?

I'm at a starparty for the next week so I won't be able to do much until
I'm home again.

Regards
Anne

aaronwmorris, 2024-09-30:
That is correct.  ~~I am not exactly sure when the next version will be released, but I expect it to be soon.~~ As soon as I wrote that, I checked github and 2.1.0 has been released.  I will update the setup code in indi-allsky to support the new version.

aaronwmorris, 2024-09-30:
Merged #1546 with the updates to build indi 2.1.0.  I tested with my Altair 224C and it does function.  The name of my camera changed slightly, so you may end up with a new camera entry.

Startupa, 2024-10-10:
Hi Aaron
I'm still having issues. I updated indi to 2.something as before but still
no camera seen. I then updated your software, it seemed to want to use an
earlier version of indi, 1.9 something or other. I then tried uninstalling
indi-Allsky. Somewhere along the line everything seems to have scrambled
the sd card and I couldn't connect to the pi. On checking with my display
and keyboard, it wasn't even booting properly.

I therefore did a clean install on the sd card. Just finished so I'll check
it tomorrow.

One question, I was asked at the end of the process whether I wanted to
start indiserver on boot. I selected yes. I hope that's right.

 I haven't done the 2 systemctl commands yet as per your instructions.
That's for tomorrow! I haven't got the new Allsky container yet so no hurry.

Anne

aaronwmorris, 2024-10-10:
The pyindi-client module shows the version of indi that it was compiled against.  There is a separate command to rebuild it against the new version, but that is not likely to cause a big problem anyway.

Starting indi-allsky at boot is the default so that is fine.

Startupa, 2024-10-11:
Hi, indi-allsky is running but I'm now having difficulty accessing its web page. What's the best way to access it easily, please?
Here's the log again.

Anne
/home/annes/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 7997472 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux raspberrypi 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

Time
Fri 11 Oct 19:17:04 BST 2024

System timezone
Europe/London

Uptime
 19:17:04 up 6 min,  2 users,  load average: 1.32, 1.00, 0.47

Memory
               total        used        free      shared  buff/cache   available
Mem:         7997472     1452732     5499388      141184     1306624     6544740
Swap:         204796           0      204796

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             3725564        0   3725564   0% /dev
tmpfs             799748     1324    798424   1% /run
/dev/mmcblk0p2 122300712 10733156 105335844  10% /
tmpfs            3998736    27896   3970840   1% /dev/shm
tmpfs               5120       16      5104   1% /run/lock
/dev/mmcblk0p1    522230    76398    445832  15% /boot/firmware
tmpfs             799744       40    799704   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
66218

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 2c:cf:67:09:46:54 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:09:46:56 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.207/24 brd 192.168.1.255 scope global dynamic noprefixroute wlan0
       valid_lft 86004sec preferred_lft 86004sec
[... 1271 lines cut ...]
  "CHARTS": {
    "CUSTOM_SLOT_1": 10,
    "CUSTOM_SLOT_2": 11,
    "CUSTOM_SLOT_3": 12,
    "CUSTOM_SLOT_4": 13
  },
  "ADSB": {
    "ENABLE": false,
    "DUMP1090_URL": "https://localhost/dump1090/data/aircraft.json",
    "CERT_BYPASS": true,
    "USERNAME": "",
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": false,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,200 (Right)\n# anchor:ra (Right Justified)\n# color:200,200,200\nAircraft"
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-10-11:
The usual method is by accessing by hostname.local, like https://hostname.local/ but this depends on MDNS (or Bonjour on apple devices).  This sometimes misbehaves and you may have to reboot your desktop to fix.

You can also access by IP:  https://192.168.1.207/

Startupa, 2024-10-11:
I'll try the ip web address next time. I used to use the host name and could remote to it fine before I started upgrading the camera. I've got bonjour on my laptop. It came with some other software. It worked this morning until I changed a few settings, max exposure and sncr. I then rebooted, changes show fine. I'll continue trying tomorrow, thanks.Anne

aaronwmorris, 2024-10-15:
Circling back to the 678 camera.  There is a possibility this is a current draw issue.  I have a touptek camera which seems to act similarly to your gpcam678.  In the high resolution mode, the camera will constantly disconnect in the highest resolution mode.

However, if I connect my camera via a powered USB hub, it operates perfectly.

I have been hunting the manual or specs for the gpcam678, but I have yet to really find them published.  The raspberry pi specs state that the USB ports are only capable of delivering 500mA of power.  The ZWO ASI678MC only draws 375mA and I would have a hard time believing the Altair camera would draw much more than that.

Startupa, 2024-10-15:
Hi again

I own a couple of fan-cooled Altair Hypercam cameras. They draw a lot of
power due to the fan from a usb port. It doesn't need a 12 volt power
source as it's not tec-cooled. There's no fan on the 678 though. I did try
my Hypercam 174 with a fan with your software and it worked fine but that's
got 5 micron pixels so doesn't have as many as the 678 which has 2 micron
ones. The file size is much bigger with the 178/678 sensors.

Altair and Toupcam are basically the same cameras but each has their own
software. The 678 worked fine on a standard usb3 port on a laptop.

Having said that the Gpcam3 178 is working ok. I've done darks which went
fine, just now waiting for Ian at Altair to finish my case. Pity the aurora
came too early!

I don't really want a powered hub in my Allsky camera! In any case Ian
switched my 678 to the 178.

Do you want me to try to find out the current drawer of their 678? (They
don't issue manuals, just post info in a support area of their website.)

Anne

aaronwmorris, 2024-10-15:
I do not know if it is really worth investigating unless you already have a measurement device available.  Something like this would be required:  https://www.amazon.com/MakerHawk-Multimeter-Voltmeter-Capacity-Resistance/dp/B07DCSNHNB

If you had a powered USB hub you could also test that.  However, if you have moved on, I think we should just let this go for now.  :-)

Startupa, 2024-10-16:
Ok, thanks for all your help. I'll close the discussion.

Anne

closed 2024-10-16 by Startupa