aaronwmorris/indi-allsky #1513: Tapo C325WB via RTSP/ONVIF using Agent DVR (Linux) instead of Contacam (Windows)

khobar95, 2024-09-17:
I saw where someone was able to integrate a Tapo C325WB camera with indi-allsky via Contacam. I'd like to try using a Linux program called Agent DVR. Thus far, I have a live stream on my Debian 12 box in a browser using localhost:8090. But I was also able to get a live stream using VLC on my windows laptop using RTSP and port 554.

Any chance someone could help me with this? Thanks very much!

aaronwmorris, 2024-09-18:
Unfortunately, indi-allsky does not support image acquisition from RTSP streams.

The Network IP web cameras people have successfully used with indi-allsky usually present the images via HTTP download in the web interface using the PyCURL Camera interface.

khobar95, 2024-09-18:
Hi Aaron and thanks for the reply!

I was hoping to copy this:
https://azmi.my.id/indi-allsky-camera-from-youtube-observatory-livestream/
except using Agent DVR for Linux in place of Contacam (Windows).

Oh well...

Paul

aaronwmorris, 2024-09-18:
Wow... I am not sure how they did that.  :-D  It obviously works, so maybe I lied to you.  I can see they are using the pyCurl camera.  Maybe Contacam has a web server where indi-allsky can download the images.

Very interesting.  Thank you for showing me this.

kippesikgithub, 2024-09-23:
I think I use kind of a same setup. Also using a POE IP camera that is connected to Indi Allsky. Feeding RTSP directly into Indi Allsky is not possible, so you need a system in between to catch the RTSP stream, and let Indi Allsky fetch an image url from there.
Thats exactly how I use it. So RTSP/ONVIF stream are connected to GO2RTC (something I alsready extensively use for Camera security in Home Assistant), and Indi Allsky fetches the image url from the camera (pointed at the sky). Only down-side, you have to configure your camera via the WEB UI. See my github for some more background info, hope this gives you some extra details. https://github.com/kippesikgithub/allsky

khobar95, 2024-09-29:
Thanks very much! I have looked over your info and will see if I can adapt
it to my setup. Currently, I have VLC capturing the stream, but I'm not
sure what to do with it after that. I've only done one capture - last
night. The video properties show 14 hours, but when I open the recorded
video in VLC, it shows only under 2 hours. It still plays beyond that so
I've no idea what's happening. I had hoped to open the recording in a
motion detection program since the camera's motion detection is not so
good, but I'm new to this and will have to do some reading!

Anyway, if I have success I'll post it. In the meantime, clear skies!
Paul

closed 2025-02-20 by aaronwmorris

kippesikgithub, 2025-05-13:
From Go2RTC you can directly livestream your camera feed btw, forgot to mention. I do the same for a birdhouse when there are birds in there.