aaronwmorris/indi-allsky #1518: typo in installation / startup instructions

morrowwm, 2024-09-19:
Just installed this promising software. In `https://github.com/aaronwmorris/indi-allsky/wiki/Getting-Started` I think

`source virtualenv/indi-allsky/bin/activate`

should be

`source ~/indi-allsky/virtualenv/bin/activate`

aaronwmorris, 2024-09-19:
I can add some clarifications, but this assumes that the checkout is at the root of the user's home directory, which it may not be located.  This is the reason for the relative path in the source command.

morrowwm, 2024-09-19:
Yeah, not a big deal.  I'm sure you have higher priorities.

morrowwm, 2024-09-19:
Explanation is sufficient.

closed 2024-09-19 by morrowwm

aaronwmorris, 2024-09-19:
This is not an invalid criticism.  Activating the virtualenv has been a source of confusion.