aaronwmorris/indi-allsky #1520: Cannot Detect svbony sv705c camera

tpaakki, 2024-09-20:
I hope this is a simple issue:

I have used ZWO cameras with the 585 sensor on my indi-allsky just fine, but when I attached my svbony sv705c I could never get a camera to detect.

It is there in the lsusb output, and I even did a rebuild on the 3rd party drivers (BUILD_INDI_CORE=false BUILD_INDI_3RDPARTY=true ./misc/build_indi.sh) but after multiple attempted restarts I cannot seem to detect the camera. The dropdown has two ZWO cameras available.

Is there something I need to do to make the system aware of the svbony?

Thank you
Support output is below

/home/tpaakki/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 7997408 kB

System: Raspberry Pi 4 Model B Rev 1.4

Linux raspberrypi 6.6.47+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.47-1+rpt1 (2024-09-02) aarch64 GNU/Linux

Time
Thu Sep 19 01:51:31 PM HST 2024

System timezone
Pacific/Honolulu

Uptime
 13:51:31 up  2:20,  2 users,  load average: 0.85, 1.53, 2.60

Memory
               total        used        free      shared  buff/cache   available
Mem:         7997408     1840608     3059324      203592     3425972     6156800
Swap:         204796           0      204796

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             3725428        0   3725428   0% /dev
tmpfs             799744     1400    798344   1% /run
/dev/mmcblk0p2 122635096 16080484 100305980  14% /
tmpfs            3998704    82248   3916456   3% /dev/shm
tmpfs               5120       16      5104   1% /run/lock
/dev/mmcblk0p1    522230    77020    445210  15% /boot/firmware
tmpfs             799740       40    799700   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
46251

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
[... 1055 lines cut ...]
    "TSL2591_GAIN_NIGHT": "GAIN_MED",
    "TSL2591_GAIN_DAY": "GAIN_LOW",
    "TSL2591_INT_NIGHT": "INTEGRATIONTIME_100MS",
    "TSL2591_INT_DAY": "INTEGRATIONTIME_100MS",
    "VEML7700_GAIN_NIGHT": "ALS_GAIN_1",
    "VEML7700_GAIN_DAY": "ALS_GAIN_1_8",
    "VEML7700_INT_NIGHT": "ALS_100MS",
    "VEML7700_INT_DAY": "ALS_100MS",
    "SI1145_VIS_GAIN_NIGHT": "GAIN_ADC_CLOCK_DIV_32",
    "SI1145_VIS_GAIN_DAY": "GAIN_ADC_CLOCK_DIV_1",
    "SI1145_IR_GAIN_NIGHT": "GAIN_ADC_CLOCK_DIV_32",
    "SI1145_IR_GAIN_DAY": "GAIN_ADC_CLOCK_DIV_1"
  },
  "CHARTS": {
    "CUSTOM_SLOT_1": 10,
    "CUSTOM_SLOT_2": 11,
    "CUSTOM_SLOT_3": 12,
    "CUSTOM_SLOT_4": 13
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-09-20:
You just need to re-run setup.sh and reconfigure the indi server to use `indi_svbony_ccd`.  A short-cut is to run `./misc/change_camera.sh` which is specific to the camera config.

Once that is done the indiserver service needs to be restarted, but you can just reboot your system.

```
systemctl --user restart indiserver
```

tpaakki, 2024-09-20:
Thank you so much. The short-cut worked perfectly.

closed 2024-09-20 by tpaakki