aaronwmorris/indi-allsky #1521: Problems installing indi-allsky on RPi4 with Pi HQ camera and latest RPi OS. What should I be using?

morrowwm, 2024-09-20:
Anything bleeding edge with what I am trying to do here? The root cause of my problems seems to be that the latest RPi OS has installed with 32-bit userland, even though I chose 64-bit OS.

I have a 4 GB Pi 4 with Pi HQ camera and USB SSD. 

1. What is the recommended version of RPi OS?
2. Should I be using docker?

### Details

Did a fresh install of Raspberry Pi OS using the Raspberry Pi Imager, choosing RASPBERRY PI OS (64-bit) dated 2024-07-04 (debian 12.7) onto an SD card, then copied it to the SSD. It all works OK, except the splash screen says it's 32-bit. (cue ominous music)

Following the [Getting Started](https://github.com/aaronwmorris/indi-allsky/wiki/Getting-Started) instructions went ok, until `./setup.sh` exited, saying I needed to build the INDI libraries with 
`./misc/build_indi.sh`

Tried that, which failed with
```
INDI 3rd party drivers will not build properly on Raspbian 11 32-bit
You must add the following parameter to /boot/config.txt and reboot:

  arm_64bit=0
```
I thought I installed 64-bit Raspbian 12 (`/etc/debian_version ` has `12.7` and `uname -m` says `aarch64`), but as suggested, I put `arm_64bit=0` in `/boot/firmware/config.txt`.

After rebooting, the INDI build succeeded but with some errors that seemed unrelated to my hardware.

Returned to `./setup.sh`, which output:
```
###############################################
### Welcome to the indi-allsky setup script ###
###############################################

Detected a custom installation of INDI in /usr/local/bin

Distribution: raspbian
Release: 12
Arch: aarch64
**Bits: 32**

CPUs: 4
Memory: 3881960 kB

INDI_DRIVER_PATH: /usr/local/bin
INDISERVER_SERVICE_NAME: indiserver
ALLSKY_SERVICE_NAME: indi-allsky
GUNICORN_SERVICE_NAME: gunicorn-indi-allsky
ALLSKY_ETC: /etc/indi-allsky
HTDOCS_FOLDER: /var/www/html/allsky
DB_FOLDER: /var/lib/indi-allsky
DB_FILE: /var/lib/indi-allsky/indi-allsky.sqlite
INSTALL_INDI: false
HTTP_PORT: 80
HTTPS_PORT: 443

Setup proceeding in 10 seconds... (control-c to cancel)

indi-allsky supports the following camera interfaces.

Wiki:  https://github.com/aaronwmorris/indi-allsky/wiki/Camera-Interfaces

     indi: For astro/planetary cameras normally connected via USB (ZWO, QHY, PlayerOne, SVBony, Altair, Touptek, etc)
     libcamera: Supports cameras connected via CSI interface on Raspberry Pi SBCs (Raspi HQ Camera, Camera Module 3, etc)
    pycurl_camera: Download images from a remote web camera
    indi_accumulator: Create synthetic exposures using multiple sub-exposures
     indi_passive: Connect a second instance of indi-allsky to an existing indi-allsky indiserver

1) indi
2) libcamera
3) pycurl_camera
4) indi_accumulator
5) indi_passive
Select a camera interface: 

```
I chose `2) libcamera` and then `1) libcamera_imx477` to match my hardware. 
```
1) libcamera_imx477	      4) libcamera_imx219	   7) libcamera_imx296_gs      10) libcamera_imx327	    13) libcamera_64mp_owlsight
2) libcamera_imx378	      5) libcamera_imx519	   8) libcamera_imx290	       11) libcamera_imx298
3) libcamera_ov5647	      6) libcamera_imx708	   9) libcamera_imx462	       12) libcamera_64mp_hawkeye
Select a libcamera interface: 1
```
The script then quits on this error:
```
Fixing git checkout permissions
**** Installing packages... ****
Hit:1 http://archive.raspberrypi.com/debian bookworm InRelease
Hit:2 http://raspbian.raspberrypi.com/raspbian bookworm InRelease
Reading package lists... Done                              
W: http://raspbian.raspberrypi.com/raspbian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libxslt1-dev' instead of 'libxslt-dev'
build-essential is already the newest version (12.9).
python3 is already the newest version (3.11.2-1).
python3 set to manually installed.
python3-dev is already the newest version (3.11.2-1).
python3-dev set to manually installed.
python3-venv is already the newest version (3.11.2-1).
python3-pip is already the newest version (23.0.1+dfsg-1+rpt1).
cmake is already the newest version (3.25.1-1).
whiptail is already the newest version (0.52.23-1+b1).
whiptail set to manually installed.
bc is already the newest version (1.07.1-3).
bc set to manually installed.
procps is already the newest version (2:4.0.2-3).
cron is already the newest version (3.0pl1-162).
git is already the newest version (1:2.39.5-0+deb12u1).
cpio is already the newest version (2.13+dfsg-7.1).
tzdata is already the newest version (2024a-0+deb12u1).
ca-certificates is already the newest version (20230311).
avahi-daemon is already the newest version (0.8-10).
libopenexr-dev is already the newest version (3.1.5-5).
libgtk-3-0 is already the newest version (1:3.24.38-2~deb12u2+rpt7+rpi1).
libgtk-3-0 set to manually installed.
libcurl4-gnutls-dev is already the newest version (7.88.1-10+rpi1+deb12u7).
libcfitsio-dev is already the newest version (4.2.0-3).
libnova-dev is already the newest version (0.16-5).
libraw-dev is already the newest version (0.20.2-2.1).
libjpeg62-turbo-dev is already the newest version (1:2.1.5-2).
libjpeg62-turbo-dev set to manually installed.
zlib1g-dev is already the newest version (1:1.2.13.dfsg-1).
liblcms2-dev is already the newest version (2.14-2).
liblcms2-dev set to manually installed.
libwebp-dev is already the newest version (1.2.4-0.2+deb12u1).
libwebp-dev set to manually installed.
python3-tk is already the newest version (3.11.2-3).
libxcb1-dev is already the newest version (1.15-1).
libxcb1-dev set to manually installed.
pkg-config is already the newest version (1.8.1-1).
ffmpeg is already the newest version (8:5.1.6-0+deb12u1+rpt1).
libgpiod2 is already the newest version (1.6.3-1+rpi1+b2).
libgpiod2 set to manually installed.
i2c-tools is already the newest version (4.3-2+b2).
i2c-tools set to manually installed.
policykit-1 is already the newest version (122-3).
dbus-user-session is already the newest version (1.14.10-1~deb12u1).
dbus-user-session set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libimath-dev : Conflicts: libilmbase-dev
 libopenexr-dev : Breaks: libilmbase-dev (< 2.5.7-3~)
E: Unable to correct problems, you have held broken packages.

###############
###  ERROR  ###
###############

The setup script exited abnormally, please try to run again...

```
Before going further, I'd like to know if I'm trying something unproven. I have another machine with identical hardware running a slightly older version of RPi OS (debian 12.5) and Thomas Jacquin allsky, which does not have this 64-bit kernel _vs._ 32-bit userland strangeness.

aaronwmorris, 2024-09-20:
There is some package conflict on 32-bit bookworm that I have not been able to track down.  The 64-bit bookworm will work for you.

morrowwm, 2024-09-20:
My mistake was using the rpi-imager. It uses a 32-bit userland version of debian 12.7 by default, even though it advertises 64-bit.  

Fixed this by going to [https://www.raspberrypi.com/software/operating-systems/](https://www.raspberrypi.com/software/operating-systems/) and choosing the 64-bit version, writing to SD card with balena-etcher or some other imager (or choose the file explicitly in rpi-imager), then proceed with Pi OS installation and indi-allsky. 

It all seems to be working. Thank you for this, Aaron!

closed 2024-09-20 by morrowwm