aaronwmorris/indi-allsky #1526: Youtube upload fails on

mrosseel, 2024-09-22:
Configured the youtube upload according to wiki instructions, any idea why I'm getting this error? Haven't changed the title template, which includes {asset_label}. 

```
2024-09-22T20:54:06.902040+02:00 allksy1 [INFO] MainProcess-14627/MainThread allsky._fileUploadWorkerStart() [579]: Starting Upload-4 worker
2024-09-22T20:54:06.901886+02:00 allksy1 [ERROR] MainProcess-14627/MainThread allsky._fileUploadWorkerStart() [572]: Upload worker exception: 
2024-09-22T20:54:06.901772+02:00 allksy1 [ERROR] MainProcess-14627/MainThread allsky._fileUploadWorkerStart() [572]: Upload worker exception: KeyError: 'asset_label'
2024-09-22T20:54:06.901661+02:00 allksy1 [ERROR] MainProcess-14627/MainThread allsky._fileUploadWorkerStart() [572]: Upload worker exception:             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-09-22T20:54:06.901540+02:00 allksy1 [ERROR] MainProcess-14627/MainThread allsky._fileUploadWorkerStart() [572]: Upload worker exception:     title = title_tmpl.format(**template_data)
2024-09-22T20:54:06.901402+02:00 allksy1 [ERROR] MainProcess-14627/MainThread allsky._fileUploadWorkerStart() [572]: Upload worker exception:   File "/home/astropolis/indi-allsky/indi_allsky/filetransfer/youtube_oauth2.py", line 83, in put
2024-09-22T20:54:06.901248+02:00 allksy1 [ERROR] MainProcess-14627/MainThread allsky._fileUploadWorkerStart() [572]: Upload worker exception:                ^^^^^^^^^^^^^^^^^^^^^^^^
2024-09-22T20:54:06.901117+02:00 allksy1 [ERROR] MainProcess-14627/MainThread allsky._fileUploadWorkerStart() [572]: Upload worker exception:     response = client.put(**put_kwargs)
2024-09-22T20:54:06.900999+02:00 allksy1 [ERROR] MainProcess-14627/MainThread allsky._fileUploadWorkerStart() [572]: Upload worker exception:   File "/home/astropolis/indi-allsky/indi_allsky/uploader.py", line 454, in processUpload
2024-09-22T20:54:06.900875+02:00 allksy1 [ERROR] MainProcess-14627/MainThread allsky._fileUploadWorkerStart() [572]: Upload worker exception:     self.processUpload(u_dict)
2024-09-22T20:54:06.900740+02:00 allksy1 [ERROR] MainProcess-14627/MainThread allsky._fileUploadWorkerStart() [572]: Upload worker exception:   File "/home/astropolis/indi-allsky/indi_allsky/uploader.py", line 139, in saferun
2024-09-22T20:54:06.900566+02:00 allksy1 [ERROR] MainProcess-14627/MainThread allsky._fileUploadWorkerStart() [572]: Upload worker exception:     self.saferun()
2024-09-22T20:54:06.900284+02:00 allksy1 [ERROR] MainProcess-14627/MainThread allsky._fileUploadWorkerStart() [572]: Upload worker exception:   File "/home/astropolis/indi-allsky/indi_allsky/uploader.py", line 108, in run
2024-09-22T20:54:06.899843+02:00 allksy1 [ERROR] MainProcess-14627/MainThread allsky._fileUploadWorkerStart() [572]: Upload worker exception: Traceback (most recent call last):
2024-09-22T20:54:04.373464+02:00 allksy1 [INFO] Capture-1-14631/Dummy-1 indi.newMessage() [353]: new Message 2024-09-22T18:54:04: [INFO] Taking a 6.14623 seconds frame... 
2024-09-22T20:54:04.332500+02:00 allksy1 [INFO] Capture-1-14631/MainThread capture.saferun() [544]: Total time since last exposure 15.0444 s
2024-09-22T20:54:04.332157+02:00 allksy1 [INFO] Capture-1-14631/MainThread capture.saferun() [513]: Image queue depth: 0
```

aaronwmorris, 2024-09-22:
Merged #1528 to fix this issue.

mrosseel, 2024-09-23:
confirmed working

closed 2024-09-23 by mrosseel