aaronwmorris/indi-allsky #1534: Database malformed

nicholasmparker, 2024-09-24:
Started getting an issue last night:
Error: stepping, database disk image is malformed (11)

I tried vacuum - did not work.  Any ideas?  I can restore the DB to a backup if needed

aaronwmorris, 2024-09-24:
The database might be toast, but try an integrity check.

```
sqlite3 /var/lib/indi-allsky/indi-allsky.sqlite "PRAGMA integrity_check;"
```

nicholasmparker, 2024-09-24:
It's toast.  🤷

aaronwmorris, 2024-09-24:
The DB image import is probably not working currently, I need to update some of the patterns.

I have this page for rebuilding the database:  https://github.com/aaronwmorris/indi-allsky/wiki/Rebuilding-the-database  if you do not have a recent backup.

aaronwmorris, 2024-09-25:
I reviewed the DB image import functionality and I think it may work in its current state.  Please feel free to try it.

aaronwmorris, 2024-11-04:
Hopefully, the database rebuilt went smoothly.  I am going to close this for now.  Please re-open if there is anything else I can do for you.

aaronwmorris, 2024-11-05:
There were some issues at one point when I had the sqlite database pragma set to disable synchronous mode in a misguided attempt to resolve locked database problems.  Setting `synchronous=off` made things faster, but it was very prone to database corruption if there is a power loss.

This has not been the case for a while and the standard setting now is `synchronous=normal`.  Database corruption has been a very rare issue.

https://www.sqlite.org/pragma.html#pragma_synchronous

nicholasmparker, 2024-11-05:
Got it - i will let you know on this thread if it happens again.  Im
setting up an imx462 version for my brother, so will tinker with it on
sqlite a bit to see if I can reproduce it.

nicholasmparker, 2024-11-10:
Happened again.  I've tried two different SD cards too, so likely not the
issue.  I wonder if its an issue with the power from the raspberry POE hat
or maybe a defective raspi

aaronwmorris, 2024-11-10:
That is very strange.  Did you have any strange error messages in the system log?

nicholasmparker, 2024-11-24:
We can go ahead and close this out.  I had tried out two different cd cards with the same results, but switched and no problems since.  Thank you!

closed 2024-11-24 by nicholasmparker