aaronwmorris/indi-allsky #1541: Problem with Image saving after update to last version.

piratinskii, 2024-09-27:
Problem with Image saving after update to last version.

Hey! I updated indi-allsky to newest version and now I have problems with image saving. Previously, I migrated DB to MariaDB, everything worked fine. Now I can see the image in Focus mode, I don't have any popups with errors, STATUS: Running, but I can't see the latest image (No image for 15 minutes) and also can't see the images on Images page (only images from previous version). I can't find something strange in logs at that moment, but I found some errors in logs in moment, when I launched the system after update for the first time:
```
2024-09-27T14:26:35.244299+03:00 raspberrypi [ERROR] MainProcess-510376 app.log_exception() [838]: Exception on /indi-allsky/ajax/system [POST]#012Traceback (most recent call last):#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 1473, in wsgi_app#012    response = self.full_dispatch_request()#012               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 882, in full_dispatch_request#012    rv = self.handle_user_exception(e)#012         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 880, in full_dispatch_request#012    rv = self.dispatch_request()#012         ^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 865, in dispatch_request#012    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]#012           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask_login/utils.py", line 290, in decorated_view#012    return current_app.ensure_sync(func)(*args, **kwargs)#012           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/views.py", line 110, in view#012    return current_app.ensure_sync(self.dispatch_request)(**kwargs)  # type: ignore[no-any-return]#012           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/indi_allsky/flask/views.py", line 3843, in dispatch_request#012    r = self.startSystemdUnit(app.config['ALLSKY_SERVICE_NAME'])#012        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/indi_allsky/flask/views.py", line 3980, in startSystemdUnit#012    r = manager.StartUnit(unit, 'fail')#012        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/dbus/proxies.py", line 72, in __call__#012    return self._proxy_method(*args, **keywords)#012           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/dbus/proxies.py", line 141, in __call__#012    return self._connection.call_blocking(self._named_service,#012           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/dbus/connection.py", line 634, in call_blocking#012    reply_message = self.send_message_with_reply_and_block(#012                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012dbus.exceptions.DBusException: org.freedesktop.systemd1.TransactionIsDestructive: Transaction for indi-allsky.service/start is destructive (indi-allsky.service has 'stop' job queued, but 'start' is included in transaction).
2024-09-27T14:26:37.932019+03:00 raspberrypi [INFO] MainProcess-510376 views.dispatch_request() [3958]: Command return: /org/freedesktop/systemd1/job/3007447
2024-09-27T14:26:45.848979+03:00 raspberrypi [ERROR] MainProcess-510376 app.log_exception() [838]: Exception on /indi-allsky/ajax/system [POST]#012Traceback (most recent call last):#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 1473, in wsgi_app#012    response = self.full_dispatch_request()#012               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 882, in full_dispatch_request#012    rv = self.handle_user_exception(e)#012         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 880, in full_dispatch_request#012    rv = self.dispatch_request()#012         ^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 865, in dispatch_request#012    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]#012           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask_login/utils.py", line 290, in decorated_view#012    return current_app.ensure_sync(func)(*args, **kwargs)#012           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/views.py", line 110, in view#012    return current_app.ensure_sync(self.dispatch_request)(**kwargs)  # type: ignore[no-any-return]#012           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/indi_allsky/flask/views.py", line 3843, in dispatch_request#012    r = self.startSystemdUnit(app.config['ALLSKY_SERVICE_NAME'])#012        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/indi_allsky/flask/views.py", line 3980, in startSystemdUnit#012    r = manager.StartUnit(unit, 'fail')#012        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/dbus/proxies.py", line 72, in __call__#012    return self._proxy_method(*args, **keywords)#012           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/dbus/proxies.py", line 141, in __call__#012    return self._connection.call_blocking(self._named_service,#012           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/dbus/connection.py", line 634, in call_blocking#012    reply_message = self.send_message_with_reply_and_block(#012                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012dbus.exceptions.DBusException: org.freedesktop.systemd1.TransactionIsDestructive: Transaction for indi-allsky.service/start is destructive (indi-allsky.service has 'stop' job queued, but 'start' is included in transaction).
2024-09-27T14:26:46.907449+03:00 raspberrypi [ERROR] MainProcess-510376 app.log_exception() [838]: Exception on /indi-allsky/ajax/system [POST]#012Traceback (most recent call last):#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 1473, in wsgi_app#012    response = self.full_dispatch_request()#012               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 882, in full_dispatch_request#012    rv = self.handle_user_exception(e)#012         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 880, in full_dispatch_request#012    rv = self.dispatch_request()#012         ^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/app.py", line 865, in dispatch_request#012    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]#012           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^#012  File "/home/pirat/Projects/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask_login/utils.py", line 290, in decorated_view#012    r [... cut ...]
2024-09-27T14:40:53.131957+03:00 raspberrypi [INFO] MainProcess-510376 views.dispatch_request() [3958]: Command return: None
2024-09-27T14:49:27.086220+03:00 raspberrypi gunicorn.indi_allsky.wsgi.error: [1204] Starting gunicorn 22.0.0
2024-09-27T14:49:27.086332+03:00 raspberrypi gunicorn.indi_allsky.wsgi.error: [1204] Listening at: unix:/var/lib/indi-allsky/gunicorn-indi-allsky.sock (1204)
2024-09-27T14:49:27.086435+03:00 raspberrypi gunicorn.indi_allsky.wsgi.error: [1204] Using worker: gthread
2024-09-27T14:49:27.099477+03:00 raspberrypi gunicorn.indi_allsky.wsgi.error: [1212] Booting worker with pid: 1212
```

aaronwmorris, 2024-09-27:
You can ignore those errors.  It appears the service was being started and stopped at the same time, so systemd was complaining.

Check the log at `/var/log/indi-allsky/indi-allsky.log`

piratinskii, 2024-09-27:
> You can ignore those errors. It appears the service was being started and stopped at the same time, so systemd was complaining.
> 
> Check the log at `/var/log/indi-allsky/indi-allsky.log`

Got you. So, this is the log:
```
2024-09-27T19:57:17.008589+03:00 raspberrypi [INFO] Image-1169-17816/MainThread processing.add() [530]: Image bits: 16, cfa: None
2024-09-27T19:57:17.026430+03:00 raspberrypi [INFO] Image-1169-17816/MainThread processing._detectBitDepth() [660]: Image max value: 65504
2024-09-27T19:57:17.027499+03:00 raspberrypi [WARNING] Image-1169-17816/MainThread processing._detectBitDepth() [699]: Updated default bit depth: 16
2024-09-27T19:57:17.036729+03:00 raspberrypi [INFO] Image-1169-17816/MainThread processing._apply_calibration() [751]: Searching for bad pixel map: gain 100, exposure >= 4.6, temp >= -273.1c
2024-09-27T19:57:17.061505+03:00 raspberrypi [WARNING] Image-1169-17816/MainThread processing._apply_calibration() [769]: Temperature matched bad pixel map not found: -273.15c
2024-09-27T19:57:17.074723+03:00 raspberrypi [WARNING] Image-1169-17816/MainThread processing._apply_calibration() [788]: Bad Pixel Map not found: ccd1 16bit 4.6146894s gain 100 bin 1 -273.15c
2024-09-27T19:57:17.074963+03:00 raspberrypi [INFO] Image-1169-17816/MainThread processing._apply_calibration() [800]: Searching for dark frame: gain 100, exposure >= 4.6, temp >= -273.1c
2024-09-27T19:57:17.095675+03:00 raspberrypi [WARNING] Image-1169-17816/MainThread processing._apply_calibration() [818]: Temperature matched dark not found: -273.15c
2024-09-27T19:57:17.108120+03:00 raspberrypi [WARNING] Image-1169-17816/MainThread processing._apply_calibration() [837]: Dark not found: ccd1 16bit 4.6146894s gain 100 bin 1 -273.15c
2024-09-27T19:57:17.108798+03:00 raspberrypi [INFO] Image-1169-17816/MainThread sqm.calculate() [21]: Exposure: 4.614689, gain: 100
2024-09-27T19:57:17.108993+03:00 raspberrypi [INFO] Image-1169-17816/MainThread sqm._generateSqmMask() [48]: Generating mask based on SQM_ROI
2024-09-27T19:57:17.109137+03:00 raspberrypi [WARNING] Image-1169-17816/MainThread sqm._generateSqmMask() [63]: Using central ROI for SQM calculations
2024-09-27T19:57:17.120615+03:00 raspberrypi [INFO] Image-1169-17816/MainThread sqm.calculate() [37]: Raw SQM average: 18334.92
2024-09-27T19:57:17.121341+03:00 raspberrypi [INFO] Image-1169-17816/MainThread sqm.calculate() [42]: Weighted SQM average: 64878.69
2024-09-27T19:57:17.121534+03:00 raspberrypi [ERROR] Image-1169-17816/MainThread processing.debayer() [1069]: No bayer pattern detected
2024-09-27T19:57:17.121670+03:00 raspberrypi [INFO] Image-1169-17816/MainThread image.processImage() [377]: Image: 1920 x 1080
2024-09-27T19:57:17.124354+03:00 raspberrypi [INFO] Image-1169-17816/MainThread processing._generateAduMask() [2673]: Generating mask based on ADU_ROI
2024-09-27T19:57:17.124734+03:00 raspberrypi [WARNING] Image-1169-17816/MainThread processing._generateAduMask() [2688]: Using central ROI for ADU calculations
2024-09-27T19:57:17.135627+03:00 raspberrypi [INFO] Image-1169-17816/MainThread processing._calculate_8bit_adu() [946]: ADU average: 18334.9 (71)
2024-09-27T19:57:17.136009+03:00 raspberrypi [INFO] Image-1169-17816/MainThread processing._convert_16bit_to_8bit() [1147]: Resampling image from 16 to 8 bits
2024-09-27T19:57:17.152540+03:00 raspberrypi [WARNING] Image-1169-17816/MainThread image.recalculate_exposure() [1641]: Found target value for exposure
2024-09-27T19:57:17.152968+03:00 raspberrypi [INFO] Image-1169-17816/MainThread image.write_mask_base_img() [1260]: Generating new mask base
2024-09-27T19:57:18.518548+03:00 raspberrypi [INFO] Capture-1-1204/MainThread capture.shoot() [1522]: Taking 5.02741909 s exposure (gain 100)
2024-09-27T19:57:18.618687+03:00 raspberrypi [INFO] Capture-1-1204/MainThread indi.set_number() [1360]: Setting Duration (s) = 5.027419090270996
2024-09-27T19:57:18.619721+03:00 raspberrypi [INFO] Capture-1-1204/MainThread capture.saferun() [513]: Image queue depth: 62
2024-09-27T19:57:18.619935+03:00 raspberrypi [WARNING] Capture-1-1204/MainThread capture.saferun() [525]: IMAGE QUEUE MAXIMUM EXCEEDED: 62 *** ADDING ADDITIONAL 10.333s DELAY BETWEEN EXPOSURES ***
2024-09-27T19:57:18.626720+03:00 raspberrypi [WARNING] Capture-1-1204/MainThread miscDb.addNotification() [966]: Not adding existing notification
2024-09-27T19:57:18.627300+03:00 raspberrypi [INFO] Capture-1-1204/MainThread capture.saferun() [544]: Total time since last exposure 11.3480 s
2024-09-27T19:57:19.695717+03:00 raspberrypi [INFO] Capture-1-1204/Dummy-1 indi.newMessage() [353]: new Message 2024-09-27T16:57:19: [INFO] Taking a 5.02742 seconds frame... 
2024-09-27T19:57:24.293956+03:00 raspberrypi [INFO] Capture-1-1204/MainThread capture.saferun() [218]: Camera last ready: 9.8s
2024-09-27T19:57:24.294432+03:00 raspberrypi [INFO] Capture-1-1204/MainThread capture.saferun() [219]: Exposure state: BUSY
2024-09-27T19:57:24.294789+03:00 raspberrypi [INFO] Capture-1-1204/MainThread capture.detectNight() [1321]: Sun altitude: -19.3
2024-09-27T19:57:24.295776+03:00 raspberrypi [INFO] Capture-1-1204/MainThread capture.detectNight() [1328]: Moon altitude: -34.9, phase 21.9%
2024-09-27T19:57:24.497450+03:00 raspberrypi [WARNING] Capture-1-1204/MainThread indi.getCcdTemperature() [891]: Camera temperature not supported
[... cut ...]
2024-09-27T19:57:41.317852+03:00 raspberrypi [WARNING] Capture-1-1204/MainThread capture.saferun() [525]: IMAGE QUEUE MAXIMUM EXCEEDED: 63 *** ADDING ADDITIONAL 10.500s DELAY BETWEEN EXPOSURES ***
2024-09-27T19:57:41.329701+03:00 raspberrypi [WARNING] Capture-1-1204/MainThread miscDb.addNotification() [966]: Not adding existing notification
2024-09-27T19:57:41.330052+03:00 raspberrypi [INFO] Capture-1-1204/MainThread capture.saferun() [544]: Total time since last exposure 11.3519 s

```

Still don't have any image :( But now I got some errors about Image worker and exposure problems

aaronwmorris, 2024-09-27:
Is it possible your storage is full?

piratinskii, 2024-09-27:
> Is it possible your storage is full?

Nope, I checked it. 
[image]
And I have script on my raspberry, so, it will clean the system automatically

piratinskii, 2024-09-27:
I think it maybe releated to the previous DB migrate. Maybe system trying to connect to the other DB or something?

aaronwmorris, 2024-09-27:
Can you post the Support Info?

piratinskii, 2024-09-27:
> Can you post the Support Info?

Sorry, how can I do it? :( I can't find it in the settings

piratinskii, 2024-09-27:
> Can you post the Support Info?

Here it is: 
```
/home/pirat/Projects/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 929100 kB

System: Raspberry Pi 3 Model B Plus Rev 1.3

Linux raspberrypi 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

Time
Fri 27 Sep 20:19:47 IDT 2024

System timezone
Asia/Jerusalem

Uptime
 20:19:47 up  5:08,  2 users,  load average: 0.55, 0.49, 0.46

Memory
               total        used        free      shared  buff/cache   available
Mem:          929100      746604      138108      247116      350816      182496
Swap:        2097148      716288     1380860

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev              322452        0    322452   0% /dev
tmpfs              92912     9660     83252  11% /run
/dev/mmcblk0p2 119646512 83997428  29552208  74% /
tmpfs             464548      184    464364   1% /dev/shm
tmpfs               5120       12      5108   1% /run/lock
tmpfs             262144   258088      4056  99% /tmp
/dev/mmcblk0p1    522230    76398    445832  15% /boot/firmware
tmpfs              92908       36     92872   1% /run/user/1000

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
51540

system python: Python 3.11.2

indiserver: /usr/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether b8:27:eb:f6:56:ff brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:a3:03:aa brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.106/24 brd 192.168.0.255 scope global noprefixroute wlan0
[... 1124 lines cut ...]
    "TSL2591_INT_NIGHT": "INTEGRATIONTIME_100MS",
    "TSL2591_INT_DAY": "INTEGRATIONTIME_100MS",
    "VEML7700_GAIN_NIGHT": "ALS_GAIN_1",
    "VEML7700_GAIN_DAY": "ALS_GAIN_1_8",
    "VEML7700_INT_NIGHT": "ALS_100MS",
    "VEML7700_INT_DAY": "ALS_100MS",
    "SI1145_VIS_GAIN_NIGHT": "GAIN_ADC_CLOCK_DIV_32",
    "SI1145_VIS_GAIN_DAY": "GAIN_ADC_CLOCK_DIV_1",
    "SI1145_IR_GAIN_NIGHT": "GAIN_ADC_CLOCK_DIV_32",
    "SI1145_IR_GAIN_DAY": "GAIN_ADC_CLOCK_DIV_1"
  },
  "CHARTS": {
    "CUSTOM_SLOT_1": 10,
    "CUSTOM_SLOT_2": 11,
    "CUSTOM_SLOT_3": 12,
    "CUSTOM_SLOT_4": 13
  }
}

```
#################################
###     end support info      ###
#################################

```

aaronwmorris, 2024-09-27:
You are out of space on the tmpfs for /tmp.  Looks like tmpfs filesystems are excluded by default from the report, I will try to fix this.

```
tmpfs 262144 258088 4056 99% /tmp
```

piratinskii, 2024-09-27:
> You are out of space on the tmpfs for /tmp. Looks like tmpfs filesystems are excluded by default from the report, I will try to fix this.
> 
> ```
> tmpfs 262144 258088 4056 99% /tmp
> ```

Thanks for this, I cleaned temp folder, restarted Indi allsky service, but it still doesn't work, unfortunately 🥲

piratinskii, 2024-09-27:
So. I think i found the error:
```
2024-09-27T22:54:40.397902+03:00 raspberrypi [ERROR] MainProcess-30900/MainThread allsky._startImageWorker() [389]: Image worker exception: sqlalchemy.exc.OperationalError: (pymysql.err.OperationalError) (1906, "The value specified for generated column 'createDate_year' in table 'image' has been ignored")
2024-09-27T22:54:40.398293+03:00 raspberrypi [ERROR] MainProcess-30900/MainThread allsky._startImageWorker() [389]: Image worker exception: [SQL: INSERT INTO image (filename, thumbnail_uuid, remote_url, s3_key, `createDate`, `createDate_year`, `createDate_month`, `createDate_day`, `createDate_hour`, `dayDate`, exposure, exp_elapsed, process_elapsed, gain, binmode, temp, night, adu, stable, moonmode, moonphase, adu_roi, sqm, stars, sync_id, calibrated, detections, kpindex, ovation_max, smoke_rating, data, exclude, width, height, camera_id) VALUES (%(filename)s, %(thumbnail_uuid)s, %(remote_url)s, %(s3_key)s, %(createDate)s, %(createDate_year)s, %(createDate_month)s, %(createDate_day)s, %(createDate_hour)s, %(dayDate)s, %(exposure)s, %(exp_elapsed)s, %(process_elapsed)s, %(gain)s, %(binmode)s, %(temp)s, %(night)s, %(adu)s, %(stable)s, %(moonmode)s, %(moonphase)s, %(adu_roi)s, %(sqm)s, %(stars)s, %(sync_id)s, %(calibrated)s, %(detections)s, %(kpindex)s, %(ovation_max)s, %(smoke_rating)s, %(data)s, %(exclude)s, %(width)s, %(height)s, %(camera_id)s) RETURNING image.id, image.uploaded]
2024-09-27T22:54:40.398983+03:00 raspberrypi [ERROR] MainProcess-30900/MainThread allsky._startImageWorker() [389]: Image worker exception: [parameters: {'filename': 'ccd_e57833c6-9f9f-4009-a49c-4585e42a4d64/exposures/20240927/night/27_22/ccd1_20240927_225318.jpg', 'thumbnail_uuid': None, 'remote_url': None, 's3_key': None, 'createDate': datetime.datetime(2024, 9, 27, 22, 53, 18, 470561), 'createDate_year': 2024, 'createDate_month': 9, 'createDate_day': 27, 'createDate_hour': 22, 'dayDate': datetime.date(2024, 9, 27), 'exposure': 0.375, 'exp_elapsed': 2.5451581478118896, 'process_elapsed': 2.213059186935425, 'gain': 100, 'binmode': 1, 'temp': -273.1499938964844, 'night': 1, 'adu': 13, 'stable': 0, 'moonmode': 0, 'moonphase': 20.899445038562387, 'adu_roi': 0, 'sqm': 6979.4113667794445, 'stars': 14, 'sync_id': None, 'calibrated': 0, 'detections': 0, 'kpindex': 1.0, 'ovation_max': 0, 'smoke_rating': -1, 'data': '{"sensor_temp_0": -273.1499938964844, "sensor_temp_1": 0.0, "sensor_temp_2": 0.0, "sensor_temp_3": 0.0, "sensor_temp_4": 0.0, "sensor_temp_5": 0.0, " ... (1092 characters truncated) ... r_23": 0.0, "sensor_user_24": 0.0, "sensor_user_25": 0.0, "sensor_user_26": 0.0, "sensor_user_27": 0.0, "sensor_user_28": 0.0, "sensor_user_29": 0.0}', 'exclude': 0, 'width': 1920, 'height': 1080, 'camera_id': 1}]
2024-09-27T22:54:40.400121+03:00 raspberrypi [ERROR] MainProcess-30900/MainThread allsky._startImageWorker() [389]: Image worker exception: (Background on this error at: https://sqlalche.me/e/20/e3q8)
2024-09-27T22:54:40.400564+03:00 raspberrypi [ERROR] MainProcess-30900/MainThread allsky._startImageWorker() [389]: Image worker exception: 
2024-09-27T22:54:40.400872+03:00 raspberrypi [INFO] MainProcess-30900/MainThread allsky._startImageWorker() [396]: Starting Image-9 worker
2024-09-27T22:54:40.401172+03:00 raspberrypi [WARNING] MainProcess-30900/MainThread processing._load_detection_mask() [2583]: No detection mask defined
2024-09-27T22:54:40.416360+03:00 raspberrypi [INFO] MainProcess-30900/MainThread allsky._queueManualTasks() [1223]: Checking for manually submitted tasks
2024-09-27T22:54:41.540361+03:00 raspberrypi [INFO] Capture-1-30907/MainThread capture.saferun() [427]: Exposure received in 4.4341 s (2.2707)
2024-09-27T22:54:41.777929+03:00 raspberrypi [INFO] Image-9-31108/MainThread processing.add() [530]: Image bits: 16, cfa: None
2024-09-27T22:54:41.792054+03:00 raspberrypi [INFO] Capture-1-30907/MainThread capture.shoot() [1522]: Taking 2.16346145 s exposure (gain 100)
2024-09-27T22:54:41.796313+03:00 raspberrypi [INFO] Image-9-31108/MainThread processing._detectBitDepth() [660]: Image max value: 27184
2024-09-27T22:54:41.797362+03:00 raspberrypi [WARNING] Image-9-31108/MainThread processing._detectBitDepth() [699]: Updated default bit depth: 15
2024-09-27T22:54:41.806808+03:00 raspberrypi [INFO] Image-9-31108/MainThread processing._apply_calibration() [751]: Searching for bad pixel map: gain 100, exposure >= 0.4, temp >= -273.1c
2024-09-27T22:54:41.829710+03:00 raspberrypi [WARNING] Image-9-31108/MainThread processing._apply_calibration() [769]: Temperature matched bad pixel map not found: -273.15c
2024-09-27T22:54:41.841964+03:00 raspberrypi [WARNING] Image-9-31108/MainThread processing._apply_calibration() [788]: Bad Pixel Map not found: ccd1 16bit 0.3750000s gain 100 bin 1 -273.15c
2024-09-27T22:54:41.842238+03:00 raspberrypi [INFO] Image-9-31108/MainThread processing._apply_calibration() [800]: Searching for dark frame: gain 100, exposure >= 0.4, temp >= -273.1c
2024-09-27T22:54:41.862441+03:00 raspberrypi [WARNING] Image-9-31108/MainThread processing._apply_calibration() [818]: Temperature matched dark not found: -273.15c
2024-09-27T22:54:41.874647+03:00 raspberrypi [WARNING] Image-9-31108/MainThread processing._apply_calibration() [837]: Dark not found: ccd1 16bit 0.3750000s gain 100 bin 1 -273.15c
2024-09-27T22:54:41.875622+03:00 raspberrypi [INFO] Image-9-31108/MainThread sqm.calculate() [21]: Exposure: 0.375000, gain: 100
[... cut ...]
2024-09-27T22:54:51.448742+03:00 raspberrypi [INFO] Capture-1-30907/MainThread capture.saferun() [513]: Image queue depth: 24
2024-09-27T22:54:51.449023+03:00 raspberrypi [WARNING] Capture-1-30907/MainThread capture.saferun() [525]: IMAGE QUEUE MAXIMUM EXCEEDED: 24 *** ADDING ADDITIONAL 4.000s DELAY BETWEEN EXPOSURES ***
2024-09-27T22:54:51.455445+03:00 raspberrypi [WARNING] Capture-1-30907/MainThread miscDb.addNotification() [966]: Not adding existing notification
2024-09-27T22:54:51.455832+03:00 raspberrypi [INFO] Capture-1-30907/MainThread capture.saferun() [544]: Total time since last exposure 4.8713 s```

aaronwmorris, 2024-09-27:
Is it possible you manually updated the mysql schema?  It is mentioning a generated column and there are no generated columns.  Is it possible you added the stuff from here:  https://github.com/aaronwmorris/indi-allsky/wiki/Image-viewer-DB-performance   ?

piratinskii, 2024-09-27:
> Is it possible you manually updated the mysql schema? It is mentioning a generated column and there are no generated columns. Is it possible you added the stuff from here: https://github.com/aaronwmorris/indi-allsky/wiki/Image-viewer-DB-performance ?

Nope, I didn't. I just previously did db migration to MariaDB by this manual: https://github.com/aaronwmorris/indi-allsky/wiki/Database-conversion in previous version of indi-allsky. Then I just updated to the last one and it's all

piratinskii, 2024-09-27:
> Is it possible you manually updated the mysql schema? It is mentioning a generated column and there are no generated columns. Is it possible you added the stuff from here: https://github.com/aaronwmorris/indi-allsky/wiki/Image-viewer-DB-performance ?

Yup, I think you're right. I'm not sure, that I did exactly this stuff, but I did something from your manuals to improve image loading. So, i think it was this stuff. How can I correct it know to work on this version ? :(

aaronwmorris, 2024-09-27:
Try the following statements.

```
DROP INDEX idx_video_dayDate_Ym on video;
DROP INDEX idx_image_createDate_Ymdh on image;

ALTER TABLE image DROP createDate_year;
ALTER TABLE image DROP createDate_month;
ALTER TABLE image DROP createDate_day;
ALTER TABLE image DROP createDate_hour;

ALTER TABLE video DROP dayDate_year;
ALTER TABLE video DROP dayDate_month;
```

Once that is done, re-run setup.sh and let it go through the database migrations.

piratinskii, 2024-09-27:
> ALTER TABLE image DROP createDate_year;

Oh, thank you very much for your help! Everything works great now!

closed 2024-09-27 by piratinskii