aaronwmorris/indi-allsky #1544: Virtualenv not working

WirthmU, 2024-09-29:
I am running indi-allksky on a new installation on a RPI5. When I try to tests the sensors I have on my system I enter the following command as in the wiki:
`source virtualenv/indi-allsky/bin/activate`

Then what I get is:
`-bash: virtualenv/indi-allsky/bin/activate: No such file or directory`

aaronwmorris, 2024-09-29:
Are you in the indi-allsky git checkout folder?

WirthmU, 2024-09-29:
I did just a regular installation as per the allsky-wiki. The webserver is up and running. I did an ssh login into the RPI as usual and entered the command:
```
as3@raspberrypi:~ $ source virtualenv/indi-allsky/bin/activate
-bash: virtualenv/indi-allsky/bin/activate: No such file or directory
```

aaronwmorris, 2024-09-29:
You just need to be inside the indi-allsky folder before activating the virtualenv

WirthmU, 2024-09-29:
That was it - it's working now.
Thank you!

closed 2024-10-07 by aaronwmorris