aaronwmorris/indi-allsky #1549: Lots of errors and status 'DOWN'

aldumil, 2024-10-02:
I would appreciate any support!

Installed fresh on a rpi 5 with a ASA676 camera plugged in (fresh OS and installed a couple times just to make sure).

On the web UI the status shows as 'STARTING' for awhile and I see errors like:

` allsky [ERROR] MainProcess-24826/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: 
 allsky [ERROR] MainProcess-24826/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: indi_allsky.exceptions.TimeOutException: Timeout finding control CCD_EXPOSURE
 allsky [ERROR] MainProcess-24826/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:     raise TimeOutException('Timeout finding control {0}'.format(name))
 allsky [ERROR] MainProcess-24826/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:   File "/home/aldumil/indi-allsky/indi_allsky/camera/indi.py", line 1344, in get_control
 allsky [ERROR] MainProcess-24826/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 allsky [ERROR] MainProcess-24826/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:     ctl_CCD_EXPOSURE = self.get_control(self.ccd_device, 'CCD_EXPOSURE', 'number')
 allsky [ERROR] MainProcess-24826/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:   File "/home/aldumil/indi-allsky/indi_allsky/camera/indi.py", line 534, in getCcdInfo
 allsky [ERROR] MainProcess-24826/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 allsky [ERROR] MainProcess-24826/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:     ccd_info = self.indiclient.getCcdInfo()
 allsky [ERROR] MainProcess-24826/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:   File "/home/aldumil/indi-allsky/indi_allsky/capture.py", line 709, in _initialize
 allsky [ERROR] MainProcess-24826/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:     self._initialize()
 allsky [ERROR] MainProcess-24826/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:   File "/home/aldumil/indi-allsky/indi_allsky/capture.py", line 187, in saferun
 allsky [ERROR] MainProcess-24826/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:     self.saferun()
 allsky [ERROR] MainProcess-24826/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:   File "/home/aldumil/indi-allsky/indi_allsky/capture.py", line 175, in run`

I believe it is the same errors repeating, and then eventually the status changes to 'DOWN'.

I am happy to provide whatever might be helpful to the wonderful person who is willing to point me in the right direction.

aaronwmorris, 2024-10-02:
Can you post the Support Info text?

aldumil, 2024-10-02:
/home/aldumil/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 8241552 kB

System: Raspberry Pi 5 Model B Rev 1.0

Linux allsky 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

Time
Wed  2 Oct 17:47:34 EDT 2024

System timezone
US/Eastern

Uptime
 17:47:34 up  2:47,  2 users,  load average: 0.02, 0.03, 0.00

Memory
               total        used        free      shared  buff/cache   available
Mem:         8241552      999504      553808       22608     6810048     7242048
Swap:         204784           0      204784

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             3945824        0   3945824   0% /dev
tmpfs             824160     6016    818144   1% /run
/dev/mmcblk0p2  60801232 10116288  47576604  18% /
tmpfs            4120768      976   4119792   1% /dev/shm
tmpfs               5120       48      5072   1% /run/lock
/dev/mmcblk0p1    522230    76398    445832  15% /boot/firmware
tmpfs             824144      160    823984   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
47400

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether 2c:cf:67:5e:18:f8 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:5e:18:f9 brd ff:ff:ff:ff:ff:ff
    inet 192.168.54.95/24 brd 192.168.54.255 scope global dynamic noprefixroute wlan0
       valid_lft 76343sec preferred_lft 76343sec
    inet6 2601:155:87:80a0:3bca:5397:22dc:c199/64 scope global dynamic noprefixroute 
       valid_lft 86379sec preferred_lft 86379sec
    inet6 fe80::d960:fcd4:a273:811c/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
[... 1126 lines cut ...]
    "TSL2591_INT_NIGHT": "INTEGRATIONTIME_100MS",
    "TSL2591_INT_DAY": "INTEGRATIONTIME_100MS",
    "VEML7700_GAIN_NIGHT": "ALS_GAIN_1",
    "VEML7700_GAIN_DAY": "ALS_GAIN_1_8",
    "VEML7700_INT_NIGHT": "ALS_100MS",
    "VEML7700_INT_DAY": "ALS_100MS",
    "SI1145_VIS_GAIN_NIGHT": "GAIN_ADC_CLOCK_DIV_32",
    "SI1145_VIS_GAIN_DAY": "GAIN_ADC_CLOCK_DIV_1",
    "SI1145_IR_GAIN_NIGHT": "GAIN_ADC_CLOCK_DIV_32",
    "SI1145_IR_GAIN_DAY": "GAIN_ADC_CLOCK_DIV_1",
    "LTR390_GAIN_NIGHT": "GAIN_9X",
    "LTR390_GAIN_DAY": "GAIN_1X"
  },
  "CHARTS": {
    "CUSTOM_SLOT_1": 10,
    "CUSTOM_SLOT_2": 11,
    "CUSTOM_SLOT_3": 12,
    "CUSTOM_SLOT_4": 13
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-10-02:
This is just a guess, but you might have a bad USB cable.  That is the most common causes of random disconnects such as this.

You may also try just using a different USB port on the Raspi.

aldumil, 2024-10-02:
> This is just a guess, but you might have a bad USB cable. That is the most common causes of random disconnects such as this.
> 
> You may also try just using a different USB port on the Raspi.

Wow. You think you are a handy person who has picked up some knowledge, and then something like this happens.

Different USB port on the PI, everything runs perfect. You are a hero Aaron!

aaronwmorris, 2024-10-03:
Fantastic!  Glad to help.

closed 2024-10-13 by aaronwmorris