aaronwmorris/indi-allsky #1553: Gallery not showing images for '0' (midnight-1am)

zaphus, 2024-10-05:
When viewing the gallery, each of the hours has the appropriate images - but the '0' hour does not appear to be loading any images, you cannot find things between 00:00 and 00:59

The server has the files, in the correct directory, it just seems to be the gallery not showing them.  This applies to both the Gallery and Images views.  I am running indi_v2024.09.1 up to and including pull request #1550 (earlier today, as of when I am writing this). 
In the image view, the list of images does not update - it continues to show whatever was loaded before you switched to the zero hour.

I only know this because someone reported a meteor last night at 12:17am and I realised I cannot view the images in either of these views.  Sadly the meteor is nowhere to be seen in the actual images.

aaronwmorris, 2024-10-05:
It appears the 0 hour is broken for both the Gallery and Image viewers.  I think I know what is going on.

aaronwmorris, 2024-10-07:
Merged #1561 to fix the views when hour is 0.

zaphus, 2024-10-08:
Confirmed Fixed

[image]

closed 2024-10-08 by zaphus