aaronwmorris/indi-allsky #1554: Very bad image on RPI5 with RP HQ IMX477 camera

babsenfred, 2024-10-05:
Hello, switched from Allsky where I had a good image quality. Did a fresh install of Allsky-INDI and have a really bad image.
I tried to set the Bayer Pattern to BGGR and SCNR to max, still no good.

Any recommendations on the settings please?

Thanks.

[image]

aaronwmorris, 2024-10-05:
This is the challenge of dealing with RAW data.  RAW data has no white balance, so you get the green bias due to the bayer matrix having 2 green pixels for every 1 red and blue.

Personally, I like the SCNR Average Neutral better than the Maximum Neutral.

You can enable the AWB settings with libcamera, which should give you results similar (or the same) as TJ's allsky.

babsenfred, 2024-10-06:
Hello Aaron, I changed the settings and the picture looks better. Still it looks dark.
The first picture is from my Allsky-INDI camera. The second one I took with my phone at the same place and time. This only to show the difference in brightness.

[image]

[image]

aaronwmorris, 2024-10-06:
Can you post your data from Support Info?

jschmied, 2024-11-15:
I had same problem and compiled a cheet sheet:

Tint Problem | Correction Needed
Too Red (Reddish tint) | Decrease Red Balance, Increase Green and/or Blue Balance
Too Blue (Bluish tint) | Decrease Blue Balance, Increase Green and/or Red Balance
Too Green (Greenish tint) | Decrease Green Balance, Increase Red and/or Blue Balance
Too Yellow (Yellowish tint) | Decrease Red and Green Balance, Increase Blue Balance
Too Magenta (Pinkish tint) | Decrease Red and Blue Balance, Increase Green Balance
Too Cyan (Cyanish tint) | Decrease Green and Blue Balance, Increase Red Balance

I use now:
Bayer Patter: RGGB
SCNR (green reduction): Average Neutral
R/G/B 0.7/1.0/1.4

I checked this by putting into processing.py:1383

        # Log the average values of B, G, R channels
        logger.info(f"B channel average: {b_avg}")
        logger.info(f"G channel average: {g_avg}")
        logger.info(f"R channel average: {r_avg}")

and I get values around 50 for all channels what means the auto-wb is not changing the picture anymore. The picture still does not look natural but I assume we are all used to cameras tuning the picture to please the eye.

haightjr, 2024-11-22:
I have been battling the same issue with a RPi 4 4GB and RPi HQ Camera.  I've used Allsky for a year now and trying to switch over because of the added features Indi-Allsky offers.   Unfortunately I have been camera down for a week now.   If I attempt to use raw mode Indi-Allsky errors out in the logs.  With jpeg, my photos come out hard green and I cannot seem to get the color settings correct.  I've attempted this after 3 wipes and installs.  

What configuration is different from AllSky and Indi-Allsky to have these IMX477 sensor issues?   Thanks. 

Error when attempting RAW image mode. 
`024-11-20T19:50:37.002536-06:00 allsky [WARNING] MainProcess-4766/MainThread processing._load_detection_mask() [2791]: No detection mask defined
2024-11-20T19:50:36.997097-06:00 allsky [INFO] MainProcess-4766/MainThread allsky._startImageWorker() [396]: Starting Image-8 worker
2024-11-20T19:50:36.996888-06:00 allsky [ERROR] MainProcess-4766/MainThread allsky._startImageWorker() [389]: Image worker exception: 
2024-11-20T19:50:36.996686-06:00 allsky [ERROR] MainProcess-4766/MainThread allsky._startImageWorker() [389]: Image worker exception: Exception: *** rawpy module not available ***
2024-11-20T19:50:36.996463-06:00 allsky [ERROR] MainProcess-4766/MainThread allsky._startImageWorker() [389]: Image worker exception:     raise Exception('*** rawpy module not available ***')
2024-11-20T19:50:36.996189-06:00 allsky [ERROR] MainProcess-4766/MainThread allsky._startImageWorker() [389]: Image worker exception:   File "/home/justin/indi-allsky/indi_allsky/processing.py", line 447, in add
2024-11-20T19:50:36.995982-06:00 allsky [ERROR] MainProcess-4766/MainThread allsky._startImageWorker() [389]: Image worker exception:             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-11-20T19:50:36.995766-06:00 allsky [ERROR] MainProcess-4766/MainThread allsky._startImageWorker() [389]: Image worker exception:     i_ref = self.image_processor.add(filename_p, exposure, exp_date, exp_elapsed, camera)
2024-11-20T19:50:36.995445-06:00 allsky [ERROR] MainProcess-4766/MainThread allsky._startImageWorker() [389]: Image worker exception:   File "/home/justin/indi-allsky/indi_allsky/image.py", line 357, in processImage
2024-11-20T19:50:36.990439-06:00 allsky [ERROR] MainProcess-4766/MainThread allsky._startImageWorker() [389]: Image worker exception:     self.processImage(i_dict)
2024-11-20T19:50:36.987026-06:00 allsky [ERROR] MainProcess-4766/MainThread allsky._startImageWorker() [389]: Image worker exception:   File "/home/justin/indi-allsky/indi_allsky/image.py", line 257, in saferun
2024-11-20T19:50:36.986618-06:00 allsky [ERROR] MainProcess-4766/MainThread allsky._startImageWorker() [389]: Image worker exception:     self.saferun()
2024-11-20T19:50:36.980605-06:00 allsky [ERROR] MainProcess-4766/MainThread allsky._startImageWorker() [389]: Image worker exception:   File "/home/justin/indi-allsky/indi_allsky/image.py", line 228, in run
2024-11-20T19:50:36.967095-06:00 allsky [ERROR] MainProcess-4766/MainThread allsky._startImageWorker() [389]: Image worker exception: Traceback (most recent call last):
2024-11-20T19:50:31.054505-06:00 allsky [WARNING] Image-7-4923/MainThread processing.populateSatelliteData() [1859]: Satellite data not found: CSS (TIANHE)
2024-11-20T19:50:31.052786-06:00 allsky [WARNING] Image-7-4923/MainThread processing.populateSatelliteData() [1859]: Satellite data not found: HST
2024-11-20T19:50:31.051346-06:00 allsky [WARNING] Image-7-4923/MainThread processing.populateSatelliteData() [1859]: Satellite data not found: ISS (ZARYA)
2024-11-20T19:50:31.012382-06:00 allsky [INFO] Capture-1-4781/MainThread capture.saferun() [464]: Exposure received in 2.4580 s (2.4480)
2024-11-20T19:50:28.555636-06:00 allsky [INFO] Capture-1-4781/MainThread capture.saferun() [581]: Total time since last exposure 15.0454 s
2024-11-20T19:50:28.555434-06:00 allsky [INFO] Capture-1-4781/MainThread capture.saferun() [550]: Image queue depth: 0
2024-11-20T19:50:28.554274-06:00 allsky [INFO] Capture-1-4781/MainThread libcamera.setCcdExposure() [263]: image command: rpicam-still --immediate --nopreview --camera 0 --raw --denoise off --gain 1 --shutter 9999 --metadata /tmp/tmpvq1zq04e.json --metadata-format json --awbgains 1,1 --output /tmp/tmpuiz_xp0o.dng
2024-11-20T19:50:28.553661-06:00 allsky [INFO] Capture-1-4781/MainThread capture.shoot() [1576]: Taking 0.01000000 s exposure (gain 1)
2024-11-20T19:50:27.301336-06:00 allsky [INFO] Capture-1-4781/MainThread indi.getTelescopeRaDec() [881]: Telescope Coord: RA 12.57, Dec 44.30
2024-11-20T19:50:27.209290-06:00 allsky [INFO] Capture-1-4781/MainThread capture.detectNight() [1382]: Moon altitude: 37.3, phase 69.3%
2024-11-20T19:50:27.209188-06:00 allsky [INFO] Capture-1-4781/MainThread capture.detectNight() [1375]: Sun altitude: 11.9
2024-11-20T19:50:27.200901-06:00 allsky [INFO] Capture-1-4781/MainThread capture.saferun() [232]: Exposure state: READY
2024-11-20T19:50:27.200609-06:00 allsky [INFO] Capture-1-4781/MainThread capture.saferun() [231]: Camera last ready: 11.4s
2024-11-20T19:50:23.944872-06:00 allsky [INFO] MainProcess-4766/MainThread allsky._queueManualTasks() [1223]: Checking for manually submitted tasks`

aaronwmorris, 2024-11-22:
Update to the latest code and re-run `setup.sh`  I just corrected an issue today with the python module requirements to ensure the rawpy module is installed.

haightjr, 2024-11-22:
@aaronwmorris Thanks for the fix.  I can confirm that fixed the issue when using the RAW settings with no errors in the logs.  I am still tuning the image balance factors to get a more natural tone without applying white balance.  If I get it sorted out I will share my image settings.

haightjr, 2024-11-23:
Now I am seeing another issue.  Night and moon mode are both showing enabled in the web interface sidebar and overlay but its 12pm.  I've verified the time/location settings in the os and Indi-Allsky are set correctly.  I am still making adjustments to Red/Green/Blue to fine tune the images. 

``justin@allsky:~ $ timedatectl
               Local time: Sat 2024-11-23 12:26:06 CST
           Universal time: Sat 2024-11-23 18:26:06 UTC
                 RTC time: n/a
                Time zone: America/Chicago (CST, -0600)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no
[image]
[image]

aaronwmorris, 2024-11-23:
Based on your timezone, I am going to assume you are in the western hemisphere.  I think the problem is you have your longitude set for the eastern hemisphere.  Your longitude should probably be negative.

haightjr, 2024-11-23:
That was it.   A minor oversight and forgot to add the -.   I've been at this tuning the images from the IMX477 for so long now.  I'm about to throw in the towel and replace the cam with a ASI676MC.

aaronwmorris, 2024-11-23:
One thing you can do with the IMX477 is enable the libcamera AWB and look to see the values that is calculates for Red and Blue in the log (it only adjusts red and blue).  It can give you a much better starting point.

I may end up having to create separate R, G, and B gain values for day and night for situations like this.

jschmied, 2024-11-23:
A problem might be, if I get a bright spot well balanced (sky) darker spots like clouds are tinted. It's visible in the histogram. This camera might need gamma correction per channel (or a better AWB algorithm). I might have a look at the code but it can take some time since I'm busy with other stuff. the libcamera AWB was not so helpful, AWB on second config tab was good together with SCNR.

[image]

jschmied, 2024-11-25:
It got better with following option:

RPi4:
--tuning-file /usr/share/libcamera/ipa/rpi/vc4/imx477_scientific.json

PRi5:
--tuning-file /usr/share/libcamera/ipa/rpi/pisp/imx477_scientific.json

see: https://forums.raspberrypi.com/viewtopic.php?t=343449

aaronwmorris, 2024-11-25:
Interesting.  I have seen the "scientific" tuning file in the folder, but I had no idea the history.  Very cool.

haightjr, 2024-12-10:
Update:  I gave up on the RPi HQ camera and installed a ASI676MC.  Its fantastic.

aaronwmorris, 2024-12-10:
Astronomy cameras do tend have much better quality.

closed 2025-02-20 by aaronwmorris