aaronwmorris/indi-allsky #1562: uploading videos have no thumbnail

screetch82, 2024-10-07:
Hi

the uploadfoldes for keogram, startrails also contain the thumbnails. the Video one does not.

[image]

[image]

here:
[image]

here is my conig:
[image]

Any ideas what I missed? Thanks.

screetch82, 2024-10-07:
e.g. this how the keograms are shown on the website:
[image]

but for the videos it just shows:
[image]

Nevertheless the shown filename also isnt correct. Should this not be a date or proper file nime?

aaronwmorris, 2024-10-08:
I will have to put this in my backlog.