aaronwmorris/indi-allsky #1567: Home Assistant Device Grouping

robbrad, 2024-10-11:
Hi @aaronwmorris  - firstly amazing project here !

I want to point out if you group the HA discovery under a subtopic it will show as a device

[image]

See the difference in the image for astrolive otherwise it displays like this in the front end of HA 

[image]

Completely up to you but it helps if the user wants to "Add a whole device to the front end"

aaronwmorris, 2024-10-11:
Interesting, I will have to play with this option.

aaronwmorris, 2024-10-12:
All indi-allsky entities are now placed under a unique discovery sub-topic.  This alone was not enough to group them as part of a single device.  I also had to add a `device` attribute for Home Assistant.

Merged #1569

robbrad, 2024-10-12:
Thanks @aaronwmorris for your perseverance on this

robbrad, 2024-10-12:
Works Great !
[image]

[image]

aaronwmorris, 2024-10-12:
Fantastic!  This has been on my to-do list for a while.  Glad it is working out for you.

closed 2024-10-12 by robbrad