aaronwmorris/indi-allsky #1576: Enhancement:  make a version of latest.jpg available without text

zaphus, 2024-10-17:
Context:  I am working on a machine-learning based cloud detector (based on a fork of: https://github.com/gordtulloch/mlCloudDetect )
I am pulling down the latest image using the web link to latest.jpg, which works great.  However I want to use a clean image without the text overlay so that I can be sure the AI is not training on any of that text (eg if all my cloud samples happen to be in October, it might use the date text as part of what it considers important in the training.
If there was a 'latest_notext.jpg' shared to the same location of latest.jpg then that would be great. I dont need indi-allsky to save all the images, just serve up the latest one.

I know it's a 1-person use case, but could also be handy for anyone else looking at AI based image processing.

aaronwmorris, 2024-10-17:
I might have functionality you can already use for this.  You can enable RAW image exports on the Config -> Image tab.  "RAW" in this context just means the unaltered/unprocessed image.  When you do this and you view images on the Image Viewer, you will see an orange/yellow link (bottom) to the RAW image.

The second part, I already have the concept of redirect views, but I had forgot the redirect view for RAW images.

The redirect views will issue a 302 redirect to the latest image of the type requested.  https://github.com/aaronwmorris/indi-allsky/wiki/Redirect-Views

Once you upgrade, you will be able to issue a request to `https://hostname.local/indi-allsky/latestraw` and you will be redirected to the latest RAW image.  A curl example is included to script the download of the latest image.

Merged #1578 to add the RAW redirect views.

aaronwmorris, 2024-10-17:
The alternative here is I could update my version of mlCloudDetect that is in my `testing/` folder.  I had modified Gord's script to integrate directly into the indi-allsky database to read the latest file.

zaphus, 2024-10-17:
Thanks - I'm not sure if I'll be running the script on the same device, or separately, which is why I was using the web access.  I can use that raw redirect for now, but does enabling raw exports that cause the number of images stored to double?  or is it only used for the export and doesnt keep a full set of night+day raws

zaphus, 2024-10-17:
FYI - I've grabbed the latest code and am using the redirected ImageRaw which works great - thanks.
I would prefer not to keep all the raw images as well, but I have enough space so not really an issue

closed 2024-10-17 by zaphus

aaronwmorris, 2024-10-17:
Merged #1579 which has separate expiration settings for RAW (and FITS) images.  You can now set a 1 day expiration for RAW images.