aaronwmorris/indi-allsky #1581: Update ./setup.sh Error

fly800, 2024-10-19:
I tried doing an update and get an error on ./setup.sh.  Unfortunately, I uninstalled indi-allsky and that didn't resolve the error.

Terminal :

admin@raspberrypi:~/indi-allsky $ ./setup.sh
###############################################
### Welcome to the indi-allsky setup script ###
###############################################

Detected a custom installation of INDI in /usr/local/bin

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 3882992 kB

INDI_DRIVER_PATH: /usr/local/bin
INDISERVER_SERVICE_NAME: indiserver
ALLSKY_SERVICE_NAME: indi-allsky
GUNICORN_SERVICE_NAME: gunicorn-indi-allsky
ALLSKY_ETC: /etc/indi-allsky
HTDOCS_FOLDER: /var/www/html/allsky
DB_FOLDER: /var/lib/indi-allsky
DB_FILE: /var/lib/indi-allsky/indi-allsky.sqlite
INSTALL_INDI: false
HTTP_PORT: 80
HTTPS_PORT: 443

Setup proceeding in 10 seconds... (control-c to cancel)

indi-allsky supports the following camera interfaces.

Wiki:  https://github.com/aaronwmorris/indi-allsky/wiki/Camera-Interfaces

             indi: For astro/planetary cameras normally connected via USB (ZWO, QHY, PlayerOne, SVBony, Altair, Touptek, etc)
        libcamera: Supports cameras connected via CSI interface on Raspberry Pi SBCs (Raspi HQ Camera, Camera Module 3, etc)
    pycurl_camera: Download images from a remote web camera
 indi_accumulator: Create synthetic exposures using multiple sub-exposures
     indi_passive: Connect a second instance of indi-allsky to an existing indi-allsky indiserver

1) indi              3) pycurl_camera     5) indi_passive
2) libcamera         4) indi_accumulator
Select a camera interface: 1

Fixing git checkout permissions
**** Installing packages... ****
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease
Ign:5 https://ppa.launchpadcontent.net/deadsnakes/ppa/ubuntu bookworm InRelease
Err:6 https://ppa.launchpadcontent.net/deadsnakes/ppa/ubuntu bookworm Release
  404  Not Found [IP: 185.125.190.80 443]
Reading package lists... Done
E: The repository 'https://ppa.launchpadcontent.net/deadsnakes/ppa/ubuntu bookworm Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

###############
###  ERROR  ###
###############

The setup script exited abnormally, please try to run again...

I hope you can help.

aaronwmorris, 2024-10-19:
It looks like the deadsnakes Python PPA was setup on this system prior to indi-allsky being installed.  Ubuntu PPAs are not usually compatible with Debian, so I would recommend removing that before proceeding.

You should be able to find the apt config in `/etc/apt/sources.list.d/`  If you remove the PPA config, the install should run normally.

fly800, 2024-10-19:
You're a godsend.  Back up and running.  Learnt another tidbit about Linux.  Thanks.