aaronwmorris/indi-allsky #1596: Show full quality image in 'Latest' tab

roop90, 2024-10-27:
Hey there,

I've noticed that the image shown on the 'Latest' tab looks way worse than on the 'Gallery' or 'Image' tab. Would it be possible to show the full image in the 'Latest' tab instead or provide an setting for this?

aaronwmorris, 2024-10-27:
The image quality is normally only an issue when viewing the images on a low resolution screen.  The image is resized to fit within the canvas and the algorithm used by the canvas does not work very well.  What device are you viewing it with?

roop90, 2024-10-27:
I use 4 different devices and all of them show bad image quality. Even when in full screen.
It mostly affects the readability of exposure settings and so on. Otherwise it's not an issue.

My Devices:
- 6,67" Smartphone with 1440x3120p Display
- 11" Tablet with 2560x1600p Display
- 14" Laptop with 1440p Display
- Desktop PC with 4k Display

Both PCs run Windows with dpi scaling.

aaronwmorris, 2024-10-28:
Would you be willing/capable of some minor javascript updates to test settings in the display code?

roop90, 2024-10-29:
> Would you be willing/capable of some minor javascript updates to test settings in the display code?

Sure. Can you provide an branch name? Then I can pull it an rebuild the container.

aaronwmorris, 2024-10-29:
There is now a branch named `canvas_quality`.  Before switching to that branch, you need to upgrade to the latest code.  I made some changes to the database schema that need in place for this branch.

roop90, 2024-11-05:
It took a while, sorry.

I switched a few times between the versions.
On mobile it got a little better, but still not good readable.

aaronwmorris, 2025-04-25:
I have merged #2095 which starts the process for me converting the canvas elements to regular img elements in indi-allsky.  For the time being, the Latest Image and Loop views have been converted to img based displays.