aaronwmorris/indi-allsky #1597: Camera Recommendation

mshetzer, 2024-10-28:
Love the app, and it's running perfectly.  I've been using an old ASI224MC, but I really enjoy have the allsky view up at night and would like to improve the image quality.  Can you recommend a camera that would preform better, and work fully with indi-allsky.

Thanks !
Matt

aaronwmorris, 2024-10-29:
The ASI676 has been getting rave reviews for all sky cameras.  It seems to be one of the best options if you want a full image circle.

The reality is that most of the newer camera sensors are going to give great results.  You can use the Camera Simulator in indi-allsky to get an idea of how much coverage a lens will give different image sensor packages.

bluthen, 2024-10-29:
@mshetzer 
With the ASI224MC what are your complaints about the image quality? What lens do you have on it?

mshetzer, 2024-10-29:
I've been struggling with the setup, and finding the camera can be very limited with gain settings, and binning.

I can't get it to go above 60 gain.  With the limitation of 120 second exposures, I can barely see the milky way.  I really want an over exposed all sky camera so I can see the faintest of clouds. Also, I'd like a bit more resolution than 1.2m.   Maybe I'm missing something.

aaronwmorris, 2024-10-29:
> I can't get it to go above 60 gain.

You are not the first to have issues with the 60 gain on the 224.  The gain for the ASI224 should go up to 100, I believe.  Do you know what version of INDI you have installed?

ayryq, 2024-12-12:
I'm running that camera with gain at 100. The included lens is not very wide, and there's no image circle visible - it fills the frame out to the corners, which means there's no place to put text. But the quality seems fine (except I have a smudge on mine and the roof is covered in snow!). I'm new and maybe not picky enough though. I have exposure limited to 28s.
You can see mine if you click through the no-certificate warnings: https://lofgrenhome.com:2024/indi-allsky/
[image]

aaronwmorris, 2024-12-12:
Very nice!  There is a 1.44mm M12 lens you could buy that would show the full image circle with your camera.  You can check that out in the Camera Simulator.  If you were to get an M12 lens, you would need to get a CS-mount to M12 adapter.

https://www.amazon.com/1-44mm-Mount-Degree-Cameras-1-44MM/dp/B0DBHZKQVG

mshetzer, 2024-12-15:
Do these lenses 1.44mm, work on the ASI224MC ?   Saw they have an infrared version.  Wondering if that would be good as a second Allsky.

Thanks !

ayryq, 2024-12-15:
I actually have a 1.44 lens ([this one](https://www.amazon.com/gp/product/B07VCF4225)) and an adapter somewhere. Won't get to test it until the weather gets better. Hopefully it fits in the cutout I made in the box. Mine does not have an infrared filter (pink trees on that camera).
I was also looking at this one: [2.1mm CS mount](https://www.amazon.com/gp/product/B0B18R4CS4)
The included lens is 2.5mm.

aaronwmorris, 2024-12-16:
The 2.5mm and 2.1mm lenses have basically the same size image circle, although the FoV differs a little.  These are both listed as the ZWO lenses in the simulator.

ayryq, 2024-12-16:
I was wrong, I had a 2.1mm lens on there. I had a dry day today so I opened it up and stood on the roof for half an hour trying to get focus with the 1.44mm lens. I think it won't go down far enough.

ayryq, 2024-12-16:
I was wrong, I had a 2.1mm lens on there. I had a dry day today so I opened it up and stood on the roof for half an hour trying to get focus with the 1.44mm lens. I think it won't go down far enough.
Hard to tell without it being in focus, but it looks like the same FOV, nearly, just smaller. Basing this on the trees visible in lower left.
[image]

aaronwmorris, 2024-12-16:
@ayryq Do you have a link to (or image of) the M12 to CS-mount adapter you are using?

ayryq, 2024-12-16:
Sure, it's [this one](https://www.amazon.com/gp/product/B0DGTYFCRM). I don't think it's the problem—it's the back end of the lens itself that's contacting the camera.
I'd like to try again in better conditions (somehow I had both sun glare and featureless skies, besides numb hands and wind trying to blow me off the roof), but I'm also looking at [this](https://www.aliexpress.com/i/2251832726956909.html).
It's too bad there's no "live view" available. The 5s or so delay makes me impatient so I make bigger steps, so it's possible I missed the focus point.
My camera is installed through a chimney to the attic, and zip-tied in place so it's not reasonable to do this inside (after the first time).
I'll go mess with it again when conditions permit.

aaronwmorris, 2024-12-16:
For the M12 lens, it may be necessary to remove the optical window from in front of the sensor to get the lens close enough to the sensor.  The front assembly of ZWO planetary cameras can be unscrewed to reveal the PCB and the optical window.

closed 2025-02-20 by aaronwmorris