aaronwmorris/indi-allsky #1602: ASI676MC hanging

aaronwmorris, 2024-10-31:
I'd be curious how you handle this setup. I'm running a setup identical to yours and found that my 676MC was hanging due to overheating in my enclosure, even with 2 40mm fans running to remove the excess heat. The adsb does run effectively when attached directly to the pi, but the usb stick gets really hot. I believe I was seeing temps around 60-65c with outside ambient temps up to 90-95f

_Originally posted by @ph03nix42 in https://github.com/aaronwmorris/indi-allsky/discussions/1548#discussioncomment-11041665_

aaronwmorris, 2024-10-31:
If you could post your Support Info bundle, it would help.  Can you qualify what you mean by "hanging"?  Is it getting stuck in a "busy" state?

ph03nix42, 2024-10-31:
Support Detail:

/home/john/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 8245648 kB

System: Raspberry Pi 5 Model B Rev 1.0

Linux indi-allsky 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux
Hostname
indi-allsky
indi-allsky 

Time
Thu 31 Oct 11:18:40 CDT 2024

System timezone
US/Central

Uptime
 11:18:40 up  3:06,  0 user,  load average: 0.43, 0.41, 0.43

Memory
               total        used        free      shared  buff/cache   available
Mem:         8245648     1156016     5971808       35904     1249376     7089632
Swap:         204784           0      204784

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             3950816        0   3950816   0% /dev
tmpfs             824576     9568    815008   2% /run
/dev/mmcblk0p2 491815528 45961148 420871188  10% /
tmpfs            4122816      752   4122064   1% /dev/shm
tmpfs               5120       48      5072   1% /run/lock
/dev/mmcblk0p1    522230    64614    457616  13% /boot/firmware
tmpfs             824560        0    824560   0% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
52350

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether d8:3a:dd:c3:cd:81 brd ff:ff:ff:ff:ff:ff
    inet 192.XXX.XX.XX brd 192.XX.XX.XX scope global dynamic noprefixroute eth0
       valid_lft 75220sec preferred_lft 75220sec
    inet6 fe80::XXX:30ea:XXXX:cbdf/64 scope link noprefixroute 
[... 1200 lines cut ...]
    "ALT_DEG_MIN": 25,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,400 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": true,
    "DAYTIME_TRACK": false,
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,400 (Right)\n# anchor:ra (Right Justified)\n# color:200,200,200\nSatellites"
  }
}

```
#################################
###     end support info      ###
#################################

Below is the "hang" message from the website that just occurred:

[image]

aaronwmorris, 2024-10-31:
First thing we can try on the software side is lowering the USB bandwidth.  On the config page, on the Camera Tab at the bottom where the Default INDI config is entered, add the following config:

```
{
    "PROPERTIES" : {
        "CCD_CONTROLS" : {
            "BandWidth" : 40
        }
    },
    "SWITCHES" : {}
}
```

ph03nix42, 2024-10-31:
copies the config in, saved and just reset. I'll let you know if it occurs again.

ph03nix42, 2024-11-01:
Unfortunately it appears I've 2 additional "hangs". One overnight and an additional this morning.

aaronwmorris, 2024-11-01:
I did not have much hope for the USB bandwidth helping.  The most common cause of hangs like this is a bad USB cable for the camera.  If you have the option, I would go with the shortest USB cable possible and USB 2.0 if you have one available.

ph03nix42, 2024-11-01:
I did shorten from the original OEM cable from ZWO which might be causing the hangs, but we'll try a couple of things and see what happens. I appreciate all the help, this is such a great project and what you've built is amazing! Thank you!

nikiosna, 2024-11-02:
I have similiar issues with an ASI676MC pro too.
I recently bought one, tested indi-allsky on my old laptop with the camera -> everything fine.
Now I bought an "odroidn2l" there I got it working once, it took three exposures but currently it just wont work anymore.
I'm constantly getting exposure state BUSY or ALERT. I aready tried reinstalling and switching between the USB2 and USB3 port. 
```
2024-11-02T16:41:49.737923+01:00 odroidn2l [INFO] Capture-2-971/MainThread indi.getTelescopeRaDec() [881]: Telescope Coord: RA 19.07, Dec 52.20
2024-11-02T16:41:49.665530+01:00 odroidn2l [INFO] MainProcess-948/MainThread allsky._queueManualTasks() [1223]: Checking for manually submitted tasks
2024-11-02T16:41:49.637513+01:00 odroidn2l [INFO] Capture-2-971/MainThread indi.getCcdTemperature() [896]: Camera temperature: 26.3
2024-11-02T16:41:49.537770+01:00 odroidn2l [INFO] Capture-2-971/MainThread capture.detectNight() [1382]: Moon altitude: -0.0, phase 1.3%
2024-11-02T16:41:49.537687+01:00 odroidn2l [INFO] Capture-2-971/MainThread capture.detectNight() [1375]: Sun altitude: 1.1
2024-11-02T16:41:49.537588+01:00 odroidn2l [INFO] Capture-2-971/MainThread capture.saferun() [232]: Exposure state: BUSY
2024-11-02T16:41:49.537168+01:00 odroidn2l [INFO] Capture-2-971/MainThread capture.saferun() [231]: Camera last ready: 179.2s
```

aaronwmorris, 2024-11-02:
It is possible there is a bug in the latest firmware of the camera.  You may try installing indi 2.0.9 and see if that helps.

When the camera goes into an ALERT state, the only way to immediately fix that is by restarting the indiserver.

```
systemctl --user restart indiserver
```

aaronwmorris, 2024-11-07:
I have this wiki page for removing the PPA.  Maybe this will help.

https://github.com/aaronwmorris/indi-allsky/wiki/Uninstall-indi-repository

nikiosna, 2024-11-08:
Hm unfortunally not. Thats basically what I have already done. `apt list --installed | grep indi` shows nothing.
```
niklas@odroidn2l:~/Projects$ indiserver
2024-11-08T07:16:15: startup: indiserver
Usage: indiserver [options] driver [driver ...]
Purpose: server for local and remote INDI drivers
INDI Library: 2.0.9
Code v2.0.9. Protocol 1.7.
```
There has to be a bug in the build script when downgrading I think (or its something special in my setup).
I manually loaded the build release zip file from https://github.com/indilib/indi-3rdparty/releases/tag/v2.0.9
Und used the build commands provided there (cmake -> make -> make install). Did work fine.

Coming back to the original issue the camera is unfortunally still hung up with indi 2.0.9
```
2024-11-08T09:05:07.959218+01:00 odroidn2l [INFO] MainProcess-967/MainThread allsky._queueManualTasks() [1223]: Checking for manually submitted tasks
2024-11-08T09:05:01.937474+01:00 odroidn2l [INFO] Capture-2-1038/MainThread indi.getCcdTemperature() [896]: Camera temperature: 27.3
2024-11-08T09:05:01.837600+01:00 odroidn2l [INFO] Capture-2-1038/MainThread capture.detectNight() [1382]: Moon altitude: -43.8, phase 40.4%
2024-11-08T09:05:01.837526+01:00 odroidn2l [INFO] Capture-2-1038/MainThread capture.detectNight() [1375]: Sun altitude: 10.1
2024-11-08T09:05:01.837445+01:00 odroidn2l [INFO] Capture-2-1038/MainThread capture.saferun() [232]: Exposure state: BUSY
2024-11-08T09:05:01.837170+01:00 odroidn2l [INFO] Capture-2-1038/MainThread capture.saferun() [231]: Camera last ready: 22.1s
2024-11-08T09:04:54.956479+01:00 odroidn2l [INFO] MainProcess-967/MainThread allsky._queueManualTasks() [1223]: Checking for manually submitted tasks
2024-11-08T09:04:50.820643+01:00 odroidn2l [INFO] Capture-2-1038/MainThread indi.getCcdTemperature() [896]: Camera temperature: 27.3
2024-11-08T09:04:50.720708+01:00 odroidn2l [INFO] Capture-2-1038/MainThread capture.detectNight() [1382]: Moon altitude: -43.9, phase 40.4%
2024-11-08T09:04:50.720635+01:00 odroidn2l [INFO] Capture-2-1038/MainThread capture.detectNight() [1375]: Sun altitude: 10.1
2024-11-08T09:04:50.720544+01:00 odroidn2l [INFO] Capture-2-1038/MainThread capture.saferun() [232]: Exposure state: BUSY
2024-11-08T09:04:50.720307+01:00 odroidn2l [INFO] Capture-2-1038/MainThread capture.saferun() [231]: Camera last ready: 11.0s
2024-11-08T09:04:47.633966+01:00 odroidn2l [INFO] Capture-2-1038/Dummy-3 indi.newDevice() [224]: new device ZWO CCD ASI676MC
2024-11-08T09:04:47.547884+01:00 odroidn2l [INFO] Capture-2-1038/Dummy-3 indi.removeDevice() [227]: remove device ZWO CCD ASI676MC
2024-11-08T09:04:45.611399+01:00 odroidn2l [INFO] Capture-2-1038/MainThread capture.saferun() [581]: Total time since last exposure 0.1572 s
2024-11-08T09:04:45.611285+01:00 odroidn2l [INFO] Capture-2-1038/MainThread capture.saferun() [550]: Image queue depth: 0
2024-11-08T09:04:45.610955+01:00 odroidn2l [INFO] Capture-2-1038/MainThread indi.set_number() [1360]: Setting Duration (s) = 0.009999999776482582
2024-11-08T09:04:45.511050+01:00 odroidn2l [INFO] Capture-2-1038/MainThread capture.shoot() [1576]: Taking 0.01000000 s exposure (gain 0)
2024-11-08T09:04:44.359750+01:00 odroidn2l [INFO] Capture-2-1038/MainThread indi.set_number() [1360]: Setting BandWidth = 40
2024-11-08T09:04:44.259880+01:00 odroidn2l [INFO] Capture-2-1038/MainThread indi.configureDevice() [743]: Setting property (number) CCD_CONTROLS
2024-11-08T09:04:43.108840+01:00 odroidn2l [INFO] Capture-2-1038/MainThread indi.set_number() [1360]: Setting Y = 1
2024-11-08T09:04:43.108644+01:00 odroidn2l [INFO] Capture-2-1038/MainThread indi.set_number() [1360]: Setting X = 1
2024-11-08T09:04:43.008792+01:00 odroidn2l [INFO] Capture-2-1038/MainThread indi.configureDevice() [743]: Setting property (number) CCD_BINNING
2024-11-08T09:04:43.008590+01:00 odroidn2l [WARNING] Capture-2-1038/MainThread indi.setCcdBinning() [1263]: Setting CCD binning to (1, 1)
2024-11-08T09:04:41.953505+01:00 odroidn2l [INFO] MainProcess-967/MainThread allsky._queueManualTasks() [1223]: Checking for manually submitted tasks
2024-11-08T09:04:41.857332+01:00 odroidn2l [INFO] Capture-2-1038/MainThread indi.set_number() [1360]: Setting Gain = 0
2024-11-08T09:04:41.757670+01:00 odroidn2l [INFO] Capture-2-1038/MainThread indi.configureDevice() [743]: Setting property (number) CCD_CONTROLS
2024-11-08T09:04:41.757572+01:00 odroidn2l [WARNING] Capture-2-1038/MainThread indi.setCcdGain() [1135]: Setting CCD gain to 0
2024-11-08T09:04:41.757179+01:00 odroidn2l [WARNING] Capture-2-1038/MainThread indi.disableCcdCooler() [928]: Cooling not supported
2024-11-08T09:04:39.754531+01:00 odroidn2l [WARNING] Capture-2-1038/MainThread indi.disableCcdCooler() [923]: Disabling CCD cooling
2024-11-08T09:04:39.754417+01:00 odroidn2l [WARNING] Capture-2-1038/MainThread capture.reconfigureCcd() [1324]: Change to day
2024-11-08T09:04:39.704175+01:00 odroidn2l [INFO] Capture-2-1038/MainThread indi.getCcdTemperature() [896]: Camera temperature: 27.3
2024-11-08T09:04:39.603823+01:00 odroidn2l [INFO] Capture-2-1038/MainThread capture.detectNight() [1382]: Moon altitude: -43.9, phase 40.4%
2024-11-08T09:04:39.603679+01:00 odroidn2l [INFO] Capture-2-1038/MainThread capture.detectNight() [1375]: Sun altitude: 10.1
2024-11-08T09:04:39.603391+01:00 odroidn2l [INFO] Capture-2-1038/MainThread capture.saferun() [232]: Exposure state: unset
[... cut ...]
2024-11-08T09:04:02.118491+01:00 odroidn2l [INFO] MainProcess-967/MainThread allsky._startup() [283]: Platform: aarch64
2024-11-08T09:04:02.118393+01:00 odroidn2l [INFO] MainProcess-967/MainThread allsky._startup() [282]: Python version: 3.11.10
2024-11-08T09:04:02.118267+01:00 odroidn2l [INFO] MainProcess-967/MainThread allsky._startup() [280]: indi-allsky config level: 20241026.0
2024-11-08T09:04:02.117775+01:00 odroidn2l [INFO] MainProcess-967/MainThread allsky._startup() [279]: indi-allsky release: indi_v2024.09.1
```

aaronwmorris, 2024-11-08:
Is it possible you are out of space on the /tmp file system?

nikiosna, 2024-11-08:
Should be fine, dont have yet moved indi-allsky to sdcard yet, but root has 4.3gb free which I think are enough for a few test frames.
```
niklas@odroidn2l:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
tmpfs           361M  1.3M  360M   1% /run
/dev/mmcblk0p2   15G  9.8G  4.3G  70% /
tmpfs           1.8G  168K  1.8G   1% /dev/shm
tmpfs           5.0M  8.0K  5.0M   1% /run/lock
/dev/mmcblk0p1  128M   17M  112M  13% /media/boot
tmpfs           361M   20K  361M   1% /run/user/1000
/dev/mmcblk1p1  234G  1.3M  222G   1% /mnt/sdcard
```

aaronwmorris, 2024-11-08:
Have you tried rebooting since recompiling indi?

nikiosna, 2024-11-08:
yes

aaronwmorris, 2024-11-08:
Try rebuilding the pyindi-client python module.

```
./misc/rebuild_pyindi-client.sh
```

nikiosna, 2024-11-08:
Did so, reboot, log says now
```
2024-11-08T16:36:13.582537+01:00 odroidn2l [INFO] Capture-2-974/MainThread indi.__init__() [136]: PyIndi version: 2.0.9
```
But camera still BUSY

aaronwmorris, 2024-11-08:
Thinking a little outside the box here...

Let's try increasing the available current for the USB ports.  The Raspi5 limits total current to 0.6a by default.

Add this to `/boot/firmware/config.txt` and reboot.

```
usb_max_current_enable=1
```

nikiosna, 2024-11-08:
Hm I will try that later. Im running an odroid need to check how I can set this there. Also will try a powered HUB.
Meanwhile I did try to connect my local KStars to the indiserver on the odroid. And once I try to toggle an exposure there the control-panel crashes. But I did notice this in my dmesg which happens on each tried exposure:
```
[ 1042.770971] usb 1-2: reset high-speed USB device number 3 using xhci-hcd
```

aaronwmorris, 2024-11-08:
Sorry, I just scrolled up to OPs post and did not check the names, so that is on me.

The reset message is not unusual, I see that all the time in my logs with ZWO cameras.

I would suggest trying a different USB cable and/or a different USB port.  This is the most likely cause of problems like this.

nikiosna, 2024-11-09:
Did some more testing. I used a powered HUB now which draws around 700ma later 1700ma.
When connecting kstars now to the indiserver, I managed to start a exposure which hangs during "Downloading".

I tried to stop the indiserver service and running `indiserver -vvv indi_asi_ccd &> log.txt` to aquire some verbose log files of the indiserver. It looks like it definitly becomes idle and tries to do some exposures both with (kstars and indi-allsky).
```
2024-11-09T10:02:19: Driver indi_asi_ccd: read setNumberVector ZWO CCD ASI676MC CCD_EXPOSURE Busy
 CCD_EXPOSURE_VALUE='0'
2024-11-09T10:02:19: Driver indi_asi_ccd: read setNumberVector ZWO CCD ASI676MC CCD_EXPOSURE Busy
 CCD_EXPOSURE_VALUE='0'
2024-11-09T10:02:19: Client 9: queuing <setNumberVector device='ZWO CCD ASI676MC' name='CCD_EXPOSURE'>
2024-11-09T10:02:19: Client 9: sending msg nq 1:
<setNumberVector device="ZWO CCD ASI676MC" name="CCD_EXPOSURE" state="Busy" timeout="60" timestamp="2024-11-09T10:02:19">
    <oneNumber name="CCD_EXPOSURE_VALUE">
0
    </oneNumber>
</setNumberVector>

2024-11-09T10:02:19: Driver indi_asi_ccd: read setTextVector ZWO CCD ASI676MC DRIVER_INFO Idle
 DRIVER_NAME='ZWO CCD'
 DRIVER_EXEC='indi_asi_ccd'
 DRIVER_VERSION='2.4'
 DRIVER_INTERFACE='6'
2024-11-09T10:02:19: Driver indi_asi_ccd: read setTextVector ZWO CCD ASI676MC DRIVER_INFO Idle
 DRIVER_NAME='ZWO CCD'
 DRIVER_EXEC='indi_asi_ccd'
 DRIVER_VERSION='2.4'
 DRIVER_INTERFACE='6'
2024-11-09T10:02:19: Client 9: queuing <setTextVector device='ZWO CCD ASI676MC' name='DRIVER_INFO'>
2024-11-09T10:02:19: Client 9: sending msg nq 1:
<setTextVector device="ZWO CCD ASI676MC" name="DRIVER_INFO" state="Idle" timeout="60" timestamp="2024-11-09T10:02:19">
    <oneText name="DRIVER_NAME">
ZWO CCD
    </oneText>
    <oneText name="DRIVER_EXEC">
indi_asi_ccd
    </oneText>
    <oneText name="DRIVER_VERSION">
2.4
    </oneText>
    <oneText name="DRIVER_INTERFACE">
6
    </oneText>
</setTextVector>

2024-11-09T10:02:19: Driver indi_asi_ccd: read setBLOBVector ZWO CCD ASI676MC CCD1 Ok
2024-11-09T10:02:19: Driver indi_asi_ccd: read setBLOBVector ZWO CCD ASI676MC CCD1 Ok
2024-11-09T10:02:19: Client 9: queuing <setBLOBVector device='ZWO CCD ASI676MC' name='CCD1'>
2024-11-09T10:02:19: Driver indi_asi_ccd: read pingRequest   
2024-11-09T10:02:19: Driver indi_asi_ccd: read pingRequest   
2024-11-09T10:02:19: Driver indi_asi_ccd: read setNumberVector ZWO CCD ASI676MC CCD_EXPOSURE Ok
 CCD_EXPOSURE_VALUE='0'
2024-11-09T10:02:19: Driver indi_asi_ccd: read setNumberVector ZWO CCD ASI676MC CCD_EXPOSURE Ok
 CCD_EXPOSURE_VALUE='0'
2024-11-09T10:02:19: Client 9: queuing <setNumberVector device='ZWO CCD ASI676MC' name='CCD_EXPOSURE'>
2024-11-09T10:02:19: Driver indi_asi_ccd: sending msg nq 1:
<pingReply uid="SetBLOB/1"/>

2024-11-09T10:02:19: Client 9: sending msg nq 2:
<setBLOBVector device="ZWO CCD ASI676MC" name="CCD1" state="Ok" timeout="60" timestamp="2024-11-09T10:02:19">
    <oneBLOB name="CCD1" size="25233408" format=".bin" len="25233408">

2024-11-09T10:02:19: Client 9: sending msg nq 2:
NABgADQAYABGAEAARgBQAGkAgABXAFAARgBAAEYAUAAjADAAVwBQACMAMABGAEAAaQBQANABgADQAYABGAEAARgBQAGkAgABXAFAARgBAAEYAUAAjADAAVwBQACMAMABGAEAAaQBQA
(NOTE: .......much much much data truncated)

2024-11-09T10:02:19: Client 9: sending msg nq 2:

    </oneBLOB>
</setBLOBVector>

2024-11-09T10:02:19: Client 9: sending msg nq 1:
<setNumberVector device="ZWO CCD ASI676MC" name="CCD_EXPOSURE" state="Ok" timeout="60" timestamp="2024-11-09T10:02:19">
    <oneNumber name="CCD_EXPOSURE_VALUE">
0
    </oneNumber>
</setNumberVector>

2024-11-09T10:02:19: Driver indi_asi_ccd: read setNumberVector ZWO CCD ASI676MC CCD_TEMPERATURE Idle
 CCD_TEMPERATURE_VALUE='24.800000000000000711'
2024-11-09T10:02:19: Driver indi_asi_ccd: read setNumberVector ZWO CCD ASI676MC CCD_TEMPERATURE Idle
 CCD_TEMPERATURE_VALUE='24.800000000000000711'
2024-11-09T10:02:19: Client 9: queuing <setNumberVector device='ZWO CCD ASI676MC' name='CCD_TEMPERATURE'>
2024-11-09T10:02:19: Client 9: sending msg nq 1:
<setNumberVector device="ZWO CCD ASI676MC" name="CCD_TEMPERATURE" state="Idle" timeout="60" timestamp="2024-11-09T10:02:19">
    <oneNumber name="CCD_TEMPERATURE_VALUE">
24.800000000000000711
    </oneNumber>
</setNumberVector>

2024-11-09T10:02:21: Client 9: read: Connection reset by peer
2024-11-09T10:02:21: Client 9: shut down complete - bye!
```
The log file looks in both cases kstars and indi-allsky similiar and if data is captured

nikiosna, 2024-11-09:
I've tried a different cable, hub/no-hub, different ports and a lowered bandwidth (tried 40 and 80) now but nothing helps.
My current state now still is: every time I start up allsky tries to do an exposure and than the cam goes BUSY without recovering.
I think I will next try a complete clean set up and try my other touptek cc from my main telescope on the odroid

aaronwmorris, 2024-11-22:
Did this ever get any better?

nikiosna, 2024-11-24:
Tried a few things and just got it working again. For me the only solution was to downgrade from ubuntu 24.04 to `ubuntu-22.04-4.9-minimal`

There seem to be some known issues with the usb bus on the odroid-n2, so better stick to some well hung os/kernel version.
https://www.reddit.com/r/ODroid/comments/cs0toc/odroid_n2_and_usb3_bus_saturation_issues/

aaronwmorris, 2024-11-24:
This is unfortunate, but not surprising.  If you look at the number of workarounds and parameters available for USB devices, its surprising USB even works as well as it does.

cgmiller76, 2024-11-25:
I am having this same issue on one of my systems.  It's running on an RPi 4B.  I'll update my logs if that helps with the overall issue.

cgmiller76, 2024-11-25:
/home/pi/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 7996380 kB

System: Raspberry Pi 4 Model B Rev 1.4

Linux Astro-INDI-AllSky-Pi 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
Hostname
Astro-INDI-AllSky-Pi
Astro-INDI-AllSky-Pi.attlocal.net Astro-INDI-AllSky-Pi.attlocal.net Astro-INDI-AllSky-Pi.attlocal.net Astro-INDI-AllSky-Pi.attlocal.net Astro-INDI-AllSky-Pi.attlocal.net Astro-INDI-AllSky-Pi.attlocal.net 

Time
Sun 24 Nov 20:37:30 EST 2024

System timezone
US/Eastern

Uptime
 20:37:30 up 7 min,  2 users,  load average: 0.45, 0.38, 0.21

Memory
               total        used        free      shared  buff/cache   available
Mem:         7996380      890716     6312420       23404      942696     7105664
Swap:         204796           0      204796

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             3724900        0   3724900   0% /dev
tmpfs             799640     1364    798276   1% /run
/dev/sda2      490615056 54456200 411163512  12% /
tmpfs            3998188      388   3997800   1% /dev/shm
tmpfs               5120       16      5104   1% /run/lock
/dev/sda1         523244    77656    445588  15% /boot/firmware
tmpfs             799636       44    799592   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
34076

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether dc:a6:32:c4:cc:13 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.161/24 brd 192.168.1.255 scope global dynamic noprefixroute eth0
       valid_lft 85943sec preferred_lft 85943sec
    inet6 2600:1700:7b11:e430::2a/128 scope global dynamic noprefixroute 
       valid_lft 3144sec preferred_lft 3144sec
    inet6 2600:1700:7b11:e430:407:d1d3:841c:f050/64 scope global dynamic noprefixroute 
[... 1339 lines cut ...]
    "CERT_BYPASS": true,
    "USERNAME": "",
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,200 (Right)\n# anchor:ra (Right Justified)\n# color:200,200,200\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": false,
    "DAYTIME_TRACK": false,
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,300 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nSatellites"
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-11-25:
@cgmiller76 You might try plugging into one of the USB2 ports to see if that helps with the situation.  I am assuming you have already tried to adjust the USB bandwidth?

cgmiller76, 2024-11-25:
I have adjusted the Bandwith to 40 as shown here.  I had not tried the USB2 port, so I just went and switched that (this is a test system that I am making sure works in order to deploy the camera in my permanent setup).  I'll update after it runs for a bit to see if that change resolves it (I hope so!)  Thanks for the help!
[image]

cgmiller76, 2024-11-25:
So far, so good!  MY ASI676MC has run all night without issue on the Raspberry Pi 4 build after plugging it into the USB2 port.  Thanks for the tip!

closed 2025-02-20 by aaronwmorris