aaronwmorris/indi-allsky #1603: Camera Not Looping

FreeMeWat, 2024-11-03:
I  went into config to try out the stretch feature and turned on SCNR, and now the camera is no longer looping. I know it has no issue connecting because when I goto SQM, I can get a live view. But the "Latest" tab is showing images from 40 minutes ago, and gallery has no new images since then. I've restarted and it's not helping. Status said STOPPED, but I went into services and started everything. Now it says running, but there's still no photos being taken :(. Please help if you can.

aaronwmorris, 2024-11-03:
Can you post your Support Info bundle?

FreeMeWat, 2024-11-03:
you know what it is? DST kicked in and it is saving images of the same timestamps, it threw it all out of what. I bet it starts working again in 13 minutes when the time can pick back up at 2 AM (right now its recording 1AM twice)

FreeMeWat, 2024-11-03:
That is why my last frame on the "latest tab" is 1:59:45, once it hit 2am - it must have triggered fall back DST and its now waiting for the next frame at 2:00:00 before it will display it.

closed 2024-11-03 by FreeMeWat