aaronwmorris/indi-allsky #1604: [ENHANCEMENT] Support OneInchEye IMX283 camera

krzysztofkuczek, 2024-11-04:
Hi,
Could you please implement support for the IMX283 camera.
There is working implementation for theallsky soft, so maybe it would be easy to do
https://github.com/AllskyTeam/allsky/issues/3935
Thank you in advance.
Kris

aaronwmorris, 2024-11-04:
Merged #1606 with support for the IMX283 camera.

krzysztofkuczek, 2024-11-05:
HI, I was able to compile inid-allsky, and it seems, to be working, but the image is turned green. The problem does not happened when the camera is accessed directly vi rpicam-hello....
[image]

aaronwmorris, 2024-11-05:
The green bias is normal.  indi-allsky accesses the RAW/DNG data by default and because there are two green pixels for every red and green, you get a green bias.

You may either adjust the R, G, and B colors manually or you can use one of the SCNR algorithms to remove the green bias.  You may also re-enable the libcamera AWB controls which will correct the colors as well.

The problem with AWB with libcamera is it requires the camera to take multiple exposures to achieve proper white balance.  During the daytime this is not so bad, but at night, a 30s exposure may require 2-3 minutes to complete.

krzysztofkuczek, 2024-11-05:
Thank you for explanation.
Have a nice day.
Kris

closed 2024-11-22 by aaronwmorris