aaronwmorris/indi-allsky #1609: Demosaicing fails in call to cvtColor due to format of image dtype

mburge, 2024-11-06:
The call to opencv's `cvtColor` in `processing.py` debayer function fails with an exception due to the image not passing a bit depth assertion.

```
        self.image = cv2.cvtColor(self.image, debayer_algorithm)
```

causes

```
indi-allsky.log:2024-11-06T01:29:53.574389-05:00 allsky [ERROR] MainProcess-1306/MainThread allsky._startImageWorker() [389]: Image worker exception: cv2.error: OpenCV(4.9.0) /io/opencv/modules/imgproc/src/demosaicing.cpp:1693: error: (-215:Assertion failed) depth == CV_8U || depth == CV_16U in function 'demosaicing'
```

I always thought this was due to my camera, ZWO CCD ASI294MC Pro, and the `ASI_IMG_RAW16` driver scaling or not scaling the camera's native 14 bit data.  I used to fix it by preceding the call with a quick cast:

```
        self.image = numpy.array(self.image, dtype=numpy.uint16) 
        self.image = cv2.cvtColor(self.image, debayer_algorithm)
```

but this no longer works.  I never submitted it as a patch as it seemed a hack. Any ideas?  Thank you!

aaronwmorris, 2024-11-06:
I do not think the issue is with the bit depth.  indi-allsky would automatically deal with the 14-bit data correctly.

This error would imply that the bayered image is not uint8 or uint16 data.  I have only ever seen uint8 or uint16 data, so I wonder what it could be.

Can you add a log message before the cv2.cvtColor() method to log the dtype of the array?

```python
logger.info('dtype: %s', str(self.image.dtype))
self.image = numpy.array(self.image, dtype=numpy.uint16) 
self.image = cv2.cvtColor(self.image, debayer_algorithm)
```

mburge, 2024-11-06:
The exception does not appear now, I wonder if it is because it is in day mode and hence a different image processing chain is being applied. The logger is showing it correctly as a uint16. I'm going to change the timezone so it is evening and see if I can duplicate the error.

aaronwmorris, 2024-11-06:
Can you post your Support Info bundle here?

mburge, 2024-11-06:
Yes, changing the timezone to somewhere it is evening does not seem to be sufficient to get Indi-all sky to change to night mode. It is still running as day mode after the time change.

/home/astra/dev/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 8245584 kB

System: Raspberry Pi 5 Model B Rev 1.0

Linux allsky 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
Hostname
allsky
customer.ashnvax2.pop.starlinkisp.net customer.ashnvax2.pop.starlinkisp.net 

Time
Thu  7 Nov 02:48:40 CST 2024

System timezone
Asia/Shanghai

Uptime
 02:48:40 up 4 min,  1 user,  load average: 0.19, 0.33, 0.17

Memory
               total        used        free      shared  buff/cache   available
Mem:         8245584     1132096     6680048       38544      558448     7113488
Swap:         204320           0      204320

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             3947712        0   3947712   0% /dev
tmpfs             824560    13824    810736   2% /run
/dev/mmcblk0p2 491513080 31835620 434694140   7% /
tmpfs            4122784      736   4122048   1% /dev/shm
tmpfs               5120       48      5072   1% /run/lock
/dev/mmcblk0p1    522232    78276    443956  15% /boot/firmware
tmpfs             824544      144    824400   1% /run/user/1000

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
56750

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether d8:3a:dd:f7:51:22 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether c8:3a:35:a0:e0:b5 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.203/24 brd 192.168.1.255 scope global dynamic noprefixroute wlan0
[... 1324 lines cut ...]
    "CERT_BYPASS": true,
    "USERNAME": "",
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,300 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": true,
    "DAYTIME_TRACK": false,
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,200 (Right)\n# anchor:ra (Right Justified)\n# color:200,200,200\nSatellites"
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-11-06:
Thanks!  That all looks normal.

You would have to change the Sun Altitude setting on the Location tab to force switching to night mode.

mburge, 2024-11-06:
Awh, good to know, I'll try that out and see if I can recreate the error.

mburge, 2024-11-07:
Nothing during night mode, but there were no stars so the stacking and registration didn't run as normal.  Will try that out soon as the clouds disperse.  I have a feeling it could be a result of that process.  If it isn't, then I'm not sure where it came from, but it will have disappeared, which is great.

aaronwmorris, 2024-11-07:
Oh well.  Just let me know if it happens again.

closed 2024-12-14 by mburge