aaronwmorris/indi-allsky #1629: Error Creating Wrapped Keogram Timelapse

Leogeddes, 2024-11-10:
Hey, I'm having the below errors when using the new wrap keogram timelapse feature, no issues with standard time lapse creation, I've tried setting scaling to 50% and this has made no difference. Any ideas what the issue could be.

Cheers in advance

2024-11-10T18:09:05.541482+13:00 allsky [ERROR] MainProcess-18464/MainThread allsky._startVideoWorker() [454]: Video worker exception: 
2024-11-10T18:09:05.541416+13:00 allsky [ERROR] MainProcess-18464/MainThread allsky._startVideoWorker() [454]: Video worker exception: ValueError: operands could not be broadcast together with shapes (2600,2636,3) (2600,2600,3) 
2024-11-10T18:09:05.541341+13:00 allsky [ERROR] MainProcess-18464/MainThread allsky._startVideoWorker() [454]: Video worker exception:                           ~~~~~~~~^~~~~~~~~~~~~~~~~~
2024-11-10T18:09:05.541259+13:00 allsky [ERROR] MainProcess-18464/MainThread allsky._startVideoWorker() [454]: Video worker exception:     image_with_keogram = (f_image * (1 - alpha_mask) + wrapped_keogram_bgr * alpha_mask).astype(numpy.uint8)
2024-11-10T18:09:05.541186+13:00 allsky [ERROR] MainProcess-18464/MainThread allsky._startVideoWorker() [454]: Video worker exception:   File "/home/admin/indi-allsky/indi_allsky/timelapse_preprocessor/preProcessorWrapKeogram.py", line 149, in wrap
2024-11-10T18:09:05.541119+13:00 allsky [ERROR] MainProcess-18464/MainThread allsky._startVideoWorker() [454]: Video worker exception:     self.wrap(i, f, self.seqfolder)
2024-11-10T18:09:05.541042+13:00 allsky [ERROR] MainProcess-18464/MainThread allsky._startVideoWorker() [454]: Video worker exception:   File "/home/admin/indi-allsky/indi_allsky/timelapse_preprocessor/preProcessorWrapKeogram.py", line 57, in main
2024-11-10T18:09:05.540927+13:00 allsky [ERROR] MainProcess-18464/MainThread allsky._startVideoWorker() [454]: Video worker exception:     self.pre_processor.main(file_list_ordered)
2024-11-10T18:09:05.540857+13:00 allsky [ERROR] MainProcess-18464/MainThread allsky._startVideoWorker() [454]: Video worker exception:   File "/home/admin/indi-allsky/indi_allsky/timelapse.py", line 99, in generate
2024-11-10T18:09:05.540780+13:00 allsky [ERROR] MainProcess-18464/MainThread allsky._startVideoWorker() [454]: Video worker exception:     tg.generate(video_file, timelapse_files)
2024-11-10T18:09:05.540655+13:00 allsky [ERROR] MainProcess-18464/MainThread allsky._startVideoWorker() [454]: Video worker exception:   File "/home/admin/indi-allsky/indi_allsky/video.py", line 436, in generateVideo
2024-11-10T18:09:05.540590+13:00 allsky [ERROR] MainProcess-18464/MainThread allsky._startVideoWorker() [454]: Video worker exception:     action_method(task, **kwargs)
2024-11-10T18:09:05.540517+13:00 allsky [ERROR] MainProcess-18464/MainThread allsky._startVideoWorker() [454]: Video worker exception:   File "/home/admin/indi-allsky/indi_allsky/video.py", line 218, in processTask
2024-11-10T18:09:05.540446+13:00 allsky [ERROR] MainProcess-18464/MainThread allsky._startVideoWorker() [454]: Video worker exception:     self.processTask(v_dict)
2024-11-10T18:09:05.540357+13:00 allsky [ERROR] MainProcess-18464/MainThread allsky._startVideoWorker() [454]: Video worker exception:   File "/home/admin/indi-allsky/indi_allsky/video.py", line 185, in saferun
2024-11-10T18:09:05.540288+13:00 allsky [ERROR] MainProcess-18464/MainThread allsky._startVideoWorker() [454]: Video worker exception:     self.saferun()
2024-11-10T18:09:05.540207+13:00 allsky [ERROR] MainProcess-18464/MainThread allsky._startVideoWorker() [454]: Video worker exception:   File "/home/admin/indi-allsky/indi_allsky/video.py", line 157, in run
2024-11-10T18:09:05.539985+13:00 allsky [ERROR] MainProcess-18464/MainThread allsky._startVideoWorker() [454]: Video worker exception: Traceback (most recent call last):

aaronwmorris, 2024-11-10:
Merged #1630 with some fixes that hopefully address the problem.  I found some mistakes in some of the calculations.

Leogeddes, 2024-11-11:
Works perfectly now, thanks Aaron.

closed 2024-11-12 by aaronwmorris