aaronwmorris/indi-allsky #1636: i2c devices not working as expected

Vertikar, 2024-11-12:
So I've just hooked up my TSL light sensor and the SI7021 temperature/humidity sensor after not having it running for a while, but left the Mlx90614 disconnected for now, and I'm getting a Remote I/O error on the connected devices, but an "No I2C device at address: 0x5a" error for the Mlx as expected.
 i2c detect shows the devices there, and a python script for the SI sensor works fine.

### Troubleshooting steps
Tried changing the I2C address in the config and back again incase something in the config file had got corrupted, but still nothing. Also updated and rebooted and still getting the same

### indi-allsky support info

https://gist.github.com/Vertikar/a50e8852810a6126c9541a040041a6ed

### output logs 
basic testing

```
karl@allsky-mobile:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- 29 -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
karl@allsky-mobile:~ $ date
Tue 12 Nov 17:23:25 AEDT 2024
karl@allsky-mobile:~ $ python SI7021.py
Humidity %RH: 46.87%
Temperature Celsius: 28.31°C
karl@allsky-mobile:~ $ date
Tue 12 Nov 17:23:33 AEDT 2024
karl@allsky-mobile:~ $
```

indi-allsky logs tailed and grepped - 
_tail -f /var/log/indi-allsky/indi-allsky.log | grep Sensor_
```
Nov 12 17:20:42 allsky-mobile [INFO] MainProcess-3861/MainThread allsky._startSensorWorker() [519]: Starting Sensor-44 worker
Nov 12 17:20:42 allsky-mobile [WARNING] Sensor-44-4266/MainThread tempSensorSi7021.__init__() [94]: Initializing [Temp-RH] Si7021 I2C temperature device @ 0x40
Nov 12 17:20:42 allsky-mobile [WARNING] Sensor-44-4266/MainThread lightSensorTsl2591.__init__() [102]: Initializing [Light] TSL2591 I2C light sensor device @ 0x29
Nov 12 17:20:43 allsky-mobile [WARNING] Sensor-44-4266/MainThread tempSensorMlx90614.__init__() [73]: Initializing [SkyTemp] MLX90614 I2C temperature device @ 0x5a
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception: Traceback (most recent call last):
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:   File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/adafruit_bus_device/i2c_device.py", line 175, in __probe_for_device
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:     self.i2c.writeto(self.device_address, b"")
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:   File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/busio.py", line 220, in writeto
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:     return self._i2c.writeto(address, buffer, stop=True)
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:   File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 60, in writeto
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:     self._i2c_bus.write_bytes(address, buffer[start:end])
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:   File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/Adafruit_PureIO/smbus.py", line 303, in write_bytes
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:     self._device.write(buf)
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception: OSError: [Errno 121] Remote I/O error
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception: During handling of the above exception, another exception occurred:
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception: Traceback (most recent call last):
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:   File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/adafruit_bus_device/i2c_device.py", line 181, in __probe_for_device
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:     self.i2c.readfrom_into(self.device_address, result)
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:   File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/busio.py", line 210, in readfrom_into
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:     return self._i2c.readfrom_into(address, buffer, stop=True)
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:   File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 67, in readfrom_into
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:     readin = self._i2c_bus.read_bytes(address, end - start)
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:   File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/Adafruit_PureIO/smbus.py", line 170, in read_bytes
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:     return self._device.read(number)
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception: OSError: [Errno 121] Remote I/O error
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception: During handling of the above exception, another exception occurred:
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception: Traceback (most recent call last):
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:   File "/opt/indi-allsky/indi_allsky/sensor.py", line 125, in run
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:     self.saferun()
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:   File "/opt/indi-allsky/indi_allsky/sensor.py", line 135, in saferun
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:     self.init_sensors()  # sensors before dew heater and fan
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:   File "/opt/indi-allsky/indi_allsky/sensor.py", line 390, in init_sensors
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:     self.sensors[2] = c_sensor(
[... cut ...]
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:     raise ValueError("No I2C device at address: 0x%x" % self.device_address)
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception: ValueError: No I2C device at address: 0x5a
Nov 12 17:20:55 allsky-mobile [ERROR] MainProcess-3861/MainThread allsky._startSensorWorker() [512]: Sensor worker exception:
Nov 12 17:20:55 allsky-mobile [INFO] MainProcess-3861/MainThread allsky._startSensorWorker() [519]: Starting Sensor-45 worker
```

Vertikar, 2024-11-12:
Also getting the same output from the sensor_test script

```
(indi-allsky) karl@allsky-mobile:/opt/indi-allsky $ misc/sensor_test.py
[WARNING]: Initializing [Temp-RH] Si7021 I2C temperature device @ 0x40
[WARNING]: Initializing [Light] TSL2591 I2C light sensor device @ 0x29
[WARNING]: Initializing [SkyTemp] MLX90614 I2C temperature device @ 0x5a
Traceback (most recent call last):
  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/adafruit_bus_device/i2c_device.py", line 175, in __probe_for_device
    self.i2c.writeto(self.device_address, b"")
  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/busio.py", line 220, in writeto
    return self._i2c.writeto(address, buffer, stop=True)
  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 60, in writeto
    self._i2c_bus.write_bytes(address, buffer[start:end])
  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/Adafruit_PureIO/smbus.py", line 303, in write_bytes
    self._device.write(buf)
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/adafruit_bus_device/i2c_device.py", line 181, in __probe_for_device
    self.i2c.readfrom_into(self.device_address, result)
  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/busio.py", line 210, in readfrom_into
    return self._i2c.readfrom_into(address, buffer, stop=True)
  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 67, in readfrom_into
    readin = self._i2c_bus.read_bytes(address, end - start)
  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/Adafruit_PureIO/smbus.py", line 170, in read_bytes
    return self._device.read(number)
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/indi-allsky/misc/sensor_test.py", line 182, in <module>
    ts.main()
  File "/opt/indi-allsky/misc/sensor_test.py", line 81, in main
    self.init_sensors()
  File "/opt/indi-allsky/misc/sensor_test.py", line 162, in init_sensors
    self.sensors[2] = c_sensor(
  File "/opt/indi-allsky/indi_allsky/devices/sensors/tempSensorMlx90614.py", line 75, in __init__
    self.mlx = adafruit_mlx90614.MLX90614(i2c, address=i2c_address)
  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/adafruit_mlx90614.py", line 113, in __init__
    self._device = i2c_device.I2CDevice(i2c_bus, address)
  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/adafruit_bus_device/i2c_device.py", line 62, in __init__
    self.__probe_for_device()
  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/adafruit_bus_device/i2c_device.py", line 184, in __probe_for_device
    raise ValueError("No I2C device at address: 0x%x" % self.device_address)
ValueError: No I2C device at address: 0x5a
```

Vertikar, 2024-11-12:
Hmm strange, so running from the indi-allsky virtualenv, I grabbed the example adafruit circuitpython code from here - https://github.com/adafruit/Adafruit_CircuitPython_SI7021/blob/main/examples/si7021_simpletest.py

and that seemed to work.

```
(indi-allsky) karl@allsky-mobile:~ $ python3 ./blinka_si7021.py

Temperature: 19.7 C
Humidity: 87.0 %

Temperature: 19.7 C
Humidity: 87.0 %

Temperature: 19.7 C
Humidity: 87.0 %

Temperature: 19.7 C
Humidity: 87.0 %
```

I did notice you import adafruit-circuitpython-si7021 in https://github.com/aaronwmorris/indi-allsky/blob/main/requirements/requirements_gpio.txt rather than adafruit_si7021 and tried it with that, but it wasn't able to import it.

```
(indi-allsky) karl@allsky-mobile:~ $ python3 ./blinka_si7021.py
  File "/home/karl/./blinka_si7021.py", line 9
    import adafruit-circuitpython-si7021
                   ^
SyntaxError: invalid syntax
(indi-allsky) karl@allsky-mobile:~ $
```

aaronwmorris, 2024-11-12:
I think your Si7021 and TSL2591 are working fine, however the MLX90614 is causing all of the problems.  You probably need to remove the reference to the MLX sensor in the indi-allsky config.

Vertikar, 2024-11-12:
> In the meantime, I have merged #1637 that will bypass sensors that cannot be initialized so the program will continue to run.

Thanks Aaron, love your work!

Vertikar, 2024-11-12:
> I think your Si7021 and TSL2591 are working fine, however the MLX90614 is causing all of the problems. You probably need to remove the reference to the MLX sensor in the indi-allsky config.

Tested this and it started working. I'll test out your fix shortly too.