aaronwmorris/indi-allsky #1645: Extra text is displaced when Moon Mode is active

wjschmitt, 2024-11-16:
I have an extra text file that gets added to the image in the correct (ie. upper left) position, except when Moon Mode is active.  Then the extra text is displaced and appears under the upper right label.  When Moon Mode is deactivated, text moves back to upper left.   I've tried reducing the font size and removing lines from the extra text, but the text will not appear in upper left when moon mode is active.

[image]
[image]

aaronwmorris, 2024-11-17:
Can you post the contents of the extra text?

wjschmitt, 2024-11-17:
One more clue - I believe this started happening after a recent update.  Don't remember the date I did it.  I have another identical setup where it doesn't happen at all.   Both systems are on v2024.09.1.
Extra text file contains the following text, everything below the line.  File is attached.
[umid.txt](https://github.com/user-attachments/files/17788664/umid.txt)

I've looked for leading or trailing spaces.  I've removed lines.  Nothing helped.
__________________________________
OAT       : 70.2 F
Humidity  : 62%
Dew Point : 56.7 F
Clouds %  : 100%
Case Temp : 80.4 F
Case Hum  : 50.6%
CPU Temp  : 96.8
Disk Used : 68.6%
Moon Phase: Waning Gibbous
Illumin % : 100%
Moonrise  : 15:30
Moonset   : 06:50

wjschmitt, 2024-11-17:
I'm attaching a shot of the test system where the issue doesn't appear.  Same extra text file.  Moon mode is active.  Labels are all normal.
[image]

aaronwmorris, 2024-11-17:
Its interesting when the bad placement happens, its showing the satellite tracking label.  I am going to investigate around the satellite tracking code.

Can you post your Support Info bundle here?

wjschmitt, 2024-11-17:
#################################
### indi-allsky support info  ###
#################################
```

Distribution: ubuntu
Release: 22.04
Arch: x86_64
Bits: 64

CPUs: 4
Memory: 8132344 kB

System: Generic PC

Linux SpeedDesk 6.8.0-47-generic #47~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Oct  2 16:16:55 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
Hostname
SpeedDesk
SpeedDesk.localdomain SpeedDesk SpeedDesk SpeedDesk 

Time
Sun Nov 17 09:21:24 AM EST 2024

System timezone
America/New_York

Uptime
 09:21:24 up 30 days, 34 min,  1 user,  load average: 1.05, 0.97, 0.47

Memory
               total        used        free      shared  buff/cache   available
Mem:         8132344     2140264      329144       14860     5662936     5639008
Swap:        3991548     1207380     2784168

Filesystems
Filesystem     1K-blocks      Used Available Use% Mounted on
tmpfs             813236      1564    811672   1% /run
/dev/sda3      131005476 103158208  21146420  83% /
tmpfs            4066172       180   4065992   1% /dev/shm
tmpfs               5120         0      5120   0% /run/lock
/dev/sda2         524252      6228    518024   2% /boot/efi
tmpfs             813232       104    813128   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info

system python: Python 3.10.12

indiserver: /usr/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: ens18: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether bc:24:11:9e:c5:54 brd ff:ff:ff:ff:ff:ff
    altname enp0s18
    inet 192.168.1.148/24 brd 192.168.1.255 scope global dynamic noprefixroute ens18
       valid_lft 65759sec preferred_lft 65759sec
    inet6 fdd4:de4c:e98e:baa2:8b0b:2c62:a80d:62ff/64 scope global temporary dynamic 
       valid_lft 1770sec preferred_lft 1770sec
    inet6 fdd4:de4c:e98e:baa2:c90f:2d7f:75e2:6d0f/64 scope global dynamic mngtmpaddr noprefixroute 
       valid_lft 1770sec preferred_lft 1770sec
    inet6 fe80::313b:6708:3258:7907/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
[... 1122 lines cut ...]
    "CERT_BYPASS": true,
    "USERNAME": "",
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,300 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": true,
    "DAYTIME_TRACK": false,
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,200 (Right)\n# anchor:ra (Right Justified)\n# color:200,200,200\nSatellites"
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-11-17:
Merged #1649 which should fix this.

I think the problem was the Satellite tracking text was being processed before the Extra Text.  The satellite text changes the cursor location, so the extra text would have been placed after the satellite tracking text.

wjschmitt, 2024-11-20:
Yes, I think this is caused by the satellite tracking labels.  It happens whether moon mode is active or not.  So I turned off satellite tracking and labels to see if that fixes it.

closed 2025-02-20 by aaronwmorris