aaronwmorris/indi-allsky #1647: How to connect to remote indiserver?

warwound, 2024-11-17:
I have 2 couple of Pi HQ cameras running on my Pi5 (runs Stellarmate OS) along with two docker instances of indi-allsky.
(One camera has wide angle lens, the other a more telephoto lens).
This has worked very well for many months.

Now I want to add a third Pi HQ camera to my setup, my options are: 

Connect the camera to a Pi Zero 2W running indi-allsky.
This would be my preferred choice, with a 3d printed case it'd be a nice compact unit.
But the Zero 2W is seriously underpowered and laggy once I get indi-allsky running.

Connect the camera to a Pi4B running indi-allsky.
The Pi4B is perfect in everyway except form factor.

Now I have an idea:
Run only indiserver on the Zero 2W (with third camera connected) and connect to it from a new docker instance of indi-allsky running on my Pi5.

I've searched for info but found no answer...
How do I configure indi-allsky to connect to a remote indiserver?

I see the indi-allsky docker documentation states: "In most cases, INDI (core and 3rd-party) will probably have to be installed on the host running the indiserver container."

I could remove the indiserver container from both of my docker instances of indi-allsky.
Then I'd need to configure both instances of indi-allsky to connect to indiserver running directly on the Pi5 (as part of Stellarmate OS).
This should save memory and improve performance.

Again how do I configure indi-allsky to connect to a remote indiserver?

Thanks.

aaronwmorris, 2024-11-17:
On the Config page for indi-allsky, under the Camera Tab right at the top, there is a config option to configure the hostname of the indiserver.

If you are using containers, the default hostname should be "indiserver.indi.allsky" which is the hostname of the docker container running the indiserver.

warwound, 2024-11-18:
Ahhh I see the indiserver hostname box now!

On my Pi4 running india-allsky I changed this to the hostname of my Pi5 running Stellarmate.
Also had to set the 'Camera Interface' drop down to 'INDI'.

My Pi4 connects to indiserver on my Pi5 but fails to capture images with the exceptions:

[quote]
2024-11-18T12:58:58.350307+00:00 pi4 [ERROR] MainProcess-856/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: 
2024-11-18T12:58:58.350072+00:00 pi4 [ERROR] MainProcess-856/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: Exception: Gain config not implemented for indi_libcamera_ccd, open an enhancement request
2024-11-18T12:58:58.349756+00:00 pi4 [ERROR] MainProcess-856/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:     raise Exception('Gain config not implemented for {0:s}, open an enhancement request'.format(indi_exec))
2024-11-18T12:58:58.349471+00:00 pi4 [ERROR] MainProcess-856/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:   File "/home/martin/Documents/indi-allsky/indi_allsky/camera/indi.py", line 1120, in getCcdGain
2024-11-18T12:58:58.349258+00:00 pi4 [ERROR] MainProcess-856/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:                 ^^^^^^^^^^^^^^^^^
2024-11-18T12:58:58.349002+00:00 pi4 [ERROR] MainProcess-856/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:     gain_info = self.getCcdGain()
2024-11-18T12:58:58.348717+00:00 pi4 [ERROR] MainProcess-856/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:   File "/home/martin/Documents/indi-allsky/indi_allsky/camera/indi.py", line 595, in getCcdInfo
2024-11-18T12:58:58.348488+00:00 pi4 [ERROR] MainProcess-856/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-11-18T12:58:58.348271+00:00 pi4 [ERROR] MainProcess-856/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:     ccd_info = self.indiclient.getCcdInfo()
2024-11-18T12:58:58.347726+00:00 pi4 [ERROR] MainProcess-856/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:   File "/home/martin/Documents/indi-allsky/indi_allsky/capture.py", line 746, in _initialize
2024-11-18T12:58:58.347528+00:00 pi4 [ERROR] MainProcess-856/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:     self._initialize()
2024-11-18T12:58:58.347349+00:00 pi4 [ERROR] MainProcess-856/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:   File "/home/martin/Documents/indi-allsky/indi_allsky/capture.py", line 189, in saferun
2024-11-18T12:58:58.347142+00:00 pi4 [ERROR] MainProcess-856/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:     self.saferun()
2024-11-18T12:58:58.346919+00:00 pi4 [ERROR] MainProcess-856/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:   File "/home/martin/Documents/indi-allsky/indi_allsky/capture.py", line 177, in run
2024-11-18T12:58:58.346165+00:00 pi4 [ERROR] MainProcess-856/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: Traceback (most recent call last):
2024-11-18T12:58:51.620889+00:00 pi4 [INFO] Capture-4-1001/MainThread indi.getCcdInfo() [560]: Detecting bayer pattern
2024-11-18T12:58:50.380528+00:00 pi4 [WARNING] Capture-4-1001/MainThread capture._initialize() [653]: Connecting to CCD device LibCamera imx477-0
2024-11-18T12:58:50.380306+00:00 pi4 [INFO] Capture-4-1001/MainThread indi.findGps() [710]: Found 0 GPSs
2024-11-18T12:58:50.379966+00:00 pi4 [INFO] Capture-4-1001/MainThread indi._findGpss() [690]: Searching for available GPS interfaces
2024-11-18T12:58:50.368227+00:00 pi4 [ERROR] Capture-4-1001/MainThread indi.findTelescope() [684]: No telescopes found
2024-11-18T12:58:50.367997+00:00 pi4 [INFO] Capture-4-1001/MainThread indi.findTelescope() [677]: Found 1 Telescopess
2024-11-18T12:58:50.367779+00:00 pi4 [INFO] Capture-4-1001/MainThread indi._findTelescopes() [668]:  Detected AZ-GTi Alt-Az Wired
2024-11-18T12:58:50.367545+00:00 pi4 [INFO] Capture-4-1001/MainThread indi._findTelescopes() [657]: Searching for available telescopes/mounts
2024-11-18T12:58:50.367313+00:00 pi4 [INFO] Capture-4-1001/MainThread indi.findCcd() [635]: Found 2 CCDs
2024-11-18T12:58:50.367080+00:00 pi4 [INFO] Capture-4-1001/MainThread indi._findCcds() [626]:  Detected LibCamera imx477-1
2024-11-18T12:58:50.366763+00:00 pi4 [INFO] Capture-4-1001/MainThread indi._findCcds() [626]:  Detected LibCamera imx477-0
2024-11-18T12:58:50.365931+00:00 pi4 [INFO] Capture-4-1001/MainThread indi._findCcds() [615]: Searching for available cameras
2024-11-18T12:58:45.415793+00:00 pi4 [INFO] Capture-4-1001/Dummy-3 indi.newDevice() [224]: new device LibCamera imx477-1
2024-11-18T12:58:45.401000+00:00 pi4 [INFO] Capture-4-1001/Dummy-3 indi.newDevice() [224]: new device moonlite_allsky
2024-11-18T12:58:45.381354+00:00 pi4 [INFO] Capture-4-1001/Dummy-3 indi.newDevice() [224]: new device LibCamera imx477-0
2024-11-18T12:58:45.377259+00:00 pi4 [INFO] Capture-4-1001/Dummy-3 indi.newDevice() [224]: new device AZ-GTi Alt-Az Wired
2024-11-18T12:58:45.362838+00:00 pi4 [INFO] Capture-4-1001/MainThread indi.serverConnected() [356]: Server connected (stellarmate.lan:7624)
2024-11-18T12:58:45.335389+00:00 pi4 [INFO] Capture-4-1001/MainThread capture._initialize() [609]: Connecting to indiserver
[/quote]

[... cut ...] camera and not a libcamera camera?

Do I need to set an 'INDI Camera Config' in the text box near the end of the Camera config tab?
Mine currently contains some default text:

[quote]
{
    "SWITCHES": {},
    "PROPERTIES": {},
    "TEXT": {}
}
[/quote]

The 'INDI custom config' webpage at https://github.com/aaronwmorris/indi-allsky/wiki/INDI-custom-config contains configs for various cameras but not for the Pi HQ camera.
Is there any documentation for the options available for the Pi HQ camera?

aaronwmorris, 2024-11-18:
Excellent!  You are the first person to attempt to use the `indi_libcamera_ccd` driver.  I just need the output of `indi_getprop` after you have tried to start indi-allsky.  I need the name of the gain property to integrate that with indi-allsky.

warwound, 2024-11-19:
Hmm I'm not sure where to get the 'indi_getprop' output.

I thought it'd be in the Pi4 (indi client) indi-allsky logs but that doesn't seem to be the case.
Running the command indi_getprop on the Pi4 didn't return anything relevant either.

However running the command indi_getprop on the Pi5 (indi server) seems to have produced some useful output.
The output of this command is attached.

[indi_getprop.txt](https://github.com/user-attachments/files/17811239/indi_getprop.txt)

aaronwmorris, 2024-11-19:
That is what I needed!  Merged #1652 with the support for setting the gain on the indi_libcamera_ccd server.

Optionally, I added a section to the wiki on how to disable the AWB mode:  https://github.com/aaronwmorris/indi-allsky/wiki/INDI-custom-config#disable-awb  I assume this will work since this is how you disable AWB on the CLI.

warwound, 2024-11-19:
Excellent support as always!

I updated my 'camera/indi.py' script, and added the 'disable AWB' custom config - all on my Pi4 running indi-allsky.
Rebooted and it successfully connected to indiserver on my Pi5 running Stellarmate.
indi-allsky webui shows 'Cameras Available: 2', and the drop down list above it has 2 options:

'LibCamera imx477-0'
This options seems to work fine.
This is my wide angle lens camera on Pi5 CSI port 0.

and 

'libcamera_imx477'
Seems not to work, always shows 'No Image for 15 minutes'

In Camera Config i then set 'INDI Camera Name' to 'LibCamera imx477-1' - it was previously blank - then reloaded the config.
indi-allsky webui now shows 'Cameras Available: 3', and the drop down list above it has 3 options:

'libcamera_imx477'
'LibCamera imx477-0'
'LibCamera imx477-1'
I selected 'LibCamera imx477-1' and there is an image from my telephoto lens camera on CSI port 1, seems to be working fine.

Thanks again for the excellent support, I'll close this issue.

warwound, 2024-12-12:
Can I reopen ths issue with a request for the Nikon DSLR driver to be supported?

I have my Nikon D5300 connected via indiserver to Kstars on my Pi5.
And docker runs indi-allsky connecting to a Pi HQ camera.
I tried to connect to indi-allsky to indiserver and my D5300, it eventually makes an initial connection then errors out.
I've attached [indi_getprop.txt](https://github.com/user-attachments/files/18114491/indi_getprop.txt) which I executed on my Pi 5.
Hoping this contains all the info needed to enable support for Nikon DSLR.

Thanks,

aaronwmorris, 2024-12-12:
The Nikon will probably work, but I have not had experience with it.  Make sure you clear out the INDI properties if you had another camera defined.  Changing cameras is a common cause of receiving timeouts when the config will not apply with the new camera.

Can you run this other script to show the descriptions of the properties?

```
source virtualenv/indi-allsky/bin/activate

./misc/camera_properties.py
```

warwound, 2024-12-13:
Ahh - I still had the INDO camera config for indi_libcamera_ccd in my config.
I replaced that with the default camera config and restarted.
At last my Nikon D5300 works with indi-allsky.

As for running the commands:

```
source virtualenv/indi-allsky/bin/activate
./misc/camera_properties.py
```

I can use the 'docker exec' command to get a bash terminal in my capture container but nowhere could I find the path  'virtualenv/indi-allsky/bin/activate'.
The source command always telling me it can't find the path.

Presumably that's unimportant as I now have my D5300 wokring?

Thanks for your help.

closed 2024-12-14 by warwound