aaronwmorris/indi-allsky #1651: Web site

Nasirius, 2024-11-18:
Hello!

is there a description somewhere of what exactly I need to set in order to access the indi allsky website from anywhere?

aaronwmorris, 2024-11-19:
There are a few options.

1. If your ISP allows, you could just setup a port forward using your router.
2. If you have a Virtual Private Server on the Internet, you could use the SyncAPI to upload images to a remote instance that could be viewed on the Internet.
3. Some people have used services like CloudFlare Tunnel to expose a private instance to the Internet.

Nasirius, 2024-11-19:
I'm too new to this unfortunately, is there any detailed description? I have an empty storage space where TJ Allsky was running, do I understand correctly that Indi Allsky won't work there?

Nasirius, 2024-11-19:
Is there a free website where syncApi works? My website doesn't support it....

1. If your ISP allows, you could just setup a port forward using your router. <------- Can anyone help with this?

aaronwmorris, 2024-11-19:
There are probably not any free places to run the server for the SyncAPI.  I really wanted to write a remote web site software in PHP for greater compatibility, however, a lot of the astrometric calculations used in indi-allsky come from python modules that just do not have equivalents in PHP.

As for port forwarding, you would probably have much better help by just googling "how to setup port forwarding" than I could give here.  There are just far too many router hardware, router software, and ISP options to give much help here.

Nasirius, 2024-11-24:
I chose another provider.
https://www.racknerd.com/

closed 2024-11-24 by Nasirius