aaronwmorris/indi-allsky #1657: Show wind units in knots

aaronwmorris, 2024-11-20:
Sorry to reopen this but I don't understand. Is there a way to show wind speed from wunderground in knots, for example?

_Originally posted by @ayryq in https://github.com/aaronwmorris/indi-allsky/issues/1404#issuecomment-2487017314_

aaronwmorris, 2024-11-20:
@ayryq I have merged #1658 with a wind speed conversion option along with the other unit selections on the Image config tab.

ayryq, 2024-11-20:
Camera config tab. And of course it's dead calm outside right now so I can't tell if it worked. Pretty cool that I can pull weather (via WUnderground) from my personal weather station.

fjungclaus, 2024-11-27:
@aaronwmorris 

Hi Aaron, 

on the camera tab I set Wind Speed Display to kph, but it still came out as m/s.

From my understanding the problem is, that within tempApiOpenWeatherMap.py and tempApiWeatherUnderground.py the comparison is done against 'kmh' instead of 'kph'.

`
diff --git a/indi_allsky/devices/sensors/tempApiOpenWeatherMap.py b/indi_allsky/devices/sensors/tempApiOpenWeatherMap.py
index ec4f5755..656e7ef2 100644
--- a/indi_allsky/devices/sensors/tempApiOpenWeatherMap.py
+++ b/indi_allsky/devices/sensors/tempApiOpenWeatherMap.py
@@ -239,7 +239,7 @@ class TempApiOpenWeatherMap(SensorBase):
         elif self.config.get('WINDSPEED_DISPLAY') == 'knots':
             current_wind_speed = self.mps2knots(wind_speed)
             current_wind_gust = self.mps2knots(wind_gust)
-        elif self.config.get('WINDSPEED_DISPLAY') == 'kmh':
+        elif self.config.get('WINDSPEED_DISPLAY') == 'kph':
             current_wind_speed = self.mps2kmph(wind_speed)
             current_wind_gust = self.mps2kmph(wind_gust)
         else:
diff --git a/indi_allsky/devices/sensors/tempApiWeatherUnderground.py b/indi_allsky/devices/sensors/tempApiWeatherUnderground.py
index 279e3a89..5decf874 100644
--- a/indi_allsky/devices/sensors/tempApiWeatherUnderground.py
+++ b/indi_allsky/devices/sensors/tempApiWeatherUnderground.py
@@ -221,7 +221,7 @@ class TempApiWeatherUnderground(SensorBase):
         elif self.config.get('WINDSPEED_DISPLAY') == 'knots':
             current_wind_speed = self.mps2knots(wind_speed)
             current_wind_gust = self.mps2knots(wind_gust)
-        elif self.config.get('WINDSPEED_DISPLAY') == 'kmh':
+        elif self.config.get('WINDSPEED_DISPLAY') == 'kph':
             current_wind_speed = self.mps2kmph(wind_speed)
             current_wind_gust = self.mps2kmph(wind_gust)
         else:
`

Thanks for all the work and energy you put into indi-allsky!
Regards, Frank

aaronwmorris, 2024-11-28:
@fjungclaus Thank you for catching this.  You can tell I could not decide between kmh and kph.  :-)  Merged #1684

closed 2024-12-17 by aaronwmorris