aaronwmorris/indi-allsky #1659: Web Only Install Not Flushing Expired Exposures

cfmorrell, 2024-11-20:
I'm running a split setup where my pi is local only and I've got a web only install that's running in AWS.  It seems that the web only instance is not cleaning up expired exposures.  Even when running ./misc/flush_images.py, the /var/www/html/allsky/images/camera/exposures folder remains untouched.  It seems like there may be a service that is include in the main install that is not included in the web only version.  I don't have this issue on my pi instance.  I was looking for a failure in a log file, but there's no indi-allsky.log file in /var/log/indi-allsky on the web only instance, just the webapp log file.

aaronwmorris, 2024-11-20:
By default, the flush images script will only flush images for camera ID 1.  If you have multiple camera entries, you will have to run it again for the additional cameras.

cfmorrell, 2024-11-20:
I've only got one camera configured.  I'll run it again to see, but the automatic cleanup isn't working either.

cfmorrell, 2024-11-20:
I see that the flush failed out saying that it was killed, but I didn't kill it.  I assume something is missing that is causing it to fail.  The camera listed is the correct one.

[image]

aaronwmorris, 2024-11-20:
Can you post your Support Info bundle?

cfmorrell, 2024-11-20:
Maybe I've got something bigger going on here.  The support bundle webpage doesn't populate anything and when I run the shell script, it errors out somewhere after lsusb.

`#################################
### indi-allsky support info  ###
#################################
```

Distribution: ubuntu
Release: 24.04
Arch: x86_64
Bits: 64

CPUs: 1
Memory: 465076 kB

System: Generic PC

Linux ip-172-31-72-85 6.8.0-1018-aws #20-Ubuntu SMP Thu Oct 10 18:14:42 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
Hostname
ip-172-31-72-85.ec2.internal
ip-172-31-72-85.ec2.internal

Time
Wed Nov 20 15:39:54 UTC 2024

System timezone
Etc/UTC

Uptime
 15:39:54 up 56 min,  2 users,  load average: 0.01, 1.66, 2.16

Memory
               total        used        free      shared  buff/cache   available
Mem:          465076      319016        6596        1560      170268      146060
Swap:              0           0           0

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root       29378688 18164036  11198268  62% /
tmpfs             232536        0    232536   0% /dev/shm
tmpfs              93016      896     92120   1% /run
tmpfs               5120        0      5120   0% /run/lock
/dev/xvda16       901520   135272    703120  17% /boot
/dev/xvda15       106832     6246    100586   6% /boot/efi
tmpfs              46504       16     46488   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info

system python: Python 3.12.3

indiserver: /usr/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute
       valid_lft forever preferred_lft forever
2: enX0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 9001 qdisc fq_codel state UP group default qlen 1000
    link/ether 12:88:08:26:9d:69 brd ff:ff:ff:ff:ff:ff
    inet 172.31.72.85/20 metric 100 brd 172.31.79.255 scope global dynamic enX0
       valid_lft 2009sec preferred_lft 2009sec
    inet6 fe80::1088:8ff:fe26:9d69/64 scope link
       valid_lft forever preferred_lft forever

User info
uid=1000(ubuntu) gid=1000(ubuntu) groups=1000(ubuntu),4(adm),20(dialout),24(cdrom),27(sudo),30(dip),105(lxd)

Process info
ubuntu      1660  0.4  3.9  35652 18176 ?        SNs  15:28   0:03 /home/ubuntu/indi-allsky/virtualenv/indi-allsky/bin/python3 /home/ubuntu/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
ubuntu      1672  0.8 24.9 684772 115904 ?       SNl  15:31   0:04 /home/ubuntu/indi-allsky/virtualenv/indi-allsky/bin/python3 /home/ubuntu/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi

Check for virtual sessions

USB info

```
###############
###  ERROR  ###
###############

The script exited abnormally, please try to run again...`

cfmorrell, 2024-11-20:
When I run those next few lines from the support script manually, I get the following:

`Check for virtual sessions

USB info

USB Permissions
find: ‘/dev/bus/usb’: No such file or directory

video device Permissions
ls: cannot access '/dev/video*': No such file or directory

v4l info
Command 'v4l2-ctl' not found, but can be installed with:
sudo apt install v4l-utils

Module info
Module                  Size  Used by
8021q                  45056  0
garp                   20480  1 8021q
mrp                    20480  1 8021q
stp                    12288  1 garp
llc                    16384  2 stp,garp
crct10dif_pclmul       12288  1
crc32_pclmul           12288  0
polyval_clmulni        12288  0
polyval_generic        12288  1 polyval_clmulni
ghash_clmulni_intel    16384  0
sha256_ssse3           32768  0
sha1_ssse3             32768  0
aesni_intel           356352  0
crypto_simd            16384  1 aesni_intel
cryptd                 24576  2 crypto_simd,ghash_clmulni_intel
psmouse               217088  0
binfmt_misc            24576  1
nls_iso8859_1          12288  1
input_leds             12288  0
serio_raw              20480  0
sch_fq_codel           24576  2
nf_conntrack          196608  0
nf_defrag_ipv6         24576  1 nf_conntrack
nf_defrag_ipv4         12288  1 nf_conntrack
libcrc32c              12288  1 nf_conntrack
dm_multipath           45056  0
msr                    12288  0
efi_pstore             12288  0
nfnetlink              20480  2
ip_tables              32768  0
x_tables               65536  1 ip_tables
autofs4                57344  2`

cfmorrell, 2024-11-20:
I turned on your command tracing option and found the following:

`+ echo 'USB info'
USB info
+ lsusb
++ catch_error
++ echo`

Apparently it doesn't like that lsusb returns nothing.

cfmorrell, 2024-11-20:
I commented out the lsusb line and got the following support bundle:

#################################
### indi-allsky support info  ###
#################################
```

Distribution: ubuntu
Release: 24.04
Arch: x86_64
Bits: 64

CPUs: 1
Memory: 465076 kB

System: Generic PC

Linux ip-172-31-72-85 6.8.0-1018-aws #20-Ubuntu SMP Thu Oct 10 18:14:42 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
Hostname
ip-172-31-72-85.ec2.internal
ip-172-31-72-85.ec2.internal 

Time
Wed Nov 20 15:51:57 UTC 2024

System timezone
Etc/UTC

Uptime
 15:51:57 up  1:08,  2 users,  load average: 10.17, 4.20, 2.46

Memory
               total        used        free      shared  buff/cache   available
Mem:          465076      331920       33744        1608      130260      133156
Swap:              0           0           0

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root       29378688 18170468  11191836  62% /
tmpfs             232536        0    232536   0% /dev/shm
tmpfs              93016      900     92116   1% /run
tmpfs               5120        0      5120   0% /run/lock
/dev/xvda16       901520   135272    703120  17% /boot
/dev/xvda15       106832     6246    100586   6% /boot/efi
tmpfs              46504       16     46488   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info

system python: Python 3.12.3

indiserver: /usr/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: enX0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 9001 qdisc fq_codel state UP group default qlen 1000
    link/ether 12:88:08:26:9d:69 brd ff:ff:ff:ff:ff:ff
    inet 172.31.72.85/20 metric 100 brd 172.31.79.255 scope global dynamic enX0
       valid_lft 3087sec preferred_lft 3087sec
    inet6 fe80::1088:8ff:fe26:9d69/64 scope link 
       valid_lft forever preferred_lft forever

User info
[... 874 lines cut ...]
    "CERT_BYPASS": true,
    "USERNAME": "",
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "ALT_DEG_MIN": 20.0,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,300 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": false,
    "DAYTIME_TRACK": false,
    "ALT_DEG_MIN": 20.0,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,200 (Right)\n# anchor:ra (Right Justified)\n# color:200,200,200\nSatellites"
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-11-20:
I will fix the support info script.  I was mainly interested in your memory, but the support info gives a lot of other info as well.

It sounds like the Linux kernel out of memory [OOM] is killing the script.  512MB is a limited environment, but I would still expect it to run.

cfmorrell, 2024-11-20:
I'm running it in an AWS t2.nano instance.  I'll bump the memory and see what happens.

cfmorrell, 2024-11-21:
I upgraded to a t2.micro instance which provide 1G of memory.  When I try to run the flush_images script, it pegs the CPU and becomes unresponsive until I reboot the VM.  I've tried monitoring memory externally, but don't have external visibility and lose the ability to see inside once everything is pegged.  CPU consumption stays at 100% until I intervene.  

[image]

aaronwmorris, 2024-11-21:
I just reviewed the code and I need to do some optimizations to the CLI flush script.  I had already applied the optimizations to the web based flush operations, but it looks like I forgot to update the flush script.

Try running the flush from the web interface, it should function a bit better.

cfmorrell, 2024-11-22:
It did work from the web interface.  I'll let it run for a while and see if it manages to keep things cleaned up automagically.

aaronwmorris, 2024-11-22:
Included with #1665 I optimized the CLI flush script for lower memory situations.

closed 2025-02-20 by aaronwmorris