aaronwmorris/indi-allsky #1668: No Image for 15 Minutes

chevyman142000, 2024-11-22:
Hello! Came across this the other day and decided to try and set it up with a Raspberry Pi 3B+ I had laying around after ordering a RPI HQ Camera IMX477. So far, I have not been able to get the camera to show up in the web interface. All I see on latest is No Image for 15 minutes. I just tried swapping in a Raspberry Pi 4 and having the same issue. I even tried re-imaging my sd card to run Bullseye to see if that would help and I'm getting the same No Image for 15 minutes. Any thoughts or suggestions on what I might be doing wrong?

Thanks!

PS. while grabbing the support info, I got a pop up that says: Date: 2024-11-22 21:17:25

Camera last ready 601s ago. Camera might be hung. Aborting exposure.

chevyman142000, 2024-11-22:
Here is the support info...Interesting that libcamera-hello says imx290...

/home/pi/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 11
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 3930356 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux allsky 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
Hostname
allsky
allsky 

Time
Fri 22 Nov 15:17:41 CST 2024

System timezone
US/Central

Uptime
 15:17:41 up 12 min,  2 users,  load average: 0.17, 0.16, 0.14

Memory
               total        used        free      shared  buff/cache   available
Mem:         3930356      507796     2721132       36780      701428     3311456
Swap:        1048572           0     1048572

Filesystems
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root      122615548 8327392 109250280   8% /
devtmpfs         1700952       0   1700952   0% /dev
tmpfs            1965176     168   1965008   1% /dev/shm
tmpfs             786072    1196    784876   1% /run
tmpfs               5120       4      5116   1% /run/lock
/dev/mmcblk0p1    261106   31380    229726  13% /boot
tmpfs             393032      20    393012   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
32128

system python: Python 3.9.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether d8:3a:dd:0b:ca:e6 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether d8:3a:dd:0b:ca:e8 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.123/24 brd 192.168.1.255 scope global dynamic noprefixroute wlan0
[... 1250 lines cut ...]
    "CERT_BYPASS": true,
    "USERNAME": "",
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,300 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": true,
    "DAYTIME_TRACK": true,
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,200 (Right)\n# anchor:ra (Right Justified)\n# color:200,200,200\nSatellites"
  }
}

```
#################################
###     end support info      ###
#################################

chevyman142000, 2024-11-22:
Apologies...I determined why the camera was showing up incorrectly. I had to update the /boot/config.txt to the correct dtoverlay=imx477. I did that, rebooted, but still seeing No Image for 15 minutes. Additionally, I've had the following 2 messages pop up:

Date: 2024-11-22 21:32:57

WARNING: Image worker was restarted more than 10 times

and 

Date: 2024-11-22 21:34:09

Image queue exceeded maximum threshold depth. System processing might be degraded.

Here is an updated support info output: 

/home/pi/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 11
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 3930356 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux allsky 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
Hostname
allsky
allsky 

Time
Fri 22 Nov 15:35:41 CST 2024

System timezone
US/Central

Uptime
 15:35:41 up 7 min,  2 users,  load average: 0.13, 0.28, 0.20

Memory
               total        used        free      shared  buff/cache   available
Mem:         3930356      604608     2024056       36788     1301692     3211732
Swap:        1048572           0     1048572

Filesystems
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root      122615548 8845032 108732640   8% /
devtmpfs         1700952       0   1700952   0% /dev
tmpfs            1965176     172   1965004   1% /dev/shm
tmpfs             786072    1200    784872   1% /run
tmpfs               5120       4      5116   1% /run/lock
/dev/mmcblk0p1    261106   31380    229726  13% /boot
tmpfs             393032      20    393012   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
34076

system python: Python 3.9.2

indiserver: /usr/local/bin/indiserver

[... 1263 lines cut ...]
    "CERT_BYPASS": true,
    "USERNAME": "",
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,300 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": true,
    "DAYTIME_TRACK": true,
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,200 (Right)\n# anchor:ra (Right Justified)\n# color:200,200,200\nSatellites"
  }
}

```
#################################
###     end support info      ###
#################################

chevyman142000, 2024-11-22:
Found this in the log and thought it might be helpful..especially the image work exception portion. 

Nov 22 15:52:50 allsky [INFO] Capture-1-1645/MainThread capture.saferun() [232]: Exposure state: READY
Nov 22 15:52:50 allsky [INFO] Capture-1-1645/MainThread capture.saferun() [231]: Camera last ready: 12.1s
Nov 22 15:52:48 allsky [INFO] MainProcess-1641/MainThread allsky._queueManualTasks() [1223]: Checking for manually submitted tasks
Nov 22 15:52:48 allsky [WARNING] MainProcess-1641/MainThread processing._load_detection_mask() [2791]: No detection mask defined
Nov 22 15:52:48 allsky [INFO] MainProcess-1641/MainThread allsky._startImageWorker() [396]: Starting Image-7 worker
Nov 22 15:52:48 allsky [ERROR] MainProcess-1641/MainThread allsky._startImageWorker() [389]: Image worker exception: 
Nov 22 15:52:48 allsky [ERROR] MainProcess-1641/MainThread allsky._startImageWorker() [389]: Image worker exception: Exception: *** rawpy module not available ***
Nov 22 15:52:48 allsky [ERROR] MainProcess-1641/MainThread allsky._startImageWorker() [389]: Image worker exception:     raise Exception('*** rawpy module not available ***')
Nov 22 15:52:48 allsky [ERROR] MainProcess-1641/MainThread allsky._startImageWorker() [389]: Image worker exception:   File "/home/pi/indi-allsky/indi_allsky/processing.py", line 447, in add
Nov 22 15:52:48 allsky [ERROR] MainProcess-1641/MainThread allsky._startImageWorker() [389]: Image worker exception:     i_ref = self.image_processor.add(filename_p, exposure, exp_date, exp_elapsed, camera)
Nov 22 15:52:48 allsky [ERROR] MainProcess-1641/MainThread allsky._startImageWorker() [389]: Image worker exception:   File "/home/pi/indi-allsky/indi_allsky/image.py", line 357, in processImage
Nov 22 15:52:48 allsky [ERROR] MainProcess-1641/MainThread allsky._startImageWorker() [389]: Image worker exception:     self.processImage(i_dict)
Nov 22 15:52:48 allsky [ERROR] MainProcess-1641/MainThread allsky._startImageWorker() [389]: Image worker exception:   File "/home/pi/indi-allsky/indi_allsky/image.py", line 257, in saferun
Nov 22 15:52:48 allsky [ERROR] MainProcess-1641/MainThread allsky._startImageWorker() [389]: Image worker exception:     self.saferun()
Nov 22 15:52:48 allsky [ERROR] MainProcess-1641/MainThread allsky._startImageWorker() [389]: Image worker exception:   File "/home/pi/indi-allsky/indi_allsky/image.py", line 228, in run
Nov 22 15:52:48 allsky [ERROR] MainProcess-1641/MainThread allsky._startImageWorker() [389]: Image worker exception: Traceback (most recent call last):
Nov 22 15:52:39 allsky [INFO] Capture-1-1645/MainThread indi.getTelescopeRaDec() [881]: Telescope Coord: RA 20.12, Dec 43.17
Nov 22 15:52:39 allsky [INFO] Capture-1-1645/MainThread capture.detectNight() [1382]: Moon altitude: -26.6, phase 51.5%
Nov 22 15:52:39 allsky [INFO] Capture-1-1645/MainThread capture.detectNight() [1375]: Sun altitude: 4.2
Nov 22 15:52:39 allsky [INFO] Capture-1-1645/MainThread capture.saferun() [232]: Exposure state: READY
Nov 22 15:52:39 allsky [INFO] Capture-1-1645/MainThread capture.saferun() [231]: Camera last ready: 1.0s
Nov 22 15:52:38 allsky [INFO] Capture-1-1645/MainThread capture.saferun() [464]: Exposure received in 1.6206 s (1.6106)
Nov 22 15:52:36 allsky [INFO] Capture-1-1645/MainThread capture.saferun() [581]: Total time since last exposure 15.0160 s
Nov 22 15:52:36 allsky [INFO] Capture-1-1645/MainThread capture.saferun() [550]: Image queue depth: 0
Nov 22 15:52:36 allsky [INFO] Capture-1-1645/MainThread libcamera.setCcdExposure() [263]: image command: libcamera-still --immediate --nopreview --camera 0 --raw --denoise off --gain 1 --shutter 9999 --metadata /tmp/tmpghtvqvev.json --metadata-format json --awbgains 1,1 --output /tmp/tmp6im2f_0d.dng

chevyman142000, 2024-11-22:
Last update for now...switch from raw/DNG to jpeg and now I'm capturing images (all green). Would like to resolve both the green and why it can't capture with raw/DNG.

Thanks!!

aaronwmorris, 2024-11-22:
I just fixed this issue this morning.  I had a problem with the python module requirements to decode the raw files.  Pull down the latest code and re-run `setup.sh` and it should resolve the problem for you.

As for the green bias, this is somewhat expected due to dealing with RAW data.  You can use one of the SCNR algorithms to remove the green bias.  You may also enable the AWB mode for libcamera, but this can have some bad side effects at night.

chevyman142000, 2024-11-23:
Thank you. I re-imaged with the latest version of Bookwork, re-installed and everything is working. Now time to configure all of the settings! Thanks for the help!

closed 2024-11-23 by chevyman142000