aaronwmorris/indi-allsky #1670: API key

Nasirius, 2024-11-24:
hello! does anyone have an idea why it is not accepting the API key?

[image]

Nasirius, 2024-11-24:
I changed the time zone, but it still doesn't match, what else can I do?
Lan
[image]
[image]

Remote:
[image]
[image]

Nasirius, 2024-11-24:
when I click on update time it doesn't do anything it just puts a "d" in red
[image]

aaronwmorris, 2024-11-24:
The timezone does not really matter.  I would suggest turning NTP back on for both systems.  If the systems times are set correctly, the only thing left is the API key being wrong.

Nasirius, 2024-11-24:
> Az időzóna nem igazán számít. Azt javaslom, hogy mindkét rendszeren kapcsolja vissza az NTP-t. Ha a rendszeridők helyesen vannak beállítva, akkor csak az API-kulcs rossz marad.

Where can I turn it on?

aaronwmorris, 2024-11-24:
On the command line, you can run the following:

```
sudo timedatectl set-ntp true
```

Nasirius, 2024-11-24:
It looks like time and the new API key solved the problem, thanks for the help! Now a new problem has arisen :) how can I share it with others without them seeing my IP address.

aaronwmorris, 2024-11-24:
Normally, people purchase their own domain they can assign a name to the the IP address.  You may also be able to use a dynamic DNS service to get a name assigned.

closed 2025-02-20 by aaronwmorris