aaronwmorris/indi-allsky #1671: Timelapses failed - nothing obvious in the logs

nicholasmparker, 2024-11-24:
Interestingly im testing out a new sensor on one allsky and the other is running as usual.  Neither one of them generated a timelapse last night.  When i re-run the timelapse it will run, but the video ends up being small in size(5 megs) and not playable

I've looked where i can think in the logs and didnt see anything obvious.  

Thank you

nicholasmparker, 2024-11-24:
/home/nicholasmparker/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 8245568 kB

System: Raspberry Pi 5 Model B Rev 1.0

Linux cthulu 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
Hostname
cthulu
cthulu.localdomain cthulu.siamese-godzilla.ts.net 

Time
Sun 24 Nov 11:34:30 CST 2024

System timezone
America/Chicago

Uptime
 11:34:30 up 48 min,  2 users,  load average: 0.00, 0.03, 0.08

Memory
               total        used        free      shared  buff/cache   available
Mem:         8245568     1424816     5760640       29984     1184000     6820752
Swap:         204784           0      204784

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             3947712        0   3947712   0% /dev
tmpfs             824560     6256    818304   1% /run
/dev/mmcblk0p2 491511928 20048780 446479672   5% /
tmpfs            4122784      944   4121840   1% /dev/shm
tmpfs               5120       48      5072   1% /run/lock
/dev/mmcblk0p1    522230    77102    445128  15% /boot/firmware
tmpfs             824544      176    824368   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
31450

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:6d:9d:66 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.247/24 brd 192.168.0.255 scope global dynamic noprefixroute eth0
       valid_lft 83484sec preferred_lft 83484sec
    inet6 fd5e:a67:947d:1d4d:b7cf:4038:a295:524a/64 scope global dynamic noprefixroute 
       valid_lft 1645sec preferred_lft 1645sec
    inet6 fe80::f4f2:f58f:c0cd:e393/64 scope link noprefixroute 
[... 1232 lines cut ...]
    "CERT_BYPASS": true,
    "USERNAME": "",
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,300 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": false,
    "DAYTIME_TRACK": false,
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,200 (Right)\n# anchor:ra (Right Justified)\n# color:200,200,200\nSatellites"
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-11-24:
Okay, that looks good.  Sorry, I deleted one of the support dumps, but I read it before I deleted it.

Okay, lets look at the log now.  Can you post the output of `grep Video /var/log/indi-allsky/indi-allsky.log`?

nicholasmparker, 2024-11-24:
[2024-11-24T00.log.zip](https://github.com/user-attachments/files/17894376/2024-11-24T00.log.zip)
Log attached.  thank you so much!

aaronwmorris, 2024-11-24:
Okay, that log cuts off right at the point where there is interesting info.  Let's try this:

```
zgrep Video /var/log/indi-allsky/indi-allsky.log*.gz
```

nicholasmparker, 2024-11-24:
[logs2.log.zip](https://github.com/user-attachments/files/17894423/logs2.log.zip)

aaronwmorris, 2024-11-24:
There is only one instance I found where the timelapse actually failed to generate:

```
/var/log/indi-allsky/indi-allsky.log-20241120.gz:2024-11-19T18:07:40.064809-06:00 beholder [INFO] Video-1-1123/MainThread timelapse.generate() [142]: FFmpeg command: ffmpeg -y -loglevel level+warning -r 50.00 -f image2 -i /tmp/tmpt9jq0nmg_timelapse/%05d.jpg -vcodec libx264 -b:v 10000k -pix_fmt yuv420p -movflags +faststart -level 3.1 /var/www/html/allsky/images/ccd_f0ba1b0f-56f5-4005-b7e8-13bba1d15036/timelapse/20241119/allsky-timelapse_ccd1_20241119_day.mp4
/var/log/indi-allsky/indi-allsky.log-20241120.gz:2024-11-19T18:11:51.009539-06:00 beholder [WARNING] Video-1-1123/MainThread video.sigterm_handler_worker() [129]: Caught TERM signal
/var/log/indi-allsky/indi-allsky.log-20241120.gz:2024-11-19T18:11:56.464696-06:00 beholder [INFO] Video-1-1123/MainThread timelapse.generate() [159]: FFMPEG ran for 256.3998 s
/var/log/indi-allsky/indi-allsky.log-20241120.gz:2024-11-19T18:11:56.464983-06:00 beholder [ERROR] Video-1-1123/MainThread timelapse.generate() [160]: FFMPEG failed to generate timelapse, return code: 255
/var/log/indi-allsky/indi-allsky.log-20241120.gz:2024-11-19T18:11:56.465062-06:00 beholder [ERROR] Video-1-1123/MainThread timelapse.generate() [161]: FFMPEG output: b'[swscaler @ 0x555633a54190] [swscaler @ 0x555633a61120] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x555633a54190] [swscaler @ 0x555633ae1220] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x555633a54190] [swscaler @ 0x555633b61320] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x555633a54190] [swscaler @ 0x555633be1420] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x555633a54190] [swscaler @ 0x555633c61520] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x555633a8e130] [swscaler @ 0x555633a54190] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x555633a8e130] [swscaler @ 0x555633ae5550] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x555633a8e130] [swscaler @ 0x555633be4920] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x555633a8e130] [swscaler @ 0x555633b66a40] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x555633a8e130] [swscaler @ 0x555633c66c40] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x555633a8e130] [swscaler @ 0x555633a54190] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x555633a8e130] [swscaler @ 0x555633ae5550] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x555633a8e130] [swscaler @ 0x555633be4920] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x555633a8e130] [swscaler @ 0x555633b66a40] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x555633a8e130] [swscaler @ 0x555633c66c40] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x555633a8e130] [swscaler @ 0x555633a54190] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x555633a8e130] [swscaler @ 0x555633ae5550] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x555633a8e130] [swscaler @ 0x555633be4920] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x555633a8e130] [swscaler @ 0x555633b66a40] [warning] deprecated pixel format used, make sure you did set range correctly\n[swscaler @ 0x555633a8e130] [swscaler @ 0x555633c66c40] [warning] deprecated pixel format used, make sure you did set range correctly\n[libx264 @ 0x5556339775e0] [warning] frame MB size (241x137) > level limit (3600)\n[libx264 @ 0x5556339775e0] [warning] DPB size (4 frames, 132068 mbs) > level limit (0 frames, 18000 mbs)\n[libx264 @ 0x5556339775e0] [warning] MB rate (1650850) > level limit (108000)\n'
/var/log/indi-allsky/indi-allsky.log-20241120.gz:2024-11-19T18:11:56.473461-06:00 beholder [ERROR] Video-1-1123/MainThread timelapse.generate() [165]: FFMPEG created broken video file, cleaning up
```

In this case, it looks like the process received a TERM signal which caused the ffmpeg process to terminate.  I cannot say why this happened, but maybe indi-allsky was restarted/reloaded in the middle of generating the timelapse.  Maybe the system was being rebooted.

nicholasmparker, 2024-11-24:
right - it looks like it completes. but then the link is to this:
[image]

https://github.com/user-attachments/assets/afb49ba0-caaa-4c6f-b03a-ce766e2e4571

aaronwmorris, 2024-11-24:
Try setting the FPS back to 25 to see if the situation improves.  Also, try multiple browsers to see if may be the browser does not like something about the codec.

aaronwmorris, 2024-11-24:
It occurred to me that maybe the resolution of the video is a problem with the `-level 3.1` option in the extra field.  Try removing that extra option.

nicholasmparker, 2024-11-25:
Interestingly I turned off the keogram wrapper and it worked.  I'll try to repto both states tomorrow

aaronwmorris, 2024-11-25:
The wrapped keogram process usually expands the size of the image, making it a square aspect ratio.  This might be causing something bad in ffmpeg.  It is difficult to say since we cannot predict how ffmpeg is going to deal with odd resolutions.

panik988, 2024-11-25:
Hello,
same problem here with piHQ cam and wrapped keogram

aaronwmorris, 2024-11-25:
@nicholasmparker @panik988 

I was able to simulator the failures on my side.  I think the problem is the resulting resolution is too big with the wrapped keogram with the HQ camera.  The image has to be expanded for the encircled keogram.

I was able to work around the issue by setting the `Pre-Scale Images` setting on the Timelapse to 75%.

panik988, 2024-11-25:
I'll try tonight and let you know

panik988, 2024-11-26:
@aaronwmorris also with pre-scale images at 75 doesn't work for me.
My setup is Raspberry Pi 4 8GB, 128GB filesystem, Raspberry picam HQ
Here my settings:
[image]

panik988, 2024-11-26:
@aaronwmorris those are the support file and the log.
The only error I found with zgrep Video /var/log/indi-allsky/indi-allsky.log*.gz is this:
2024-11-25T07:43:34.868481+01:00 raspberrypi [ERROR] Video-2-1114/MainThread video.generateKeogramStarTrails() [1471]: Not enough frames to generate star trails timelapse: 198

[support file.txt](https://github.com/user-attachments/files/17922227/support.file.txt)
[indi-allsky.zip](https://github.com/user-attachments/files/17922233/indi-allsky.zip)

aaronwmorris, 2024-11-26:
It looks like you restarted the program in the middle of pre-processing the images, however, that is not what is really interesting.  Based on my calculations, your system appears to be averaging 9s to pre-process each image.  Based on that estimate, it would require **8 hours** to pre-process all of the images.  This is obviously too much time.

I guess I did not fully consider the performance considerations of the resolution of the HQ camera and pre-processing the images.  I am going to have to refactor the pre-processor to pre-scale the image data before processing them.  Currently, the image is scaled after being processed.

I hope I can get the pre-preprocessing time per image to 2s (or less).  It would still take ~2 hours to pre-process 3200 images at 2s each, but that would be considerably better.

aaronwmorris, 2024-11-27:
Merged #1681 which will scale the image before processing.  I would recommend setting the pre-scale setting to 50% if you have a Raspberry Pi with a HQ camera.  This should significantly reduce the processing time.

panik988, 2024-11-28:
@aaronwmorris now is working :-)
Thank you for your support

wjschmitt, 2024-12-08:
My startrail timelapses are generating, but the regular timelapses are not.  On the latest version 2024.09.1.   I used to get them.  I think this changed recently, maybe with this new version.  I don't see a Pre-scale Image setting.  I'm also running a RPi4 with HQ.  But I run indi-allsky on a VM with indiserver on the Pi.   

[image]

aaronwmorris, 2024-12-08:
The pre-scale setting is in a slightly newer release, I just have not bumped the version number.  I would recommend upgrading anyway, there were a number of fixes I implemented that might correct the issue you are hitting.

wjschmitt, 2024-12-12:
> @wjschmitt The pre-scale setting is in a slightly newer release, I just have not bumped the version number. I would recommend upgrading anyway, there were a number of fixes I implemented that might correct the issue you are hitting.

OK, I did the upgrade and now no timelapses at all generate, even the startrails, which were generating before.

aaronwmorris, 2024-12-12:
```
[error] width not divisible by 2 (2219x1787)\n[vost#0:0/libx264 @ 0x62f05fba1340] [error] Error while opening encoder - maybe incorrect parameters such as bit_rate, rate, width or height.
```

This is caused by the FFMPEG output scaling being set to 50%.  In this case, ffmpeg is not smart enough to resolve this.  Because the 50% size is not exactly divisible by 2, it does not generate a file.

closed 2025-02-20 by aaronwmorris