aaronwmorris/indi-allsky #1675: Camera Images Not Displaying & Status Down

cgmiller76, 2024-11-25:
I updated some hardware on my system (added a POE Hat, USB GPS Module, & DHT22 sensor.  I had to run a few updates including running back through setup to enable GPIO.  Afterwards as I was getting ready to remount after everything tested out fine, I had an issue with the camera cable pulling free.  I corrected that, but now images aren't showing up and the Status is Down even after multiple reboots.  I have comfirmed via rpicam-hello that the camera is working - so not sure what has gone wrong.  Appreciate any help!

cgmiller76, 2024-11-25:
/home/pi/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 8245568 kB

System: Raspberry Pi 5 Model B Rev 1.0

Linux Astro-AllSky2-Pi 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
Hostname
Astro-AllSky2-Pi
Astro-AllSky2-Pi.attlocal.net Astro-AllSky2-Pi.attlocal.net unknown2ccf67101e0e.attlocal.net unknown2ccf67101e0e.attlocal.net Astro-AllSky2-Pi.attlocal.net Astro-AllSky2-Pi.attlocal.net 

Time
Sun Nov 24 08:15:17 PM EST 2024

System timezone
US/Eastern

Uptime
 20:15:17 up  1:09,  2 users,  load average: 0.26, 0.26, 0.27

Memory
               total        used        free      shared  buff/cache   available
Mem:         8245568     1156128     4692640       87616     2591552     7089440
Swap:         524272           0      524272

Filesystems
Filesystem      1K-blocks       Used Available Use% Mounted on
udev              3947712          0   3947712   0% /dev
tmpfs              824560       6384    818176   1% /run
/dev/nvme0n1p2 1921205552 1735022996  88516864  96% /
tmpfs             4122784       1120   4121664   1% /dev/shm
tmpfs                5120         48      5072   1% /run/lock
tmpfs               76800         32     76768   1% /home/pi/allsky/tmp
/dev/nvme0n1p1     523248      77936    445312  15% /boot/firmware
tmpfs              824544        176    824368   1% /run/user/1000

sysctl info
vm.swappiness = 1

Thermal info
cpu-thermal
34750

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:10:1e:0e brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.122/24 brd 192.168.1.255 scope global dynamic noprefixroute eth0
       valid_lft 82231sec preferred_lft 82231sec
    inet6 2600:1700:7b11:e430::1c/128 scope global dynamic noprefixroute 
       valid_lft 2796sec preferred_lft 2796sec
[... 1312 lines cut ...]
    "CERT_BYPASS": true,
    "USERNAME": "",
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,300 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": false,
    "DAYTIME_TRACK": false,
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,200 (Right)\n# anchor:ra (Right Justified)\n# color:200,200,200\nSatellites"
  }
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-11-25:
Check the log at `/var/log/indi-allsky/indi-allsky.log` and see if there are any exceptions occurring.

cgmiller76, 2024-11-25:
2024-11-24T21:53:39.712226-05:00 Astro-AllSky2-Pi [INFO] MainProcess-6722/MainThread allsky._queueManualTasks() [1223]: Checking for manually submitted tasks
2024-11-24T21:53:50.545459-05:00 Astro-AllSky2-Pi [ERROR] Sensor-1-6732/MainThread sensor.update_sensors() [433]: SensorReadException: DHT sensor not found, check wiring
2024-11-24T21:53:52.714869-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: Traceback (most recent call last):
2024-11-24T21:53:52.715678-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/capture.py", line 177, in run
2024-11-24T21:53:52.715942-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:     self.saferun()
2024-11-24T21:53:52.716070-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/capture.py", line 189, in saferun
2024-11-24T21:53:52.716167-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:     self._initialize()
2024-11-24T21:53:52.716242-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/capture.py", line 683, in _initialize
2024-11-24T21:53:52.716326-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:     self.indiclient.configureGpsDevice(gps_config)
2024-11-24T21:53:52.716387-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 785, in configureGpsDevice
2024-11-24T21:53:52.716448-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:     self.configureDevice(self.gps_device, *args, **kwargs)
2024-11-24T21:53:52.716690-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 744, in configureDevice
2024-11-24T21:53:52.717063-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:     self.set_number(device, k, v)
2024-11-24T21:53:52.717098-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 1353, in set_number
2024-11-24T21:53:52.717131-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:     c = self.get_control(device, name, 'number')
2024-11-24T21:53:52.717156-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-11-24T21:53:52.717184-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:   File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 1344, in get_control
2024-11-24T21:53:52.717216-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:     raise TimeOutException('Timeout finding control {0}'.format(name))
2024-11-24T21:53:52.717250-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: indi_allsky.exceptions.TimeOutException: Timeout finding control GPS_REFRESH_PERIOD
2024-11-24T21:53:52.717280-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: 
2024-11-24T21:53:52.717309-05:00 Astro-AllSky2-Pi [INFO] MainProcess-6722/MainThread allsky._startCaptureWorker() [341]: Starting Capture-301 worker
2024-11-24T21:53:52.725028-05:00 Astro-AllSky2-Pi [INFO] MainProcess-6722/MainThread allsky._queueManualTasks() [1223]: Checking for manually submitted tasks
2024-11-24T21:53:52.730227-05:00 Astro-AllSky2-Pi [INFO] Capture-301-8549/MainThread indi.__init__() [129]: creating an instance of IndiClient
2024-11-24T21:53:52.730582-05:00 Astro-AllSky2-Pi [INFO] Capture-301-8549/MainThread indi.__init__() [136]: PyIndi version: 2.1.0
2024-11-24T21:53:52.731335-05:00 Astro-AllSky2-Pi [INFO] Capture-301-8549/MainThread capture._initialize() [609]: Connecting to indiserver
2024-11-24T21:53:52.732326-05:00 Astro-AllSky2-Pi [INFO] Capture-301-8549/MainThread indi.serverConnected() [356]: Server connected (localhost:7624)
2024-11-24T21:53:52.733208-05:00 Astro-AllSky2-Pi [INFO] Capture-301-8549/Dummy-3 indi.newDevice() [224]: new device GPS NMEA
2024-11-24T21:53:52.735939-05:00 Astro-AllSky2-Pi [INFO] Capture-301-8549/Dummy-3 indi.newDevice() [224]: new device Telescope Simulator
2024-11-24T21:53:52.736578-05:00 Astro-AllSky2-Pi [WARNING] MainProcess-6722/MainThread allsky._periodic_tasks() [1330]: Periodic tasks triggered
2024-11-24T21:53:52.736676-05:00 Astro-AllSky2-Pi [INFO] MainProcess-6722/MainThread allsky._periodic_tasks() [1345]: Creating aurora update task
[... cut ...]
2024-11-24T21:54:51.955275-05:00 Astro-AllSky2-Pi [ERROR] Sensor-1-6732/MainThread sensor.update_sensors() [433]: SensorReadException: DHT sensor not found, check wiring
2024-11-24T21:54:52.883767-05:00 Astro-AllSky2-Pi [INFO] Capture-303-8576/Dummy-3 indi.newMessage() [353]: new Message 2024-11-25T02:54:52: [ERROR] Failed to connect to 192.168.1.1@50000: No route to host. 
2024-11-24T21:54:57.791000-05:00 Astro-AllSky2-Pi [INFO] MainProcess-6722/MainThread allsky._queueManualTasks() [1223]: Checking for manually submitted tasks

cgmiller76, 2024-11-25:
So essentially looks like this recurring set of errors:
2024-11-24T21:54:05.898876-05:00 Astro-AllSky2-Pi [ERROR] Sensor-1-6732/MainThread sensor.update_sensors() [433]: SensorReadException: DHT sensor not found, check wiring
2024-11-24T21:54:18.747351-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: Traceback (most recent call last):
2024-11-24T21:54:18.747649-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: File "/home/pi/indi-allsky/indi_allsky/capture.py", line 177, in run
2024-11-24T21:54:18.747695-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: self.saferun()
2024-11-24T21:54:18.747882-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: File "/home/pi/indi-allsky/indi_allsky/capture.py", line 189, in saferun
2024-11-24T21:54:18.747984-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: self._initialize()
2024-11-24T21:54:18.748072-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: File "/home/pi/indi-allsky/indi_allsky/capture.py", line 683, in _initialize
2024-11-24T21:54:18.748174-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: self.indiclient.configureGpsDevice(gps_config)
2024-11-24T21:54:18.748525-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 785, in configureGpsDevice
2024-11-24T21:54:18.748610-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: self.configureDevice(self.gps_device, *args, **kwargs)
2024-11-24T21:54:18.748651-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 744, in configureDevice
2024-11-24T21:54:18.748684-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: self.set_number(device, k, v)
2024-11-24T21:54:18.748720-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 1353, in set_number
2024-11-24T21:54:18.748750-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: c = self.get_control(device, name, 'number')
2024-11-24T21:54:18.748778-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-11-24T21:54:18.748804-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 1344, in get_control
2024-11-24T21:54:18.748830-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: raise TimeOutException('Timeout finding control {0}'.format(name))
2024-11-24T21:54:18.748856-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception: indi_allsky.exceptions.TimeOutException: Timeout finding control GPS_REFRESH_PERIOD
2024-11-24T21:54:18.748889-05:00 Astro-AllSky2-Pi [ERROR] MainProcess-6722/MainThread allsky._startCaptureWorker() [334]: Capture worker exception:

aaronwmorris, 2024-11-25:
The problem appears to be the GPS module.  Did you setup GPSd on this system?

The following script can kickstart this process...

```
./misc/setup_gpsd.sh
```

cgmiller76, 2024-11-25:
I had installed that.  I just ran it again - along with the instructions from the Wiki:

####################################################
### Welcome to the indi-allsky GPSD setup script ###
####################################################

This script sets up GPSD

Distribution: debian
Release: 12
Arch: aarch64

Serial port: /dev/ttyACM0

Setup proceeding in 10 seconds... (control-c to cancel)

**** Installing packages... ****
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Get:3 http://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]                
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease                                     
Fetched 55.4 kB in 1s (82.2 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
telnet is already the newest version (0.17+2.4-2+deb12u1).
gpsd is already the newest version (3.22-4.1+b1).
gpsd-tools is already the newest version (3.22-4.1+b1).
gpsd-clients is already the newest version (3.22-4.1+b1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
**** Ensure user is a member of the dialout group ****

Use "cgps" to test your GPS adapter

GPSD is now installed... enjoy

cgmiller76, 2024-11-25:
sudo apt-get install ntpsec
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
ntpsec is already the newest version (1.2.2+dfsg1-1+deb12u1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

cgmiller76, 2024-11-25:
I figured out the issue - I had turned GPS Enable ON.  I have this system inside right now, so when I run cgps it isn't getting a fix.  I turned GPS Enable OFF, and it's back working as expected.  I didn't realize it would cause the image capture to fail if it couldn't get a fix.  Thanks for taking the time to troubleshoot with me!

closed 2024-12-17 by aaronwmorris