aaronwmorris/indi-allsky #1686: SYNCAPI/Passive Not working

jshall77, 2024-11-28:
Hello,

I am attempting to set up secondary sites to mirror my RPI's running Indi-Allsky, however it appears that the scripts used for creating the new user and API key don't work or need more dependencies than are already installed.  Is there another way that the SYNCAPI feature can be implemented?  I even tried setting up a "standard" installation of indi-allsky that mirrored my current config, other than the secondary is running on an Ubuntu VM vs RPI, and still didn't have any luck.

aaronwmorris, 2024-11-29:
Did you have the python virtualenv activated when running the user management commands?

```
source virtualenv/indi-allsky/bin/activate

./misc/usertool.py
```

jshall77, 2024-11-29:
Well, on Ubuntu 22.04 I can't even get that far.  Running the "web_only_setup" script generates the attached error.

Running it on RPI generates the second error attachment.
[image]
[image]

aaronwmorris, 2024-11-30:
This usually happens when flask is installed outside of the virtualenv or maybe in the user home directories.

jshall77, 2024-11-30:
So I abandoned the "web_only_setup" and just installed as normal, selecting "Indi-Passive" as my camera config, then ran the API script and that worked fine.  I was able to set up 2 API-Sync VMs to mirror my current Allsky devices.  However, I do still have a couple questions:
Does this sync all previous data? ie. images, timelapses, etc. or just what is there from the point of configuring the Sync?
If I change a retention setting on the secondary instance, does this also change on the primary or am I able to have separate retention settings?

Thanks as always for your awesome work!
Jeff

aaronwmorris, 2024-11-30:
The retention settings are separate on both sides.  However, because the image expiration occurs in the capture process which does not run on the remote side, you will have to manually expire images using the `./misc/expire_images.py` script.

The previous images and videos are not synced automatically, but you should be able to manually sync them using the `./misc/upload_sync.py` script.

closed 2025-02-20 by aaronwmorris