aaronwmorris/indi-allsky #1688: error on ./setup.sh

graememilne, 2024-11-30:
[image]
[image]

Followed the instructions as per page 
then ./setup.sh
then did this as required ,BUILD_INDI_CORE=true BUILD_INDI_3RDPARTY=false ./misc/build_indi.sh
once that was done reverted to 
./setup.sh

(i installed on a rpi4b 2gb which installed ok after a couple of re-installs. even installed a bme280 sensor,working.
Had the indi-allsky software up and running)

This installation is on a rpi4b 4gb model with errors.
Could it be that rawpy is out of date ??

how can this be sorted please.

aaronwmorris, 2024-11-30:
This is some sort of basic git issue.  I would try to fully upgrade the packages on the system to see if it helps.

```
sudo apt-get update

sudo apt-get dist-upgrade

reboot
```

aaronwmorris, 2024-11-30:
Can you post the output of `./misc/support_info.sh` ?

graememilne, 2024-11-30:
Is there a way to save it to a file for posting ,please

aaronwmorris, 2024-11-30:
You can just post the text without an attachment.

graememilne, 2024-11-30:
./misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 3882912 kB

System: Raspberry Pi 4 Model B Rev 1.5

Linux raspberrypi 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
Hostname
raspberrypi

Time
Sat 30 Nov 18:28:32 GMT 2024

System timezone
Europe/London

Uptime
 18:28:32 up 51 min,  3 users,  load average: 0.00, 0.01, 0.00

Memory
               total        used        free      shared  buff/cache   available
Mem:         3882912      398820     2653256       23080      931392     3484092
Swap:         524284           0      524284

Filesystems
Filesystem     1K-blocks    Used Available Use% Mounted on
udev             1670400       0   1670400   0% /dev
tmpfs             388292    1324    386968   1% /run
/dev/mmcblk0p2 122322964 8933832 107155812   8% /
tmpfs            1941456     216   1941240   1% /dev/shm
tmpfs               5120      16      5104   1% /run/lock
/dev/mmcblk0p1    522230   68386    453844  14% /boot/firmware
tmpfs             388288      48    388240   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
29206

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

IP Info
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 2c:cf:67:21:72:f2 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:21:72:f3 brd ff:ff:ff:ff:ff:ff
    inet 192.168.8.117/24 brd 192.168.8.255 scope global dynamic noprefixroute wlan0
       valid_lft 83330sec preferred_lft 83330sec
    inet6 fe80::9a6f:86e5:26d9:3995/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
[... 480 lines cut ...]
WTForms==3.2.1
zarr==2.18.3
zipp==3.21.0
```

indi-allsky config (passwords redacted)
Traceback (most recent call last):
  File "/home/allsky/indi-allsky/config.py", line 6, in <module>
    from indi_allsky.config import IndiAllSkyConfigUtil
  File "/home/allsky/indi-allsky/indi_allsky/config.py", line 29, in <module>
    app = create_app()
          ^^^^^^^^^^^^
  File "/home/allsky/indi-allsky/indi_allsky/flask/__init__.py", line 90, in create_app
    app.config.from_file(flask_config, load=json.load)
  File "/home/allsky/indi-allsky/virtualenv/indi-allsky/lib/python3.11/site-packages/flask/config.py", line 293, in from_file
    with open(filename, "r" if text else "rb") as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] Unable to load configuration file (No such file or directory): '/etc/indi-allsky/flask.json'

```
###############
###  ERROR  ###
###############

The script exited abnormally, please try to run again...

aaronwmorris, 2024-11-30:
(The error at the end of the support info was expected since the setup has not fully completed)

This is a really weird error.  I tested locally using several iterations and I am not having a problem.  Do you have the ability to test with another SDcard?

graememilne, 2024-11-30:
i found a spare 32gb sd card just now,will try it now

I was using a sandisk 128gb sd card

graememilne, 2024-11-30:
[image]

This error appeared near the end of ./setup.sh

aaronwmorris, 2024-11-30:
I think this is an issue with the python wheel mirror or maybe issues with your internet.  I would suggest rebooting your internet router.  If that does not help, you may try editing the `/etc/resolv.conf` file of your Pi and setting 8.8.8.8 as your DNS server.

graememilne, 2024-11-30:
Hi Aaron
I tried the above but had no joy.
Googled the problem and had a look at https://stackoverflow.com/questions/21277806/fatal-early-eof-fatal-index-pack-failed.
Tried this, git config --global core.compression 0
Did not work.
Then tried this,git config --global core.compression 9
It solved the issue

[image]

[image]

Aaron ,you were right about the internet issue .
It must have been timing out before file downloaded for what ever reason.
I wish i had contacted you earlier today as it would have saved me a lot of time.
I am now going to connect the BME280 sensor again and check the charts.
Hopefully put it outside on the tripod and test tomorrow evening.
I think this software is brilliant and can i say thank you very much for helping me resolve this irritating issue.

aaronwmorris, 2024-11-30:
Fantastic!  I have seen similar problems with my home internet, so I completely understand what you are going through.

graememilne, 2024-11-30:
[image]

[image]

BME280 up and running 
Will swap over the 32gb sd card for the 128gb card tomorrow.
Then put all the hardware into the enclosure again.

[image]

graememilne, 2024-11-30:
[image]
[image]
[image]

graememilne, 2024-11-30:
Hi Aaron
Would it be possible to have on the web page the sunrise sunset times bottom left of page.
sunrise 0823 sunset 1531 
The above is an example for the 1st of December 2024.
If too much coding involved ,no worries as it was just a suggestion.

graememilne, 2024-11-30:
Excellent,thank you very much.

aaronwmorris, 2024-12-01:
Merged #1690 which has the sunrise, sunset, moon-rise, and moon-set variables.

The wikis are updated with the new variables:
https://github.com/aaronwmorris/indi-allsky/wiki/Web-Status-Configuration
https://github.com/aaronwmorris/indi-allsky/wiki/Image-Labels

graememilne, 2024-12-01:
[image]
Hi Aaron
Thank you very much for the help you have given me.
I was gob smacked when i seen an email from you,to say you had implemented the request for sunrise/sunset times.
I have now assembled the hardware into the enclosure and now outside ready to test overnight.
I realise that there is still some tweeking of the settings to be done .
Meantime i will find my way round the software and make adjustments when required.
Looking forward to seeing results from overnight captures.
Thank you once again Aaron.

aaronwmorris, 2024-12-01:
I have been thinking about adding these variables for some time, and it is an obvious improvement without too much work.  I just have to be careful with the fact that at certain latitudes, the sun does not rise/set.  That is why the times are a string.  A blank value is returned for those situations.

graememilne, 2024-12-03:
[image]

Hi Aaron
Wondering if this is correct for sunrise,sunset ??
Would value your input.

aaronwmorris, 2024-12-03:
Given the Latitude 57.2 and Longitude -2.1 and the UTC timezone, the online calculators agree with the sunrise and sunset times within a few minutes.  I have the atmospheric refraction calculations disabled which is why the times differ by a few minutes from other sources.

closed 2025-02-20 by aaronwmorris