aaronwmorris/indi-allsky #1696: Grey / Blank Images Returned from Camera

StarGeezerPhil, 2024-12-04:
Had this issue intermittently before and a restart seemed to fix it - @aaronwmorris at the time I believe we both put this down to minimum exposure being too low?

But, now I have minimum exposures set (0.0005) and getting the same result.

You can see below the ** ADU average: 224.8 ** is completely wrong - simply wouldn't happen with an exposure like this.

For context, I took down my camera this morning to give it a clean, update and fresh darks.

It updated successfully, then I flushed my darks and ran the tempsigmaclip method - but it failed. 
I then restarted indi-allsky to take a look and all images are coming back like this.

I've power cycled and sudo reboot-ed the device repeatedly with the same result.

Raw export (compressed to JPG so I can upload here):
[image]

Log:

```
Dec  4 11:10:42 SkyCam [WARNING] Image-3-1190/MainThread processing.rotate_angle() [1211]: Rotation in 0.5647 s
Dec  4 11:10:42 SkyCam [WARNING] Image-3-1190/MainThread image.recalculate_exposure() [1744]: New calculated exposure: 0.00500000
Dec  4 11:10:41 SkyCam [INFO] Image-3-1190/MainThread processing._convert_16bit_to_8bit() [1156]: Resampling image from 16 to 8 bits
Dec  4 11:10:41 SkyCam [INFO] Image-3-1190/MainThread mode2_mtf.stretch() [70]: Stretch in 1.1686 s
Dec  4 11:10:41 SkyCam [INFO] MainProcess-901/MainThread allsky._queueManualTasks() [1223]: Checking for manually submitted tasks
Dec  4 11:10:40 SkyCam [INFO] Image-3-1190/MainThread processing._calculate_8bit_adu() [955]: ADU average: 224.8 (112)
Dec  4 11:10:40 SkyCam [INFO] Image-3-1190/MainThread image.processImage() [412]: Image: 4056 x 3040
Dec  4 11:10:40 SkyCam [WARNING] Image-3-1190/MainThread processing.debayer() [1071]: Overriding CFA pattern: BGGR
Dec  4 11:10:40 SkyCam [INFO] Image-3-1190/MainThread sqm.calculate() [42]: Weighted SQM average: 2647.40
Dec  4 11:10:40 SkyCam [INFO] Image-3-1190/MainThread sqm.calculate() [37]: Raw SQM average: 224.74
Dec  4 11:10:40 SkyCam [WARNING] Image-3-1190/MainThread sqm._generateSqmMask() [63]: Using central ROI for SQM calculations
Dec  4 11:10:40 SkyCam [INFO] Image-3-1190/MainThread sqm._generateSqmMask() [48]: Generating mask based on SQM_ROI
Dec  4 11:10:40 SkyCam [INFO] Image-3-1190/MainThread sqm.calculate() [21]: Exposure: 0.000100, gain: 1
Dec  4 11:10:40 SkyCam [INFO] Image-3-1190/MainThread processing._calibrate() [739]: Black level: 4096
Dec  4 11:10:40 SkyCam [WARNING] Image-3-1190/MainThread processing._apply_calibration() [846]: Dark not found: ccd1 16bit 0.0001000s gain 1 bin 1 -273.15c
Dec  4 11:10:40 SkyCam [WARNING] Image-3-1190/MainThread processing._apply_calibration() [826]: Temperature matched dark not found: -273.15c
Dec  4 11:10:40 SkyCam [INFO] Image-3-1190/MainThread processing._apply_calibration() [807]: Searching for dark frame: gain 1, exposure >= 0.0, temp >= -273.1c
Dec  4 11:10:40 SkyCam [WARNING] Image-3-1190/MainThread processing._detectBitDepth() [701]: Updated default bit depth: 9
Dec  4 11:10:40 SkyCam [INFO] Image-3-1190/MainThread processing._detectBitDepth() [662]: Image max value: 281
Dec  4 11:10:40 SkyCam [INFO] Image-3-1190/MainThread processing.add() [532]: Image bits: 16, cfa: BGGR
Dec  4 11:10:39 SkyCam [ERROR] Image-3-1190/MainThread processing.get_astrometric_data() [1914]: HST next pass error: that satellite seems to stay always below your horizon
Dec  4 11:10:39 SkyCam [INFO] Capture-3-1188/MainThread capture.saferun() [464]: Exposure received in 1.6941 s (1.6940)
Dec  4 11:10:37 SkyCam [INFO] Capture-3-1188/MainThread capture.saferun() [581]: Total time since last exposure 0.1641 s
Dec  4 11:10:37 SkyCam [INFO] Capture-3-1188/MainThread capture.saferun() [550]: Image queue depth: 0
Dec  4 11:10:37 SkyCam [INFO] Capture-3-1188/MainThread libcamera.setCcdExposure() [263]: image command: libcamera-still --immediate --nopreview --camera 0 --raw --denoise off --gain 1 --shutter 100 --metadata /tmp/tmp2uhceuwa.json --metadata-format json --awbgains 1,1 --output /tmp/tmp_6aljkh_.dng
Dec  4 11:10:37 SkyCam [INFO] Capture-3-1188/MainThread capture.shoot() [1576]: Taking 0.00010001 s exposure (gain 1)
```

aaronwmorris, 2024-12-04:
The grey image issue is something that can be caused by the data scaling when the image data is converted to 8-bit data.  The challenge is when indi-allsky ingests RAW data, it is not known what scale the data is presented.

Lets say you have a 12-bit camera.  Some cameras will output the 12-bit data in a 12-bit space where the max value is 4095.  Other cameras/drivers will upscale the 12-bit data to the full 16-bit space where the max value is 65535.  indi-allsky tries to auto-detect the space by sampling the maximum value in the image.  At very low exposures, with these low noise cameras, there is a very low ceiling.  This can trick indi-allsky to think 12-bit data is actually 10-bits, so when the data is bit-shifted to 8-bits for the image export, this will result in a grey image.  indi-allsky thinks the exposure is at the expected level because the average brightness is at the correct point.

The fix for this is to explicitly set the bit depth of the image on the Image tab in the config.  You will have to review the log to see what the proper setting is for your camera.

StarGeezerPhil, 2024-12-04:
Brilliant, thank you @aaronwmorris I'll take a look and get this fixed.

Any ideas re: temp sensor seeming to disappear?

aaronwmorris, 2024-12-04:
For the temperature, I would suggest trying to update your libcamera system packages.  An update might have some fixes.

```
sudo apt-get update

sudo apt-get dist-upgrade
```

closed 2025-02-20 by aaronwmorris