aaronwmorris/indi-allsky #1701: Feature suggestion:  Image stretching 'simulator'

zaphus, 2024-12-05:
This is a suggestion to help people get the best image stretching settings for their setup:  Add a page that lets you simulate the stretching settings on a previously saved Raw image.
Currently the only way to see what stretching does (that I am aware of) is:  Wait for a clear night, when it is dark enough and the milky way (or stars) are visible.  Change the settings, apply them, switch to the live view and wait for more images. Switch back to settings, change the numbers, repeat.

I would much rather do this during the day with a saved raw image and a simpler set of steps on a single screen.  If I have captured a raw image, theoretically I could try stretching that with the settings and play around until I found something that stretched the milky way nicely.  If it was fast enough it could even be done with sliders, but even a simple button to apply would let me play with the settings quickly to find what works in my circumstances.
This could be extended to more than just stretching - so an alternative would be to have a "Simulation Camera" you could switch to, which uses a raw image that you have previously captured (or a sequence in a loop) then you can try during the day with that and also see how other settings impact the view - although the current interface would still be clunky to use since there is no sample image displayed along with the settings, adjusting still requires a lot of button presses to tweak, view, tweak again, etc

aaronwmorris, 2024-12-05:
I am already ahead of you.  :-D  There is a feature you can use under tools.  It is the FITS Processor.  You can test image stretching, stacking, and levels adjustments in real-time.

If you want to see how well it works at night, you still need to wait for a clear night to save some FITS files.  You can enable saving FITS under the Image tab let it run a few minutes and then turn it off.

If you navigate to the FITS Processor, it just loads the last available FITS file.  You can also select individual files in the Image Viewer.  I am not totally happy with the interface for selecting the FITS files, but I am working on it.

zaphus, 2024-12-07:
Well, there you go!
That is perfect, now I just need some clear sky to grab some samples and I can do everything I need - apologies for not seeing that in the first place!

closed 2024-12-07 by zaphus

aaronwmorris, 2024-12-07:
Its no big deal.  It is an unusual feature.

aaronwmorris, 2024-12-08:
For what it is worth, I have merged #1705 which has a new FITS Image viewer which should make it easier to navigate the FITS files that are saved.