aaronwmorris/indi-allsky #1708: Mounting NFS Share from Synology Nas

haightjr, 2024-12-10:
I want to archive a year's worth of images, videos, and other media to an NFS share on my local Synology NAS.

Currently, I have INDI AllSky Cam running on a Raspberry Pi 4 and an INDI AllSky Web instance deployed in Proxmox. My goal is to store all media directly on the INDI AllSky Web instance (up to 1 year) while minimizing SD card usage on the Pi.

Would NFS offload be a viable option? Alternatively, can I simply mount the NFS share to the INDI AllSky Web OS (Ubuntu Server 24) and redirect the images folder within its configuration?

Thank you!

aaronwmorris, 2024-12-10:
It should be able to work with NFS.  When you change the Image Folder setting, you will also need to re-run the web_only_setup.sh script again to update the web server config to allow image access.

haightjr, 2024-12-10:
That works perfectly, thank you! I successfully mounted the NFS share to the OS and updated the images folder. It appears that images are being transferred from Indi-AllSky-Cam to Indi-AllSky-Web and then saved to the NFS share. Do I need to configure anything else on Indi-AllSky-Cam to ensure its storage doesn’t fill up?

If I understand correctly, with SyncAPI configured, images taken by Indi-AllSky-Cam are transferred to Indi-AllSky-Web, which then saves and retrieves them from the NFS share. Where should I set the retention policy for 365 days—on Indi-AllSky-Web?  Thank you.

haightjr, 2024-12-11:
The 64GB storage on the camera filled up overnight, so I need to check the sync settings. My ultimate goal is to have the photos from the camera saved directly to the NAS share, accessible by the web server, and not stored on the SD card.

aaronwmorris, 2024-12-11:
There is technically no retention policy on the remote instance.  The expiration of images and videos run within the capture process and that does not operate on the remote server.  There is a script to manually expire images:  `./misc/expire_images.py`

haightjr, 2024-12-14:
I’m not sure why the 64GB SD card filled up within the first 24 hours. After clearing out old media files on the Pi, the issue hasn’t reoccurred. Images are now successfully transferring from the Pi to the web VM and then to the NFS share on the Synology NAS. I still need to tweak some settings and figure out why time-lapse videos aren’t being created, but otherwise, everything is running smoothly. You can view my AllSky camera at allsky.my7h.com .

aaronwmorris, 2025-02-20:
Very nice!

closed 2025-02-20 by aaronwmorris