aaronwmorris/indi-allsky #1709: ./setup.sh

graememilne, 2024-12-10:
[image]

Hi Aaron
I have had to re-install indi-allsky due to Raspberry pi internet issues.
Clean install of RPI os was required.

I have an error message (see photograph)
How can i resolve this as it is stopping the installation from completing.

Regards
Graeme

aaronwmorris, 2024-12-10:
You may try changing your DNS server on your raspberry pi to 8.8.8.8 or 1.1.1.1 (in /etc/resolv.conf).  I have seen issues like this in the past, and I have always suspected that there might be an old or python wheel mirror out that is cached in DNS.

graememilne, 2024-12-10:
The only txt in file was nameserver and my router ip address.
At the moment the problem has disappeared.
Another problem has shown itself again.
Had a look at the website and rawpy is now 021.3.
Would this have caused this problem?
[image]

aaronwmorris, 2024-12-10:
Another user had a similar issue.  Try setting the following

```
git config --global core.compression 9
```

https://github.com/aaronwmorris/indi-allsky/issues/1688

graememilne, 2024-12-10:
Hi Aaron
Alas not working this time.
At a loss with this.

graememilne, 2024-12-10:
Eventually completed installation.
This worked .........
git config --global core.compression 0
export GIT_TRACE_PACKET=1

Thank you very much for your help Aaron
Regards
Graeme

graememilne, 2024-12-11:
Hi Bluthen

It was a frustrating evening with the software and it was late about 0025utc uk time.
 I eventually got the meteor camera outside.
I was also tired, so i have to apologise  to you for lines you supplied, to try out.
Originally it worked .
But when i check the charts, the bme280 was not working .
Tried the i2c tools,no number showed which means not connected.
That is when i did another ic2 check.
With the indi-allsky software i did another ./setup again .
This is when i ran into the same problem.
I tried different combinations of your lines which eventually sorted the problem.

Checked the indi charts but the BME280 sensor still did not work.
So pulled the unit out of the case .
Discovered i had got the blue and yellow wires  back to front .

Blue was on the SCL
Yellow was on the SDA

Correct wiring is ....

Red  - VCC 3.3
Yellow - SCL i2c
Blue - SDA I2C
Black - Ground  

hope this helps someone who has this sensor .

Over night temperature was approx -6 degrees,this morning  about 0750 it was -3.
Interesting to see the results .
Recorded temperature -2 @0820utc  Met office Dyce airport.
And the dome is clear so the heater is doing its job.
On a finishing note i googled JSQM and now  i know what it means.
But what is SQM Delta ??

graememilne, 2024-12-11:
[image]
Thought i like to share this . 
On checking the star trail and timelapse captured this.
Meteor camera was a RPI HQ with a RPI wide angled 6mm lens

closed 2025-02-20 by aaronwmorris