aaronwmorris/indi-allsky #1713: run indi-allsky on startup.

graememilne, 2024-12-12:
Was running during today ,then sometime in the afternoon it said the software was down .
Tried to reboot ,but still no luck.
i can ssh into the rpi.
I ran this....sudo systemctl --type=service.
No sign of the indi-allsky service.
Where is it ?
Can i recover to get it working again ??

aaronwmorris, 2024-12-13:
The indi-allsky service runs under the user systemd session.  You have to run systemctl as the indi-allsky service user.

```
systemctl --user status indi-allsky.service
```

Also, the indi-allsky service unit is not auto-started, it is started by the indi-allsky timer 2 minutes after boot.  This gives ample time for any firmwares to load and the system to settle after boot.

```
systemctl --user status indi-allsky.timer
```

graememilne, 2024-12-13:
Good morning Aaron
For what ever reason i had to enable the timer and the service.
Now the other issue is that i cannot get the web page up.
How can i recover the web page ??
[image]

As all ways thank you for help,it is most appreciated.
Graeme

aaronwmorris, 2024-12-13:
The web interface runs as a gunicorn service behind an Apache reverse proxy.  The log file is at `/var/log/indi-allsky/webapp-indi-allsky.log`.

```
systemctl --user status gunicorn-indi-allsky
```

graememilne, 2024-12-13:
[image]
It appears web was not active for some reason.
That done now enabled ,web page still not showing.
Hope you can solve this ?

aaronwmorris, 2024-12-13:
I would suggest re-running the setup.sh script.  It would normally fix errors like this.

graememilne, 2024-12-13:
Would I lose the timelapse videos etc ?

aaronwmorris, 2024-12-13:
No, the setup.sh script just performs system setup to ensure the services and dependencies are configured correctly.  It is designed to be re-run often and does not delete anything.

graememilne, 2024-12-13:
Oh,brilliant thank you

graememilne, 2024-12-14:
Sorry to say ,i have tried different combinations from this below,but this time not working. 
git config --global core.compression 0
git config --global http.postBuffer 157286400
export GIT_TRACE_PACKET=1
export GIT_TRACE=1
export GIT_CURL_VERBOSE=1

It appears to be getting stuck on this line.

[image]

aaronwmorris, 2024-12-14:
I do not know why installing this module from source is being so difficult for people.  I have updated how the requirements are handled in #1715 and rawpy should install from a wheel on 64-bit platforms.  Pull down the latest code and see how this works for you.

graememilne, 2024-12-15:
Hi Aaron

Sorry i did not get back to you sooner,

Thank you very much for the update.
I am delighted to say it worked first time .
This update has made a huge difference now.

As i have UFW installed this caused an issue of not being able to display
the web page.
I have put in a couple of rules for port 80 and 443 .
This now allows me to display the web page and viw the videos ,datax etc.
Thank you once again for your help.

closed 2025-02-20 by aaronwmorris