aaronwmorris/indi-allsky #1714: IUC2600KPA horizontal lines

mburge, 2024-12-14:
This is an issue we discussed on CN, I've attached the support information (had to edit There was an error creating your Issue: body is too long (maximum is 65536 characters). Still haven't figured it out, seems to happen across the board, with and without stacking on, day, night, moon. Thanks for any ideas.

/home/astra/dev/indi-allsky/misc/support_info.sh: line 62: warning: command substitution: ignored null byte in input
#################################
### indi-allsky support info  ###
#################################
```

Distribution: debian
Release: 12
Arch: aarch64
Bits: 64

CPUs: 4
Memory: 8245584 kB

System: Raspberry Pi 5 Model B Rev 1.0

Linux wolke 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux

Time
Fri 13 Dec 23:06:57 EST 2024

System timezone
America/New_York

Uptime
 23:06:57 up 53 min,  2 users,  load average: 0.60, 0.54, 0.95

Memory
               total        used        free      shared  buff/cache   available
Mem:         8245584     2103312     2267568       58384     4032288     6142272
Swap:         204784           0      204784

Filesystems
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             3947712        0   3947712   0% /dev
tmpfs             824560     5760    818800   1% /run
/dev/mmcblk0p2 245406872 29300016 203623312  13% /
tmpfs            4122784      800   4121984   1% /dev/shm
tmpfs               5120       48      5072   1% /run/lock
/dev/mmcblk0p1    522230    78266    443964  15% /boot/firmware
tmpfs             824544      160    824384   1% /run/user/1000

sysctl info
vm.swappiness = 60

Thermal info
cpu-thermal
22650

system python: Python 3.11.2

indiserver: /usr/local/bin/indiserver

User info
uid=1000(astra) gid=1000(astra) groups=1000(astra),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),102(input),105(render),106(netdev),115(lpadmin),993(gpio),994(i2c),995(spi),999(systemd-journal)

gpsd user info
id: ‘gpsd’: no such user

Process info
astra      71316  0.3  0.0  83760  3712 ?        Rs   22:57   0:01 /usr/local/bin/indiserver -p 7624 indi_simulator_telescope indi_toupcam_ccd
astra      71318  0.0  0.1  42160 13312 ?        S    22:57   0:00 indi_simulator_telescope
astra      71319  0.3  5.5 980080 455680 ?       Sl   22:57   0:02 indi_toupcam_ccd
astra      71363  0.0  0.2  33936 24064 ?        SNs  22:58   0:00 /home/astra/dev/indi-allsky/virtualenv/indi-allsky/bin/python3 /home/astra/dev/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
astra      71365  1.5  3.0 1115744 253568 ?      SNl  22:58   0:08 /home/astra/dev/indi-allsky/virtualenv/indi-allsky/bin/python3 /home/astra/dev/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
astra      72654  6.0  1.6 517520 138176 ?       SNsl 23:06   0:02 /home/astra/dev/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
astra      72666  5.6  1.9 558352 157456 ?       Sl   23:06   0:02 /home/astra/dev/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
astra      72668 10.8 13.4 1714304 1108960 ?     Rl   23:06   0:04 /home/astra/dev/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run
[... 1198 lines cut ...]
    "USERNAME": "",
    "PASSWORD": "REDACTED",
    "PASSWORD_E": "REDACTED",
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "AIRCRAFT_LABEL_TEMPLATE": "{id:s} {distance:0.1f}km {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:15,300 (Left)\n# anchor:la (Left Justified)\n# color:200,200,200\nAircraft"
  },
  "SATELLITE_TRACK": {
    "ENABLE": true,
    "DAYTIME_TRACK": true,
    "ALT_DEG_MIN": 20,
    "LABEL_ENABLE": true,
    "LABEL_LIMIT": 10,
    "SAT_LABEL_TEMPLATE": "{title:s} {alt:0.1f}° {dir:s}",
    "IMAGE_LABEL_TEMPLATE_PREFIX": "# xy:-15,200 (Right)\n# anchor:ra (Right Justified)\n# color:200,200,200\nSatellites"
  },
  "TIMELAPSE_PRE_PROCESSOR": "standard"
}

```
#################################
###     end support info      ###
#################################

aaronwmorris, 2024-12-17:
Remind me, are you running this directly off your Pi5 or through a powered USB hub?

mburge, 2024-12-17:
Interesting, I hadn't thought of that, let me run it without that and see. I'm also going to move it to a new PI5 instance just in case it is hardware related.

aaronwmorris, 2024-12-20:
Any luck?

aaronwmorris, 2025-04-24:
Closing this for now.  Let me know if you need anymore help!

closed 2025-04-24 by aaronwmorris