aaronwmorris/indi-allsky #1718: Scripts for focuser not working on Raspberry Pi 5 OS 12 bookworm

dbrdirect, 2024-12-15:
Using stepper motor hardware from previous Raspberry Pi 4 installation worked great. Scripts may need to be updated to reflect GPIO change with RPI 5. The attached script, compliments of chat gpt. It works has a work around but i'd rather control from indi-allsky. Unfortunately, I lack the expertise to make the changes myself in the .py scripts /indi-allsky/indi_allsky/devices/focusers.
[stepper_motor_focuser.txt](https://github.com/user-attachments/files/18139750/stepper_motor_focuser.txt)

aaronwmorris, 2024-12-15:
You should pull down the latest release and re-run setup.sh.  The latest version uses the python module that works with Pi5.

dbrdirect, 2024-12-15:
That worked! Thank you! This is why you get paid the big bucks...

closed 2024-12-15 by dbrdirect