aaronwmorris/indi-allsky #1728: GPS not being detected

Vertikar, 2024-12-21:
Able to see GPS via cgps and gpsmon

Logs are showing 0 GPSs found

```
2024-12-21T15:39:57.962943+11:00 rpi5-mobile [INFO] Capture-1-1197/MainThread indi._findGpss() [690]: Searching for available GPS interfaces
2024-12-21T15:39:57.962989+11:00 rpi5-mobile [INFO] Capture-1-1197/MainThread indi.findGps() [710]: Found 0 GPSs
```

Support info shared here: https://gist.github.com/Vertikar/50ea8a511fd10611ce6f5ac61e468df6

I setup gpsd manually, but also ran the setup gps script to make sure it would work.

aaronwmorris, 2024-12-22:
You will need to re-run setup.sh and enable the `indi_gpsd` server in the indiserver setup.  Make sure the indiserver is restarted (it will ask you as part of setup.sh).

Vertikar, 2024-12-22:
> You will need to re-run setup.sh and enable the `indi_gpsd` server in the indiserver setup. Make sure the indiserver is restarted (it will ask you as part of setup.sh).

Thanks! I'll try that tomorrow, running my mobile setup and it rained today so brought it inside.

Vertikar, 2024-12-23:
That sorted it, thanks!
Might be worthwhile adding that to the GPS wiki page, as I did have a look there initially. Tried to see if I could edit it myself, but couldn't find the edit button.

closed 2024-12-23 by Vertikar

morganjrees, 2025-04-12:
just had the same issue re ran the ./setup.sh and en though all setup but was not promoted for the GPS options, I do recall seeing something about GPS in first install but dint need it at that time.

aaronwmorris, 2025-04-12:
@morganjrees Once the indiserver is defined from the original setup, it will prompt if you want to reconfigure the indiserver when you re-run setup.sh.  Only when you reconfigure the indiserver will it ask about the GPS setup.

morganjrees, 2025-04-14:
@aaronwmorris This seems to be the only options:

[image]