aaronwmorris/indi-allsky #1743: Feature suggestion:  Year-long Keogram (Keogram of Keograms?)

zaphus, 2024-12-26:
As we approach the new year, I've been experimenting with the idea of a year-long keogram.  In my experiment I take a few lines of pixels from the center of each nights keogram (horizontally) - them combine those into a slowly increasing year-long keogram
This (I hope) will give me a keogram of what the sky looks like as the year progresses

The times dont align on the keograms, since nights have different lengths - which seems fine, but an alternative would be to scale for the longest night.  Seems like a lot more work, but it would show the length of the nights as well!
Since there would only be 365ish nights, choosing 3 pixels from each image would give a little over 1000 in width

Here is an example of December so far for me, using 3 pixels width per night and with time progressing to the right. Scaled down to a 250px height to be similar to the standard keograms:

[image]

I think there is potential for something interesting there, even though it will take a year to complete the image!

zaphus, 2024-12-26:
thinking about this further, I am really taking a few pixels from the center of the original images.  Maybe theres a version where every 30 minutes (for example) you take a 3x3 (for example) from the center of the most recent image, and paste that into a giant image in the correct location (x-axis is date, y-axis is time) - then you can potentially have the full 24hrs vertically, showing midnight-midnight each day.   I'm going to try this.  Could also be taken from the raw unstretched image

zaphus, 2024-12-28:
continuing to reply to myself - I've managed to build a script to do this, taking sample every 10 minutes (actually 9.5 minutes, so that I never miss one due to system delays).  Here is what the last 11 days looks like, the background being black may be hiding a few misses where the system was down and unable to capture anything.  It also looks good at 1 pixel rather than 3, but thats an easy modification.

[image]

aaronwmorris, 2024-12-28:
I have been thinking about this.  I am considering adding a database table to store the center pixel value of each image.  Ultimately, the a year long keogram is the center line of the keogram of each day, which is just the center pixels.

I have not quite settled on a strategy yet.

caphector, 2024-12-31:
I've wondered about creating analemmas from an all-sky camera. When I experimented with my current exposures I have waaaaaaaaaaay too much sun bloom to make the path of the analemma visible. I had a script running to snap a photo at solar noon every day and I assembled those into this: https://youtu.be/NicaNGACzhA

aaronwmorris, 2025-02-19:
Merged #1894 with the initial support for long term keogram generation.

This support includes a new dedicated database table to store the data for the keogram.  Unfortunately, it does not pull data from the existing keograms.  It starts gathering data at the point to upgrade to this code.

The new long term keogram function stores the values of 5 pixels from every image taken directly to a database table.  One very positive attribute to this system is it is forgiving of missing data.  There will be gaps, but the data will still line up properly.

There is still many enhancements that can be made to the code that generates the keogram, but it is more important that the data starts to be gathered.

aaronwmorris, 2025-02-19:
I even created a script to test creating data, storing the data in a database, and querying the data to generate a long term keogram.

This is test code output for my location for a full year:

[image]

zaphus, 2025-02-19:
> I've wondered about creating analemmas from an all-sky camera. When I experimented with my current exposures I have waaaaaaaaaaay too much sun bloom to make the path of the analemma visible. I had a [...]

There is a way to do this, in theory - if indi-allsky snapped one picture at a specific time each day (say Noon but offset from the other images being taken) but with different settings such that the image was not washed out.  I dont think my camera can get to a low enough shutter speed+gain but it might be possible with some experimentation - this image would not go into the normal location, you dont want it in your timelapses, 
Then you could build the anelemma from those images.

closed 2025-08-24 by aaronwmorris